ZoneMinder/zoneminder #2737: Valgrind results

dsl, 2019-10-22:
Development version, built from source on Debian Buster x64.

Ran valgrind on zma and zmc. Results were very good. This wasn't a long term test, as the machine struggles to keep up with the camera framerate (achieves ~ 3fps, camera set to 10fps).

One possible issue in zma:

`1 errors in context 1 of 20:
Conditional jump or move depends on uninitialised value(s)
    at 0x4838CC5: __strlen_sse2 (vg_replace_strmem.c:462)
    by 0x71B09EE: vfprintf (vfprintf.c:1638)
    by 0x71D779F: vsnprintf (vsnprintf.c:114)
    by 0x71B76BE: snprintf (snprintf.c:33)
    by 0x18989E: snprintf (stdio2.h:67)
    by 0x18989E: Monitor::Analyse() (zm_monitor.cpp:1605)
    by 0x15DB2A: main (zma.cpp:165)
Uninitialised value was created by a heap allocation
    at 0x4835DEF: operator new(unsigned long) (vg_replace_malloc.c:334)
    by 0x189816: Monitor::Analyse() (zm_monitor.cpp:1600)
    by 0x15DB2A: main (zma.cpp:165)`

It also detected a leak from libjpeg in zma, but that might be due to the process receiving a shutdown signal:

`168 bytes in 1 blocks are definitely lost in loss record 225 of 241
    at 0x483577F: malloc (vg_replace_malloc.c:299)
    by 0x4B434C7: jinit_memory_mgr (in /usr/lib/x86_64-linux-gnu/libjpeg.so.62.2.0)
    by 0x4B16E08: jpeg_CreateCompress (in /usr/lib/x86_64-linux-gnu/libjpeg.so.62.2.0)
    by 0x17AE52: Image::WriteJpeg(char const*, int, timeval, bool) const (zm_image.cpp:995)
    by 0x1F1771: Event::WriteFrameImage(Image*, timeval, char const*, bool) (zm_event.cpp:305)
    by 0x1F367C: Event::AddFramesInternal(int, int, Image**, timeval**) (zm_event.cpp:465)
    by 0x1F4ED4: Event::AddFrames(int, Image**, timeval**) (zm_event.cpp:445)
    by 0x189995: Monitor::Analyse() (zm_monitor.cpp:1624)
    by 0x15DB2A: main (zma.cpp:165)`

zmc also has a couple of minor "leaks", look like one offs though:

`4 bytes in 1 blocks are definitely lost in loss record 2 of 242
    at 0x4837B65: calloc (vg_replace_malloc.c:752)
    by 0x511DF66: mysql_init (in /usr/lib/x86_64-linux-gnu/libmariadb.so.3)
    by 0x16A153: zmDbConnect() (zm_db.cpp:39)
    by 0x16BD00: Logger::databaseLevel(int) (zm_logger.cpp:363)
    by 0x16E57F: Logger::initialise(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Logger::Options const&) (zm_logger.cpp:216)
    by 0x16FD4F: logInit(char const*, Logger::Options const&) (zm_logger.cpp:607)
    by 0x15DE1B: main (zmc.cpp:190)`

and

`128 bytes in 1 blocks are definitely lost in loss record 220 of 242
    at 0x4837B65: calloc (vg_replace_malloc.c:752)
    by 0x511DF85: mysql_init (in /usr/lib/x86_64-linux-gnu/libmariadb.so.3)
    by 0x16A153: zmDbConnect() (zm_db.cpp:39)
    by 0x16BD00: Logger::databaseLevel(int) (zm_logger.cpp:363)
    by 0x16E57F: Logger::initialise(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Logger::Options const&) (zm_logger.cpp:216)
    by 0x16FD4F: logInit(char const*, Logger::Options const&) (zm_logger.cpp:607)
    by 0x15DE1B: main (zmc.cpp:190)`

connortechnology, 2019-10-22:
Unfortunately our most worrisome leak won't be reported by valgrind.  It is in gpu mem allocated for hardware accelerated h264 decoding. @pliablepixels reports that we leak gpu mem on each decoder restart.

dsl, 2019-10-22:
Yeah, it was really just a quick run to catch any low hanging fruit. Is the gpu memory leak logged somewhere?

connortechnology, 2019-10-22:
No I havn't even confirmed it myself.  It requires a flakey video stream causing the capture code to reset.

closed 2019-12-28 by stale[bot]