ZoneMinder/zoneminder #2739: cat: /etc/zm/conf.d/zmcustom.conf: No such file or directory

Malathi15, 2019-10-23:
I'm using ubuntu 16.04 , I followed the steps from **Easy Way: Debian Jessie** when I tried to run this commend `cat /etc/zm/conf.d/zmcustom.conf` 

I got this error
`cat: /etc/zm/conf.d/zmcustom.conf: No such file or directory`

I have no file called zmcustom.conf. help me to solve this.
Thanks in Advance

connortechnology, 2019-10-23:
zmcustom.conf is a file that might get created on upgrade. It is not generated during clean install.  It is ok for it to not exist.  It is a file where you would put changes to the default config values.

closed 2019-10-23 by connortechnology

Malathi15, 2019-10-24:
@connortechnology  Thanks for your reply.
 After completing all the steps in Easy Way: Debian Jessie, when I open `http://ip/zm`
I got this message
`Cannot write to content dirs('events','images'). Check that these exist and are owned by the web account user`
events and images directors are exist. Help me to solve this.

Thanks in Advance