ZoneMinder/zoneminder #2740: Documentation updates checklist for 1.34

pliablepixels, 2019-10-23:
**USER GUIDE:**
https://zoneminder.readthedocs.io/en/latest/userguide/components.html
- [ ] Remove zmf  @connortechnology 
- [ ] review rest is ok @connortechnology 
- [ ] Do we need to mention ES anywhere is components and link to the ES docs or leave it out for now, esp. since ES invocation is built into ZM now (opinions from @connortechnology, owner @pliablepixels  )

https://zoneminder.readthedocs.io/en/latest/userguide/gettingstarted.html
- [ ] Redo all screenshots to show 1.34 @pliablepixels 
- [ ] See if we can do boxes with an object editor like draw.io so we can move around later instead  of redrawing @pliablepixels 
- [ ] Correct authentication to explain tokens and legacy @pliablepixels 
- [ ] Rework understanding web console to include new 1.34 stuff @pliablepixels 
- [ ] Review/fix Adding new monitors as needed @pliablepixels 

https://zoneminder.readthedocs.io/en/latest/userguide/definemonitor.html#
- [ ] remove remote type and other fixes in defining monitors process @pliablepixels 
- [ ] review all tabs in this process (except X10) and update to 1.34 @pliablepixels  
- [ ] X10 part needs review/correction

https://zoneminder.readthedocs.io/en/latest/userguide/definezone.html
- [ ]  Content seems ok. Don't recall anything new here
- [ ] Merge zoneminder zones for dummies into this manual @pliablepixels 

https://zoneminder.readthedocs.io/en/latest/userguide/viewmonitors.html
 - [ ] use a different image @pliablepixels 

https://zoneminder.readthedocs.io/en/latest/userguide/filterevents.html
- [ ] review all parts and fix as needed 

https://zoneminder.readthedocs.io/en/latest/userguide/viewevents.html
-  [ ] Add FIFO stuff

https://zoneminder.readthedocs.io/en/latest/userguide/options.html
There are 14 sections:
- [ ] @pliablepixels  will review/update Display, System, Config, Web, Images, Logging, BW, Users
- [ ] review update Servers, Paths, Network, Email, Upload, X10

https://zoneminder.readthedocs.io/en/latest/userguide/cameracontrol.html
- [ ] Seems to be ok. Not sure if any of the PTZ changes Isaac made affect this. @connortechnology to modify if anything needs to be done here

https://zoneminder.readthedocs.io/en/latest/userguide/cameracontrol.html#controlling-monitors
- Seems to be ok. Not sure if any of the PTZ changes Isaac made affect this. @connortechnology to modify if anything needs to be done here

https://zoneminder.readthedocs.io/en/latest/userguide/mobile.html
- [ ] review, remove eyeZM completely @pliablepixels 
- [ ] remove zmview as well - no longer maintained @pliablepixels 
- [ ] add surveillance pro (if its still maintained)

[ ] Do we need to add a brief section on machine learning? Its a hot topic, lots of people use OD. Does it need to have a brief mention in this manual with a link to the ES/OD docs? @connortechnology  to review @pliablepixels to author, if approved

https://zoneminder.readthedocs.io/en/latest/userguide/logging.html
- [ ] @connortechnology to review. Should this section not explain to do to INFO/DEBUG logs, how to log to a file, how to inspect?

https://zoneminder.readthedocs.io/en/latest/api.html
- [ ] Updated already with tokens/auth @pliablepixels 
- [ ] Updated `stateful` need @connortechnology 
- [ ] overall review for consistency @pliablepixels 

https://zoneminder.readthedocs.io/en/latest/faq.html
- [ ] General review needed to see if all suggestions are valid
- [ ] Storage space calculator link 

https://zoneminder.readthedocs.io/en/latest/contributing.html
- [ ] Seems fine

pliablepixels, 2019-10-23:
**INSTALL GUIDE**:

pliablepixels, 2019-10-24:
Ok. On ES, I don't plan to import all doc. Simply show it as an optional component in the component diagram and a brief para at the right place (where there is an option to ES enablement) with a link to the already available ES documentation. That way ES document updates are not dependent on ZM releases.

jcwren, 2020-02-10:
Not sure if this should be a new issue for documentation, a program change, or both. When defining a new zone, the error dialog for missing fields do not correspond with the names on the page or the documentation at https://zoneminder.readthedocs.io/en/latest/userguide/definezone.html.

minimum alarm pixel count is presumably the Min field of 'Min/Max Alarmed Area'.
minimum filter pixel count is presumably the Min field of  'Min/Max Filtered Area'.
minimum blob pixel count is presumably the Min field of 'Min/Max Blob Area'.
minimum blob count is presumably the Min field of 'Min/Max Blobs'.

The Min/Max Alarmed Area documentation states "A good starting point for both the minimum and maximum are 0 and 0, indicating that any number of alarmed pixels (or any percentage) greater than 0 will trigger an alarm.", however attempting to use 0 for the minimum throws a dialog that says 'You must specify the minimum alarm pixel count'. Entering 1 and 0 allows entry, not sure if it works yet (haven't tried to trip the alarm).

[image]

jcwren, 2020-02-10:
Additionally, there are some new fields in the Misc tab when defining a monitor that are not covered on the https://zoneminder.readthedocs.io/en/latest/userguide/definemonitor.html page (Motion Frame Skip, Analysis Update Delay, Default Rate, Default Method For Live View, Signal Check Points, Signal Check Colour).

pliablepixels, 2020-02-10:
Please see if you can make PRs for missing elements

jcwren, 2020-02-10:
> Please see if you can make PRs for missing elements

I'd like to, but I have no idea what they mean. I'm new to ZoneMinder, and trying to get the object detection working. No luck yet, but I'll figure it out :)

bearmeister, 2020-03-12:
Greetings, 
I noticed there is some email tokens available which do not seem to appear in the current documentation, namely ; 

%EI1A%  generateImage($Event, $first_alarm_frame, 'analyse');
%EIMA% generateImage($Event, $max_alarm_frame, 'analyse');

These appear to have been introduced as part of pull 1814

1768 fix email tokens for h264 storage #1814
> vpoluyaktov commented on Mar 17, 2017
> Also two new email tokens were added:
> %EI1A% - attach first alarm "analyze" frame (similar to existing %EI1% but analyze frame)
> %EIMA% - attach alarm "analyse" frame having max score (similar to existing %EIM% but analyze frame)

Kind regards

PhantomGames-glitch, 2020-06-07:
Beta tester for the update

sojab0on, 2025-06-06:
i wanted to test the WSL guide , but still not posted its been more then 3 years now