ZoneMinder/zoneminder #2744: Error when forcing alarm on monitor using API v2

RollingMeadows, 2019-10-28:
Running latest ZoneMinder 1.33.14 built using sources from Github repo on
Fedora Linux (Nginx)

As recommended I switch to the recent API v2.  Most of the calls work fine
except when forcing alarms on monitors

Both of these options are enabled
OPT_USE_API
OPT_USE_LEGACY_API_AUTH

Interesting the following command worked fine under the legacy API, but now both new and old fail with the error listed below

curl -d token=ACCESS_TOKEN http://localhost/api/monitors/alarm/id:1/command:on.json  NOTE: (status|on|off) all produce the error
 

{"success":false,"data":{"name":"An Internal Error Has Occurred.","message":"An Internal Error Has Occurred.","url":"\/api\/monitors\/alarm\/id:1\/command:on.json?p=%2Fmonitors%2Falarm%2Fid%3A1%2Fcommand%3Aon.json","exception":{"class":"InternalErrorException","code":500,"message":"Internal Server Error"}}}

The server log shows:
----
[error] 915#0: *456 FastCGI sent in stderr: "PHP message: PHP Fatal error:  Call to a member function Read() on a non-object in /usr/share/html/zm/api/app/Controller/MonitorsController.php on line 303" while reading response header from upstream, client: 127.0.0.1, server: localhost, request: "POST /api/monitors/alarm/id:1/command:on.json HTTP/1.1", upstream: "fastcgi://127.0.0.1:9000", host: "localhost:8520"

Tried setting AUTH_RELAY to none, plain, and hashed. But errors persists.  Any suggestions on what may be happening...

pliablepixels, 2019-11-01:
If you pass the token as as query parameter it will work. As it turns out CakePHP discards payload data for HTTP GET requests as only POST/PUT APIs generally have data payloads.

RollingMeadows, 2019-11-02:
Indeed passing token to the query URL fixes this issue.  Closing issue

closed 2019-11-02 by RollingMeadows