ZoneMinder/zoneminder #2745: ZoneMinder consistent behavior on extracting alarm/snapshot/specific frames for object detection

pliablepixels, 2019-10-29:
Keeping this as an issue for @connortechnology  to look into:

Given that a lot of folks are using object detection, a common issue that is reported is that the image that is used for analysis while the event is recording is not the same as the image that is used if the detection is invoked _after_ the event ends. This causes object detection failure and folks are always confused on why detection works well if run manually.

The latest issue that discusses this is [here](https://github.com/pliablepixels/zmeventnotification/issues/165)

Summarizing ideal behavior so we can address this:

a) `alarm` image should be immediately written when the event triggers 
b) `snapshot` image should be immediately written when the event triggers - with the caveat that snapshots can change over time as the event proceeds

To simplify implementation the following suggestions were discussed on slack
c) We write `alarm.jpg` and `snapshot.jpg` irrespective of whether jpeg store is enabled or disabled in `zm_event.cpp`
d) We simplify `image.php` to search for `snapshot.jpg` and `alarm.jpg` instead of doing DB searches when `fid=alarm|snapshot` is provided

**Open:**
e) This is one tiny issue, however. If you have jpeg store _off_ then you can't do `fid=nn` while the video is recording. This is because the mp4 is still recording and seems the toc isn't written yet. Curious if there is a way around it?

pliablepixels, 2019-10-31:
@selvanair the changes were merged to master - I don't use JPEG mode, but will set one up and test. Feel free to test master.

selvanair, 2019-10-31:
@pliablepixels did a quick test, and works well. With jpeg frames saved, the correct snapshot image is now returned and all other fid values work as before. Thanks.

For me the saved mp4 has a start time jitter of +/-1 seconds from the actual event start likely because of only one the `I` frame per second in the source. That makes frames generated on the fly mismatch the database scores by several frames. The alarm and snapshot frames are not affected by this but other frames are. This is another reason why I'm sticking with storing jpegs -- useful for experimenting with OD in frames between alarm and snapshot etc.

@connortechnology Thanks for this fix.

closed 2019-11-01 by pliablepixels