ZoneMinder/zoneminder #2747: The start point of stored event video is not quiet right

selvanair, 2019-11-01:
ZM Version: _1.33.14_
OS: _GNU/Linux Debian 10 (buster)_

Some relevant parameters:

_analysis fps = 5
captured video is 15 fps nominal
pre-event-count = 10 (i.e. 2 seconds using 5 fps).
image buffer = 100_

**What happens:**
The stored video often starts less than 2 seconds before the alarm start time
**Expected:**
The pre-event segment should be at least 2 seconds (could be more because of the need to start at an IDR frame (keyframe)).

Looking at debug logs I see the captured packet queue is routinely cleared keeping at least 10 frames which leads to 11 frames in the queue in the worst case to 40 frames in the best case at any time. That is 0.7 sec to 2.7 sec of video, very often much less than the required 2 seconds (in my case).

```
Log:
10/31/19 20:53:22.104711 zmc_m1[26779].DB3-zm_packetqueue.cpp/131 [Clearing all but 10 frames, queue has 30]
10/31/19 20:53:22.104926 zmc_m1[26779].DB3-zm_packetqueue.cpp/190 [Deleted 15 packets, 15 remaining]
```

Consistent with this, the logs also show that `clear_unwanted_packets()` called by `CaptureAnd Record()` doesn't  find a point in the packet queue to start the event recording and ends up keeping the whole queue.

```
Log:
10/31/19 20:53:56.058295 zmc_m1[26779].DB1-zm_packetqueue.cpp/255 [Didn't find a frame before event start time. keeping all]
```

I think the pre-event frame count (10 in my case) should be scaled by capture_fps/analysis_fps while trimming the queue. Same logic should also apply with `clear_unwanted_packets()` called before event clip storage. Even better, convert that number internally to a time interval while manipulating packets as capture-fps is somewhat variable.

selvanair, 2019-11-04:
> a camera at 15fps means that 10 pre-event frames is 2/3 of a second. You should set it to 30 to get 2 seconds of pre-event.

That would retain enough packets in the video buffer, but will give incorrect number of pre-event frames. As far as I can see, jpeg frames are always defined using the analysis fps and that will save 30 frames lasting for 6 seconds. Even if jpegs are not saved, the frames data in the db will have 30 pre-event frames with an approximate delta time of 0.2 seconds (5 fps). Same with post event frames. 

And, frames recreated from the video will be off too as they are generated using the data in the db.

Looking at the code, all reference to frames refer to the analysis phase (i.e. at analysis fps), so using that appropriately while computing offsets for the video seems to be the right fix.

selvanair, 2019-11-04:
I have fixed this locally by
(i) keep enough frames in the video buffer (packet queue)  to cover the pre-event duration.
(ii) add an offset to frame's delta time saved in the db to account for the difference between video start time and event start time. The former could be a few frames earlier depending on where the key-frame appears.

Note: (ii) is needed to correctly generate frames from the video as delta is passed as the seek time to ffmpeg and should refer to the start of the saved video clip. 

If that sounds reasonable, I can open a PR.

connortechnology, 2019-11-30:
Would love a PR.  To properly fix this requires a huge refactor, which won't happen until 1.35.  
I agree that a quick fix would be to figure out how many additional frames will be stored in the video and adjust the deltas.

selvanair, 2019-11-30:
Fixed by PR #2768

closed 2019-11-30 by selvanair