ZoneMinder/zoneminder #2748: permission denied to unifi nvr camera

junster1, 2019-11-02:
CentOS Linux release 7.7.1908 (Core)
zoneminder-1.32.3-1.el7.x86_64 via RPMFusion
Unifi G3 cameras via unifi NVR

- ZoneMinder Source Type:
```ffmpeg
rtsp://192.168.1.x:7447/XXXXXXXXXXXXXXXXXXXX_0
```

**Describe the bug**
I can connect to the device and capture packets via ffmpeg directly with the following command line.
```ffmpeg -i rtsp://192.168.1.x:7447/XXXXXXXXXXXXXXXXXXXX_0 my.mp4
Input #0, rtsp, from 'rtsp://192.168.1.x:7447/XXXXXXXXXXXXXXXX_0':
  Metadata:
    title           : FCECDAD89881_0
  Duration: N/A, start: 0.078095, bitrate: N/A
    Stream #0:0: Audio: aac (LC), 11025 Hz, mono, fltp
    Stream #0:1: Video: h264 (Main), yuvj420p(pc, bt709, progressive), 1920x1080 [SAR 189:190 DAR 168:95], 30 fps, 30 tbr, 90k tbn, 60 tbc
Stream mapping:
  Stream #0:1 -> #0:0 (h264 (native) -> h264 (libx264))
  Stream #0:0 -> #0:1 (aac (native) -> aac (native))
Press [q] to stop, [?] for help
[aac @ 0x19d05a0] Too many bits 6408.707483 > 6144 per frame requested, clamping to max
[h264 @ 0x1ae0f20] error while decoding MB 64 10, bytestream -10
[h264 @ 0x1ae0f20] concealing 6945 DC, 6945 AC, 6945 MV errors in I frame
[libx264 @ 0x19b88a0] using SAR=189/190
[libx264 @ 0x19b88a0] using cpu capabilities: MMX2 SSE2Slow SlowCTZ
[libx264 @ 0x19b88a0] profile High, level 4.0
[libx264 @ 0x19b88a0] 264 - core 148 r2795 aaa9aa8 - H.264/MPEG-4 AVC codec - Copyleft 2003-2017 - http://www.videolan.org/x264.html - options: cabac=1 ref=3 deblock=1:0:0 analyse=0x3:0x113 me=hex subme=7 psy=1 psy_rd=1.00:0.00 mixed_ref=1 me_range=16 chroma_me=1 trellis=1 8x8dct=1 cqm=0 deadzone=21,11 fast_pskip=1 chroma_qp_offset=-2 threads=18 lookahead_threads=3 sliced_threads=0 nr=0 decimate=1 interlaced=0 bluray_compat=0 constrained_intra=0 bframes=3 b_pyramid=2 b_adapt=1 b_bias=0 direct=1 weightb=1 open_gop=0 weightp=2 keyint=250 keyint_min=25 scenecut=40 intra_refresh=0 rc_lookahead=40 rc=crf mbtree=1 crf=23.0 qcomp=0.60 qpmin=0 qpmax=69 qpstep=4 ip_ratio=1.40 aq=1:1.00
Output #0, mp4, to 'my.mp4':
  Metadata:
    title           : FCECDAD89881_0
    encoder         : Lavf57.83.100
    Stream #0:0: Video: h264 (libx264) (avc1 / 0x31637661), yuvj420p(pc), 1920x1080 [SAR 189:190 DAR 168:95], q=-1--1, 30 fps, 15360 tbn, 30 tbc
    Metadata:
      encoder         : Lavc57.107.100 libx264
    Side data:
      cpb: bitrate max/min/avg: 0/0/0 buffer size: 0 vbv_delay: -1
    Stream #0:1: Audio: aac (LC) (mp4a / 0x6134706D), 11025 Hz, mono, fltp, 66 kb/s
    Metadata:
      encoder         : Lavc57.107.100 aac
```
**Expected behavior**
In the logs I see permission denied, not sure where that is occurring.
```Nov 02 00:39:06 zmfilter_2[13491]: INF [Checking filter Update DiskSpace update disk space returned 0 events ]
Nov 02 00:39:12 zmc_m2[13693]: INF [zmc_m2] [Priming capture from rtsp://192.168.1.X:7447/XXXXXXXXXXXXXXXXXXX_0]
Nov 02 00:39:12  zmc_m2[13693]: ERR [zmc_m2] [Unable to open input rtsp://192.168.1.X:7447/XXXXXXXXXXXXXXXXX_0 due to: Permission denied]
Nov 02 00:39:12 zmc_m2[13693]: ERR [zmc_m2] [Failed to prime capture of initial monitor]
```
**Debug Logs**
```

<insert debug logs here, please make sure they are within the ``` quotes so they are formatted properly>

```

junster1, 2019-11-02:
I am also able to connect using VLC media player just fine with the URL via the unifi NVR to the camera.

connortechnology, 2019-11-04:
My initial thought is: Do you need to put username/pasword in the url?

You are going to have to update to 1.33.14.  It has better ffmpeg logging.
https://zmrepo.zoneminder.com

junster1, 2019-11-05:
I upgraded to that version and am still getting the same error in systemctl status zoneminder.

`
[root@zoneminder zoneminder]# systemctl status zoneminder
● zoneminder.service - ZoneMinder CCTV recording and security system
   Loaded: loaded (/usr/lib/systemd/system/zoneminder.service; enabled; vendor preset: disabled)
  Drop-In: /usr/lib/systemd/system/zoneminder.service.d
           └─zm-httpd.conf
   Active: active (running) since Mon 2019-11-04 22:30:06 CST; 5s ago
  Process: 6983 ExecStop=/usr/bin/zmpkg.pl stop (code=exited, status=0/SUCCESS)
  Process: 6999 ExecStart=/usr/bin/zmpkg.pl start (code=exited, status=0/SUCCESS)
 Main PID: 7014 (zmdc.pl)
   CGroup: /system.slice/zoneminder.service
           ├─7014 /usr/bin/perl -wT /usr/bin/zmdc.pl startup
           ├─7044 /usr/bin/zmc -m 2
           ├─7063 /usr/bin/perl -wT /usr/bin/zmfilter.pl --filter_id=1 --daemon
           ├─7066 /usr/bin/zma -m 2
           ├─7078 /usr/bin/perl -wT /usr/bin/zmfilter.pl --filter_id=2 --daemon
           ├─7091 /usr/bin/perl -wT /usr/bin/zmaudit.pl -c
           ├─7103 /usr/bin/perl -wT /usr/bin/zmwatch.pl
           ├─7116 /usr/bin/perl -wT /usr/bin/zmtelemetry.pl
           └─7126 /usr/bin/perl -wT /usr/bin/zmstats.pl

Nov 04 22:30:05 zoneminder zmfilter_2[7078]: INF [Scanning for events using filter id '2']
Nov 04 22:30:05 zoneminder zmdc[7014]: INF ['zmwatch.pl' starting at 19/11/04 22:30:05, pid = 7103]
Nov 04 22:30:05 zoneminder zmdc[7103]: INF ['zmwatch.pl' started at 19/11/04 22:30:05]
Nov 04 22:30:06 zoneminder zmwatch[7103]: INF [Watchdog starting, pausing for 30 seconds]
Nov 04 22:30:06 zoneminder zmdc[7116]: INF ['zmtelemetry.pl' started at 19/11/04 22:30:06]
Nov 04 22:30:06 zoneminder zmdc[7014]: INF ['zmtelemetry.pl' starting at 19/11/04 22:30:06, pid = 7116]
Nov 04 22:30:06 zoneminder zmdc[7014]: INF ['zmstats.pl' starting at 19/11/04 22:30:06, pid = 7126]
Nov 04 22:30:06 zoneminder. zmdc[7126]: INF ['zmstats.pl' started at 19/11/04 22:30:06]
Nov 04 22:30:06 zoneminder systemd[1]: Started ZoneMinder CCTV recording and security system.
Nov 04 22:30:06 zoneminder zmstats[7126]: INF [Stats Daemon starting in 30 seconds]
[root@zoneminder zoneminder]# ls -arlt
total 4
drwxrwxrwx.  2 apache apache    6 Nov  2 23:17 .
drwxr-xr-x. 11 root   root   4096 Nov  3 03:26 ..
[root@zoneminder zoneminder]# rpm -qa |grep zoneminder
zoneminder-common-1.33.14-1.495.20191103gitg4e16e9052.el7.x86_64
zoneminder-1.33.14-1.495.20191103gitg4e16e9052.el7.x86_64
zoneminder-httpd-1.33.14-1.495.20191103gitg4e16e9052.el7.x86_64
`

There are no logs in /var/log/zoneminder.

As far as auth, I don't have to supply it either on the command line to ffmpeg or VLC and it works, so I don't think that is the issue..

connortechnology, 2019-11-30:
Change the following settings under Options -> Logging:
LOG_LEVEL_FILE => Debug
LOG_FFMPEG => checked
LOG_DEBUG => checked
LOG_DEBUG_TARGET => _zmc
LOG_DEBUG_LEVEL => 3
LOG_DEBUG_FILE=> empty

closed 2020-02-06 by stale[bot]