ZoneMinder/zoneminder #2750: Hyperlinks to nph-zms are incorrect when the web console is accessed trough a tunneled connection.

mkaniewski, 2019-11-03:
**Describe Your Environment**
- Version of ZoneMinder: v1.32.3
- How you installed ZoneMinder: FreeBSD pkg
- Full name and version of OS: FreeBSD 12.0-RELEASE 
- Browser name and version (if this is an issue with the web interface): Google Chrome 76

**Describe the bug**
My ZM web console listen on port 80 (HTTP, nginx). I use an SSH tunnel to connect to it remotely:
`ssh user@zmwebserver -L 11111:localhost:80`
therefore all my requests to the ZM console looks like that:
`http://localhost:11111/zm/something`

Most of the things works correctly however when I try to play any camera stream I get a wrong link, eg:
`<img id="liveStream1" src="localhost:80/cgi-bin/nph-zms?(...)>`

It is caused by the fact that `Server.php` uses `$_SERVER['SERVER_PORT'];` to construct the URL. However in my case the port used by the server is not the one I use in the SSH tunnel. In a result the generated link is incorrect.

To fix this problem I came up with the bellow fix:
```
--- a/web/includes/Server.php
+++ b/web/includes/Server.php
@@ -89,7 +89,11 @@ class Server extends ZM_Object {
   }
 
   public function UrlToZMS( $port = null ) {
-    return $this->Url($port).$this->PathToZMS();
+    if ( $port != NULL ) {
+      return $this->Url($port).$this->PathToZMS();
+    } else {
+      return $this->PathToZMS();
+    }
   }
```
So if no special port is specified (it would be specified only if the `ZM_MIN_STREAMING_PORT` option would in use) then I let ZM web console to generate a relative link.

Let me know if this solution is acceptable then I will do a pull request.

**To Reproduce**
Connect to the ZM Web interface using SSH tunnel with a port different then the port used by the ZM web server.

**Expected behavior**
I expect that if I use `http://localhost:11111` in my web browser then all hyperlinks generated in ZM web console will:
1. be relative or
2. have the same domain:port at the beginning of an URL.

connortechnology, 2019-11-04:
This would break the multi-server case.  zms for live view has to run on the server that zmc is running on.  So a relative url might not reference the right server.  So you would have to add a check for $this->{'Id'} as well to determine that we are using the default single-server case.

I'll commit some code.  You will need to update to latest master to test.

closed 2019-11-04 by connortechnology

wjwoods, 2019-11-07:
The commit broke multi-port for me.  Could it be addressed by checking for both $this->Id and $port before returning just the URL?

@@ -91,7 +91,9 @@
   public function UrlToZMS( $port = null ) {
     if ( $this->Id() ) {
       return $this->Url($port).$this->PathToZMS();
-    }
+    } elseif ( $port != NULL ) {
+      return $this->Url($port).$this->PathToZMS();
+    } 
     return $this->PathToZMS();
   }

mkaniewski, 2019-11-17:
@connortechnology Thanks for your patch. It resolves my initial issue however I find out that exactly the same problem (wrong port number in an URL) is present in when I try to:
1. watch an event,
2. see a camera view using "Montage" tab.

URLs on these pages are generated by `web/includes/Event.php` and `web/includes/Monitor.php`, respectively. Both of these files use function `Url()` from `Server.php` so my current solution is to modify this function:
```
--- a/web/includes/Server.php
+++ b/web/includes/Server.php
@@ -97,6 +97,9 @@ class Server extends ZM_Object {
   }
 
        public function Url( $port = null ) {
+    if ( $this->Id() == null and $port == null ) {
+       return '';
+    }
     $url = $this->Protocol().'://';
                $url .= $this->Hostname();
     if ( $port ) {
```
It works for all three cases I already had.

@wjwoods I think your problem Was fixed in https://github.com/ZoneMinder/zoneminder/commit/59b9220dddb644986c5fc62a75043a0c3bbc2f30.

connortechnology, 2019-11-21:
I merged something similarlet me know if it works for you.

mkaniewski, 2019-11-21:
Yes, commit https://github.com/ZoneMinder/zoneminder/commit/1dc83700c0ca263f9b352039cd40de1956537795 fixed all my issues. Thanks!

commit d218d2ee (2019-11-04): If the default server case, don't add the Url component.  So this will return just the path, using whatever servername/port combo is currently in use. Fixes #2750