ZoneMinder/zoneminder #2751: I want the option to build from source

sharonfiles, 2019-11-11:
I want the option to build from source, and I don't want anyone doing anything for me, which would be the point of installing a security system to begin with.

SteveGilvarry, 2019-11-11:
Who said you couldn't? Install from Source Discouraged means packages are a better option for most people, built by someone else or build your own packages. But you want to install from source go for it. Will really satisfy that need to understand every detail.
Closing out as not a bug. Discussions etc we use the forums for, to keep actual bug tracker clean, thanks for your understanding

closed 2019-11-11 by SteveGilvarry

sharonfiles, 2019-11-12:
This feels like censorship.

connortechnology, 2019-11-12:
It's not. Your post is still here for all to see.