ZoneMinder/zoneminder #2762: multiserver and storage

Cramac, 2019-11-23:
Hello. 
Add disk data on other servers to the database

id | Name | Path | Type | Scheme | Server | DiskSpace | Mark
-- | -- | -- | -- | -- | -- | -- | --
3 | Dav | /var/cache/zoneminder/events | local | Medium | Dav | 253.06GB of 1.78TB |  
0 | Default | /var/cache/zoneminder/events | local | Medium |   | 253.06GB of 1.78TB |  
2 | Elg | /var/cache/zoneminder/events | local | Medium | Elg | 253.06GB of 1.78TB |  
1 | Kur | /var/cache/zoneminder/events | local | Medium | Kur | 253.06GB of 1.78TB |  
5 | Kur7tb | /var/cache/zoneminder/events7tb | local | Medium | Kur | 253.06GB of 1.78TB

in mysql

INSERT INTO `Storage` (`Id`, `Path`, `Name`, `Type`, `Url`, `DiskSpace`, `Scheme`, `ServerId`, `DoDelete`) VALUES
(0, '/var/cache/zoneminder/events', 'Default', 'local', NULL, 0, 'Medium', NULL, 1),
(1, '/var/cache/zoneminder/events', 'Kur', 'local', NULL, 0, 'Medium', 3, 1),
(2, '/var/cache/zoneminder/events', 'Elg', 'local', NULL, 0, 'Medium', 4, 1),
(3, '/var/cache/zoneminder/events', 'Dav', 'local', NULL, 127174589533, 'Medium', 2, 1),
(5, '/var/cache/zoneminder/events7tb', 'Kur7tb', 'local', NULL, 9675922155, 'Medium', 3, 1);

should be:
dav - 1.78tb
kur  - 145mb
kur7tb - 7tb

Cramac, 2019-11-24:
I also suggest introducing verification when deleting the storage. If there are entries on it, prohibit deleting. That there would be no errors in consequence

connortechnology, 2019-11-26:
The diskspace values are returned from php's disk_free_space functions.  So to get proper values, each server needs to have each area mounted and each to a unique path.

Cramac, 2019-11-27:
if so, 5 storage should have a different size.
he with a unique path

connortechnology, 2019-12-01:
How about we see the output of df on the machine hosting zm?

Cramac, 2019-12-02:
server main
`root@camera:~# df
Файл.система                 1K-блоков Использовано   Доступно Использовано% Cмонтировано в
udev                           2989464            0    2989464            0% /dev
tmpfs                           610376          920     609456            1% /run
/dev/mapper/camera--vg-root 1916153032    192573524 1626221480           11% /
tmpfs                          3051868      1328448    1723420           44% /dev/shm
tmpfs                             5120            0       5120            0% /run/lock
tmpfs                          3051868            0    3051868            0% /sys/fs/cgroup
/dev/sda1                       240972       141152      87379           62% /boot
tmpfs                           610372            0     610372            0% /run/user/0`

server 2
`root@camera2:~# df
Filesystem      1K-blocks     Used  Available Use% Mounted on
udev              8155064        0    8155064   0% /dev
tmpfs             1643892     1284    1642608   1% /run
/dev/sdb2       152268328 33547096  110916716  24% /
tmpfs            11507236  4406456    7100780  39% /dev/shm
tmpfs                5120        0       5120   0% /run/lock
tmpfs             8219452        0    8219452   0% /sys/fs/cgroup
/dev/loop0          91264    91264          0 100% /snap/core/8039
/dev/loop1          90624    90624          0 100% /snap/core/7270
/dev/sdb1          523248     6152     517096   2% /boot/efi
/dev/sda1      7751366384 43684476 7316964248   1% /var/cache/zoneminder/events7tb
tmpfs             1643888        0    1643888   0% /run/user/0`

connortechnology, 2019-12-02:
So you don't have events7tb mounted on server main, so it doesn't know anything about it.  
Which server are you pointing your browser at ? Judging by the size of / you are hitting main.  
Both servers need the fs's mounted at the same place. Use nfs or samba to achieve this.

Cramac, 2019-12-02:
Yes, they understood correctly. Why complicate and mount everything in one place? Isn't it better to remotely update local disk information?
can be storage table add another field:
**DiskSpaceAvailable**
and transfer the size of the given disk into it?
and display data from a table...

connortechnology, 2019-12-02:
I await your PR

Cramac, 2019-12-02:
my PR?

connortechnology, 2019-12-02:
I'm suggesting that if you want ZM to work differently than it does, you may have to code it yourself.  PR stands for Pull Request.

Cramac, 2019-12-05:
there are too many tied to these paths. you can add in the filter so that each server only cleans its own.

Cramac, 2019-12-05:
so is the audit. Now it deletes from all servers records older than a day (set in the zm config).

Cramac, 2019-12-06:
that's what happened. Suddenly someone comes in handy. So far, only an zmaudit. It takes the server name from the config (/etc/zm/), finds its ID in the database, and then uses only the storage of this server (this was implemented)

[zmaudit.pl.txt](https://github.com/ZoneMinder/zoneminder/files/3931416/zmaudit.pl.txt)

connortechnology, 2019-12-06:
Please note that you should not be running zmaudit continuously anymore.  You should only run it manually if you have had a crash or something.  And you can specify on it's command line which storage area to look at.

Still, at this time, multi-server assumes that every server has access to all storage areas. There are many places that make this assumption.

Cramac, 2019-12-20:
if you are interested, my filter option
[zmfilter.pl.txt](https://github.com/ZoneMinder/zoneminder/files/3988659/zmfilter.pl.txt)
on each server it processes only its storages and monitors

closed 2020-02-28 by stale[bot]