ZoneMinder/zoneminder #2764: api - cookies do not come

Cramac, 2019-11-27:
zm 1.33.14~20191122121850-bionic

login:
curl -XPOST -d "user=admin&pass=pass" -c cook.txt http://video.yotabit.ru/zm/api/host/login.json

return:
``{"access_token":"eyJ0eXAiOiJKV1QiLCJh.....qPCa292S7oGzGTM","access_token_expires":3600,"refresh_token":"eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ....w4MOOX7IKy8h4qw9-XMXIva8mB0W68","refresh_token_expires":86400,"credentials":"auth=199ad676.....09d6757f84","append_password":0,"version":"1.33.14","apiversion":"2.0"}

but no cookies
`[root@host zm]# ls
[root@host zm]# curl -XPOST -d "user=admin&pass=pass" -c cook.txt http://video.yotabit.ru/zm/api/host/login.json
{"access_token":"eyJ0eXAiOiJK...v9dmICcnRY","access_token_expires":3600,"refresh_token":"eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI...QtEiVA","refresh_token_expires":86400,"credentials":"auth=199a...09d6757f84","append_password":0,"version":"1.33.14","apiversion":"2.0"}
[root@host zm]# ls
[root@host zm]#
`

pliablepixels, 2019-11-27:
cookies are not automatic anymore. https://zoneminder.readthedocs.io/en/latest/api.html

closed 2019-11-27 by connortechnology

Cramac, 2019-11-28:
CakePHP: the rapid development php framework
Not Found
Error: The requested address '/zm/api/monitors.json&token=eyJ0eXAiOiJK.....U_RPATY8-EE' was not found on this server.

CakePHP: the rapid development php framework
CakePHP 2.10.8

Cramac, 2019-11-28:
change
zm/api/monitors.json&token
to
zm/api/monitors.json?token