ZoneMinder/zoneminder #2766: Unable to reorder cameras in the console view

novadarkstar, 2019-11-29:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit]
ZM v1.32.3
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]
RPM
- Full name and version of OS
OpenSuSE Leap 15.1
- Browser name and version (if this is an issue with the web interface)
Mozilla Firefox 68.2.0esr (64-bit)

**If the issue concerns a camera**
I don't think the cameras are at issue.

**Describe the bug**
 I am using ZM v1.32.3 running under OpenSuSE Leap 15.1 and have set up ZM to handle a dozen Amcrest cameras of various types. I am unable to persist changes I want to make in the order of the camera monitors. In the console view, I click on the up/down arrows to drag and drop the cameras into a new position but after a few seconds the camera I moved reverts back to it's old position. The ReadTheDocs does not appear to have kept abreast of this version of ZM and shows a different GUI from the one I see, which explicitly allows the user to set the camera order number. So that document is not helpful.

**To Reproduce**
Steps to reproduce the behavior:
1. Add a set of cameras in the console view.
2. Click on the Up Down arrows to drag and drop a camera into a new position/order.
3.Wait for a few seconds
4. The camera order reverts back to the original order, the new order is not persisted and there
seems to be no way to explicitly save the new order after a change is made.

**Expected behavior**
I expect the new order of camera placement to persist and the camera order numbers to change to reflect the new positions

**Debug Logs**
Nothing happens in the logs when I attempt to reorder the cameras, so I don't think the log files have anything pertinent to this problem.
```

connortechnology, 2019-11-29:
Try click and dragging.

novadarkstar, 2019-11-29:
Isaac -  That is exactly what I do, and I can move a camera from one 
position to another temporarily. But it stays in that new position for 
only a few seconds and then reverts back to it's original position. In 
other words it is not persistent and the underlying order position does 
not apparently change to reflect the new positions of all the cameras.

      Marc....

novadarkstar, 2019-11-29:
Isaac -  In the future please wait and give the reporter of a bug report 
a chance to respond before closing a bug report.

      Thanks,   Marc..

novadarkstar, 2019-11-29:
Please reopen this bug report. While I suspect that in most situations the ability to drag and drop a camera from one position to another works OK, otherwise there would already be complaints about this problem, this is not a matter of being a simple novice mistake. I am a very advanced computer scientist with over 40 years of experience with software engineering. So I probably have already tried the obvious measures and suspect there is something in my environment that is preventing this feature from working properly, for me. Whatever it is, it is a bug that needs to be addressed and will require the help from engineers with a lot of insight into the software and how is actually works.

That said, I have just observed an oddity which may be causing this bug (this is just an educated guess). The camera order numbers begin with 2 and increase to 12. There is no camera order number of 1. I think I remember, when first starting out setting ZM up and I had created a camera setup to play with and then I deleted it. I suspect that the software did not properly reset the camera order numbers so this may be another bug, and/or the cause of this issue of not being able to reorder the cameras.  Again I am just guessing but my guess is based on a lot of years doing software development myself so I do have an ability to guess where software problems may be occurring. I would ask that the software developers take a look see into this part of the software to see if doing a white box analysis would reveal anything.

SteveGilvarry, 2019-11-29:
I am sure you will also appreciate that any bugs will tested and addresses in the current software code base. I will check to see if we can replicate the issue in latest master. Does sound like something in your particular set steps.
But you may also want to try that as we a closing on the next release and a lot of issues from 1.32 are addressed.

connortechnology, 2019-11-30:
https://github.com/ZoneMinder/zoneminder/issues/2014
Follow to find the fix, or update.

novadarkstar, 2019-12-12:
I don't have the time right now to set up GIT, download the source code and build it with the fix and then figure out how to update the firmware with the new binaries, so will wait for the next release of firmware from Amcrest. So I think it is OK to close this bug report for now... Thanks for the help.

closed 2019-12-12 by novadarkstar