ZoneMinder/zoneminder #2770: x264 impossible with arm raspberry pi 4

AntoineFaget, 2019-12-01:
Hi, i have zoneminder on a raspberry pi 4 with zoneminder 1.32.3, raspbian for the os.
I have two jennov ip cam encoding in H264+ (not h264) or 265+
my cam are Jennov Caméra IP PoE HD 1080P IP66 MSC313E-V02, frame rate are 10 - 15 for slave stream resolution is 640 x 480 (I did not manage to get master stream with 1080p only slave stream in rtsp).
I am trying to use x264 encode to convert h264+ to h264 but doesn't work with zoneminder
I followed a tuto (http://jollejolles.com/installing-ffmpeg-with-h264-support-on-raspberry-pi/)  to make ffmpeg able to do the job. it work fine in the terminal but the option x264 encode is still grayed in zoneminder.
The browser name is chromium.
when i use camera passthrough the soft catch event but files are corrupt (only one jpeg frame ok)
On my regular pc with amd processor it work fine, i can choose x264 encode and replay video.
If someone know how i can do...

connortechnology, 2019-12-01:
You have to use our builds, not the raspbian ones.  
go to zmrepo.zoneminder.com

AntoineFaget, 2019-12-03:
thank you for your quick answer, I will try probably this weekend.

AntoineFaget, 2019-12-08:
Hi, i have tried, but i have a raspberry pi  4 fedora linux support raspberry pi 3 but not 4. I can't install zoneminder with arm support...

closed 2019-12-20 by connortechnology

zhangjc, 2021-02-22:
I spend a few days and find out the way. Need rebuild ffmpeg with h264_mmal enable. The change is here. https://github.com/zhangjc/zoneminder-rpi-ffmpeg