ZoneMinder/zoneminder #2777: Benefits from archived events

marcschh, 2019-12-12:
With the latest stable and even with the 1.33.15 I cannot find the benefit of archived events.
There is no option in the filter mask to include/exclude them.
It would be good to have an option to only select archived events or to exclude them from the filter

They are not protected from accidental deleting.
For example: if you archived events of the past x-days and then you select the weekly or monthly preselection and delete them all. the archived events are also deleted.
From my point of view they should not be shown up in any of the "hourly/daily/weekly/monthly" filter which you can select per camera or overall from the CONSOLE.
archived events should only came up in the "archived filter" which is already there

tommycw1, 2019-12-12:
They are protected from autodelete when disk full filter runs.

connortechnology, 2019-12-15:
Agreed about not being able to delete them. Don't agree with the rest.  

Fixed in 4d9d4ba9573c084661f8895d26711b0964457cb9

closed 2019-12-15 by connortechnology