ZoneMinder/zoneminder #2778: New Thumbnail Overview Page

marcschh, 2019-12-12:
The Event List have the option to show thumbnails of the event.
But even if I increase the thumbnail size to (for example 80px) they are still to get an idea of what may happened there and I always need view manually all events.

What about a new event overview page which shows an much bigger thumbnail and does not show to much information of the events, so that there is no LIST VIEW anymore, but an matrix (like most photo organiozer apps have) of the thumbnail with really small text (time/date should be enough) below each image.

tommycw1, 2019-12-12:
I have my thumbnails set to 128px and seems good to view at FHD res.  If
you store as video, mouseover even plays video.  Running latest stable
(1.32)

connortechnology, 2019-12-14:
I use 200.  What needs to be done is alter the layout to make better use of the space and be more responsive.

closed 2019-12-20 by connortechnology