ZoneMinder/zoneminder #2781: online play h265

Cramac, 2019-12-17:
Hello.
how to watch online video in h265 ?

```
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from '88221.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2mp41
    title           : Zoneminder Security Recording
    encoder         : Lavf57.83.100
  Duration: 01:00:00.56, start: 0.000000, bitrate: 651 kb/s
    Stream #0:0(und): Video: hevc (Main) (hev1 / 0x31766568), yuvj420p(pc), 1920x1080, 650 kb/s, 10.83 fps, 20.42 tbr, 90k tbn, 90k tbc (default)
    Metadata:
      handler_name    : VideoHandler
At least one output file must be specified
```
[image]

connortechnology, 2019-12-17:
Most browsers can't play it. In zm switch codec to mjpeg. You can also set this in the monitor to be default

Cramac, 2019-12-17:
if you put mjpeg in the codec, then there is nothing at all
[image]
[image]

joelsdc, 2019-12-19:
Per your screenshot, you have selected *MP4*. You have to select *MJPEG*. 

You can see the codec selector on the top right corner of the GUI.

Cramac, 2019-12-19:
...
> if you put mjpeg in the codec, then there is nothing at all
> [image]

connortechnology, 2019-12-19:
You may actually have to consult your logs. We are not psychic.

Cramac, 2019-12-19:
[image]

connortechnology, 2019-12-20:
You need to update.  I recently fixed a bug in zms that caused the mjpeg streaming to fail.

Cramac, 2019-12-20:
updated, mjpeg shows, mp4 never seems

joelsdc, 2019-12-29:
AFAIK the current way to stream H264 is MP4 and H265 is MJPEG

closed 2020-03-05 by stale[bot]