ZoneMinder/zoneminder #2782: Cannot build/install the Docker version of ZoneMinder from source

pree-t1, 2019-12-19:
Environment details
- ZoneMinder development version 1.33.16, just cloned tip of the master
- Followed the instruction from documentation: https://zoneminder.readthedocs.io/en/stable/installationguide/packpack.html
- OS Ubuntu 18.04 (bionic)

When I try to run the below command as per instructions:

`OS=ubuntu DIST=bionic utils/packpack/startpackpack.sh
`

It starts off just fine, but then gets stuck at a step where it asks for the geographic location. I have attached the screenshot.
[image]

I tried entering 2, 12, America, US. However, it just hangs there and even Ctrl+C doesn't help. I just need to the close the shell. 

Can someone point out what is going on here? I just want to be able to build the docker image from the zoneminder source and run it.

connortechnology, 2019-12-19:
There is just too much to configure in the default docker image. You can use my image which has all the zm dependencies already installed and configured.  This is what we are using for package builds.

add DOCKER_REPO=iconzm/packpack

pree-t1, 2019-12-19:
I made that change in <zm root dir>/packpack/packpack file and the build passed after that. It created a build folder at <zm root dir>/build with all the deb/tar files. However, it doesn't seem to build a docker image. I don't find that in the documentation either. How do build the final docker image that I use to run zoneminder?

connortechnology, 2019-12-19:
packpack is not for building docker images.  It is for building packages using docker to provide a clean build environment.

I'm not sure what the point is of using source built packages in a docker.... just use the existing docker images... 

That being said... once you have docker running... I think you can "log" into it and build from source like normal, using the do_debian_package.sh script.

closed 2019-12-20 by pree-t1