ZoneMinder/zoneminder #2791: frontend sends bad ptz command

bhearsum, 2020-01-02:
**Describe Your Environment**
- Version of ZoneMinder: 1.32.3
- How you installed ZoneMinder: default ubuntu eoan package
- Ubuntu Eoan
- Firefox Beta

**If the issue concerns a camera**
- Amcrest IP2M-841

**Describe the bug**
The web frontend sends POST requests to `/zm/index.php` with bodies such as:
```
view=request
request=control
id=6
control=Up
yge=32 
```

Which results in errors like `Invalid control parameter: Up` and `Can't access Up member of object of class ZoneMinder::Control::Amcrest_HTTP`. The former comes from https://github.com/ZoneMinder/zoneminder/blob/98f7fb63b907bf83dab8a9f8fc596800dc2d4d8a/web/includes/control_functions.php#L11, which makes me think there's something mismatched between what the backend wants, and what the frontend sends.

Digging through the HTML, I see that the onClick handler with `onclick="controlCmd('Up',event,0,-1)"`.

If I'm reading the code correctly, this ultimately comes from https://github.com/ZoneMinder/zoneminder/blob/d4435368bc420564d395cb4980c3ad0d3f1c5e6e/web/includes/Control.php#L212, and I'm guessing that `moveRoot` ends up as an empty string.

connortechnology, 2020-01-02:
You should have Can Move Continuous checked in your control settings.  Monitor->Control->Edit -> Move, check Can Move Continuous

bhearsum, 2020-01-03:
That did it. Thanks!

closed 2020-01-03 by bhearsum