ZoneMinder/zoneminder #2792: CMakeLists.txt install paths problem

danielfullmer, 2020-01-03:
In `nixpkgs` from the NixOS distribution, the default build environment for cmake projects sets both environment variables `CMAKE_INSTALL_PREFIX` and `CMAKE_INSTALL_LIBEXECDIR` to an absolute path.

This line in `CMakeLists.txt` sets `ZM_CGIDIR` like so:
https://github.com/ZoneMinder/zoneminder/blob/89844f72f49b07c90597413fb4722486441fad26/CMakeLists.txt#L143

Consequently, `ZM_CGIDIR` has two absolute paths concatenated--which isn't correct.

See: https://cmake.org/cmake/help/v3.16/module/GNUInstallDirs.html
An alternative would be to use `CMAKE_INSTALL_FULL_<dir>` instead of concatenating `CMAKE_INSTALL_PREFIX` and `CMAKE_INSTALL_<dir>`.

Downstream issue: https://github.com/NixOS/nixpkgs/pull/76855

veprbl, 2020-01-03:
One comment:
The `CMAKE_INSTALL_LIBEXECDIR` is not supposed to be a full path, but we (downstream) override it and this approach might be a bug on our side. However, switching to `CMAKE_INSTALL_FULL_LIBEXECDIR` seems like a reasonable change for zoneminder.

danielfullmer, 2020-01-03:
@veprbl The wording "If the value is not already an absolute path, ...." for `CMAKE_INSTALL_FULL_<dir>` at https://cmake.org/cmake/help/v3.16/module/GNUInstallDirs.html would suggest that it could be an absolute path--or am I misunderstanding?

veprbl, 2020-01-03:
@danielfullmer Yes, it seems like the documentation implies that non-`_FULL` variables can be full paths. Then the only "correct" option should be to use the `CMAKE_INSTALL_FULL_` variables or use a `GNUInstallDirs_get_absolute_install_dir()` function [1].

[1] https://cmake.org/cmake/help/v3.16/module/GNUInstallDirs.html#macros

jtojnar, 2020-01-03:
Alternately, using a [function for joining paths](https://stackoverflow.com/questions/57382992/joining-paths-in-cmake/57383772#57383772) would be even better. Unfortunately, CMake does not support such essential thing natively https://gitlab.kitware.com/cmake/cmake/issues/19568.

jtojnar, 2020-03-03:
Unstale.

PR #2867 (2020-03-04): cmake: use CMAKE_INSTALL_FULL_
CMAKE_INSTALL_FULL_ is to be used as a definition of the full path.

https://cmake.org/cmake/help/latest/module/GNUInstallDirs.html

Fixes: #2792
closed 2020-03-04 by connortechnology