ZoneMinder/zoneminder #2794: Errors in logs when motion detection is enabled

pree-t1, 2020-01-03:
Environment details

ZoneMinder development version 1.33.16, just cloned tip of the master
Followed the instruction from documentation: https://zoneminder.readthedocs.io/en/1.32.3/
OS Ubuntu 18.04 (bionic)

Everything seems to be fine, however, when I select "Modect" function, I see some errors in the logs (screenshot below).
[image]

I looked at a perl script: [zmdc.pl.in](https://github.com/ZoneMinder/zoneminder/blob/master/scripts/zmdc.pl.in/#L684)

It seems to be the state when the alarm gets triggered, however, how can the analysis time be 0? It looks like an error to me since the script is repeatedly exiting with return code 14.

Can someone explain this behavior?

connortechnology, 2020-01-03:
Analyse can be 0 if zma hasn't analysed any frames yet. The question is: why not?  Is this just a race condition? Do we need to check to see how long zma has been running before looking for an analysis time?  possibly.

 These logs lines are a little suspicious because they should zma being told to restart because zmwatch logs that it is going to restart zma. I wonder if there is a sorting problem in the log display.

pree-t1, 2020-01-03:
I see. I also tried running the below command and then looked at logs:

`sudo zma -m 1`

[image]

I would expect the log bahavior should be same from both command line or the web interface. Isn't it?

pree-t1, 2020-01-07:
It was probably an issue with the setup. I no longer see it. Therefore, I am closing it for now.

closed 2020-01-07 by pree-t1