ZoneMinder/zoneminder #2795: zm_create.sql: ERROR 1064 (42000) at line 741

new-gen23, 2020-01-03:
**Describe Your Environment**
- Version of ZoneMinder [1.32.3-2build1 on raspberry pi armhf]
- How you installed ZoneMinder [apt-get install from regular ubuntu repo]
- Ubuntu 19.10

**Describe the bug**
I followed the installation notes from here: https://zoneminder.readthedocs.io/en/stable/installationguide/ubuntu.html#easy-way-ubuntu-16-04
At the point where I setup the mysql database, with 

>  mysql -uroot  < /usr/share/zoneminder/db/zm_create.sql
I get the error:

> ERROR 1064 (42000) at line 741: You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near '('admin'),'',1,'View','Edit','Edit','Edit','Edit','Edit','Edit','','')' at line 1

I checked the Zoneminder Git and found that the file has been updated multiple times. I don't know if the 

**Expected behavior**
The setup script should initialize the mysql database correctly.

new-gen23, 2020-01-03:
Here is the file provided by the ubuntu repository:
```
-- MySQL dump 10.13 Distrib 5.6.13, for Linux (i686)
--
-- Host: localhost Database: zm
-- ------------------------------------------------------
-- Server version 5.6.13

/*!40101 SET @OLD_CHARACTER_SET_CLIENT=@@CHARACTER_SET_CLIENT */;
/*!40101 SET @OLD_CHARACTER_SET_RESULTS=@@CHARACTER_SET_RESULTS */;
/*!40101 SET @OLD_COLLATION_CONNECTION=@@COLLATION_CONNECTION */;
/*!40101 SET NAMES utf8 */;
/*!40014 SET @OLD_UNIQUE_CHECKS=@@UNIQUE_CHECKS, UNIQUE_CHECKS=0 */;
/*!40014 SET @OLD_FOREIGN_KEY_CHECKS=@@FOREIGN_KEY_CHECKS, FOREIGN_KEY_CHECKS=0 */;
/*!40101 SET @OLD_SQL_MODE=@@SQL_MODE, SQL_MODE='NO_AUTO_VALUE_ON_ZERO' */;
/*!40111 SET @OLD_SQL_NOTES=@@SQL_NOTES, SQL_NOTES=0 */;

--
-- Current Database: `zm`
--

CREATE DATABASE /*!32312 IF NOT EXISTS*/ `zm`;

USE `zm`;

--
-- Table structure for table `Config`
--

DROP TABLE IF EXISTS `Config`;
CREATE TABLE `Config` (
  `Id` smallint(5) unsigned NOT NULL default '0',
  `Name` varchar(32) NOT NULL default '',
  `Value` text NOT NULL,
  `Type` tinytext NOT NULL,
  `DefaultValue` text,
  `Hint` tinytext,
  `Pattern` tinytext,
  `Format` tinytext,
  `Prompt` tinytext,
  `Help` text,
  `Category` varchar(32) NOT NULL default '',
  `Readonly` tinyint(3) unsigned NOT NULL default '0',
  `Requires` text,
  PRIMARY KEY (`Name`)
) ENGINE=InnoDB;

--
-- Table structure for table `ControlPresets`
--

DROP TABLE IF EXISTS `ControlPresets`;
CREATE TABLE `ControlPresets` (
  `MonitorId` int(10) unsigned NOT NULL default '0',
  `Preset` int(10) unsigned NOT NULL default '0',
  `Label` varchar(64) NOT NULL default '',
  PRIMARY KEY (`MonitorId`,`Preset`)
) ENGINE=InnoDB;

--
-- Table structure for table `Controls`
--

DROP TABLE IF EXISTS `Controls`;
CREATE TABLE `Controls` (
  `Id` int(10) unsigned NOT NULL auto_increment,
  `Name` varchar(64) NOT NULL default '',
  `Type` enum('Local','Remote','Ffmpeg','Libvlc','cURL','WebSite') NOT NULL default 'Local',
  `Protocol` varchar(64) default NULL,
  `CanWake` tinyint(3) unsigned NOT NULL default '0',
[... 2450 lines cut ...]
      particular bit rate result in individual frames being at a
      lower quality.
      ', Category = 'highband', Readonly = '0', Requires = '';
insert into Config set Id = 176, Name = 'ZM_WEB_H_VIDEO_MAXFPS', Value = '30', Type = 'integer', DefaultValue = '30', Hint = 'integer', Pattern = '(?^:^(\d+)$)', Format = ' $1 ', Prompt = 'What the maximum frame rate for streamed video should be', Help = '
      When using streamed video the main control is the bitrate which
      determines how much data can be transmitted. However a lower
      bitrate at high frame rates results in a lower quality image.
      This option allows you to limit the maximum frame rate to
      ensure that video quality is maintained. An additional
      advantage is that encoding video at high frame rates is a
      processor intensive task when for the most part a very high
      frame rate offers little perceptible improvement over one that
      has a more manageable resource requirement. Note, this option
      is implemented as a cap beyond which binary reduction takes
      place. So if you have a device capturing at 15fps and set this
      option to 10fps then the video is not produced at 10fps, but
      rather at 7.5fps (15 divided by 2) as the final frame rate must
      be the original divided by a power of 2.
      ', Category = 'highband', Readonly = '0', Requires = '';
insert into Config set Id = 177, Name = 'ZM_WEB_H_SCALE_THUMBS', Value = '0', Type = 'boolean', DefaultValue = 'no', Hint = 'yes|no', Pattern = '(?^i:^([yn]))', Format = ' ($1 =~ /^y/) ? \'yes\' : \'no\' ', Prompt = 'Scale thumbnails in events, bandwidth versus cpu in rescaling', Help = '
      If unset, this option sends the whole image to the browser
      which resizes it in the window. If set the image is scaled down
      on the server before sending a reduced size image to the
      browser to conserve bandwidth at the cost of cpu on the server.
      Note that ZM can only

connortechnology, 2020-01-07:
Unfortunately 1.32.3 is not compatible with the mysql that comes with 1910.  Your best bet would be to upgrade to our latest master version.  

sudo add-apt-repository ppa:iconnor/master

closed 2020-01-07 by connortechnology

blackPantherOS, 2020-02-25:
Same error with MariaDB 10.4.12
Tested with latest git as well...
Which the compatible SQL version?