ZoneMinder/zoneminder #2796: Undefined error

alabamatoy, 2020-01-04:
**Describe Your Environment**
- Version of ZoneMinder 1.32.3
- How you installed ZoneMinder Iconnor's PPA
- Full name and version of OS Ubuntu 18.04 server, Bionic 
- Browser name and version (if this is an issue with the web interface) - various, doesnt seem to matter

**If the issue concerns a camera**
Believe this is N/A

**Describe the bug**
Hundreds of these in log:

ZM log as displayed in console:
2020-01-04 08:59:21 | web_js |   | 29424 | ERR | undefined - () | ?view=event

Syslog:
Jan  4 08:59:21 servername web_php[29420]: ERR [undefined - ()]

Apache2 access log:
[Sat Jan 04 08:59:21.826377 2020] [php7:notice] [pid 29420] [client 192.168.1.81:52859] ERR [undefined - ()], referer: https://192.168.1.126/zm/index.php?view=event&eid=2480&filter[Query][terms][0][attr]=MonitorId&filter[Query][terms][0][op]==&filter[Query][terms][0][val]=5&page=1&popup=1

Explanation: 192.168.1.81 is a Windows 10 box using MS Edge, and 192.168.1.126 is the ZM server web interface.

**To Reproduce**
Steps to reproduce the behavior:
Unknown.  Seems to occur randomly.

**Expected behavior**
No errors.

**Debug Logs**
`01/04/20 08:59:21.767904 web_php[29420].DBG [View: request Request: status Action: ] at /usr/share/zoneminder/www/index.php line 209
01/04/20 08:59:21.824988 web_php[29420].DBG [View: request Request: log Action: ] at /usr/share/zoneminder/www/index.php line 209
01/04/20 08:59:21.825086 web_js[29420].ERR [undefined - ()] at ?view=event line 
01/04/20 08:59:21.888720 web_php[29420].DBG [View: view_video Request:  Action: ] at /usr/share/zoneminder/www/index.php line 209
01/04/20 08:59:21.889680 web_php[29420].DBG [Path: /zmdata/zoneminder/events/5/2020-01-04/2480/2480-video.mp4] at /usr/share/zoneminder/www/views/view_video.php line 43
01/04/20 08:59:21.889760 web_php[29420].DBG [Using Range bytes=0-] at /usr/share/zoneminder/www/views/view_video.php line 71
01/04/20 08:59:21.889791 web_php[29420].DBG [Using Range 0 1970170 size: 1970171, length: 1970171] at /usr/share/zoneminder/www/views/view_video.php line 78
01/04/20 08:59:48.227260 web_php[29420].DBG [View: request Request: log Action: ] at /usr/share/zoneminder/www/index.php line 209
01/04/20 08:59:50.404200 web_php[29420].DBG [View: request Request: log Action: ] at /usr/share/zoneminder/www/index.php line 209
01/04/20 08:59:54.524620 web_php[29420].DBG [View: request Request: log Action: ] at /usr/share/zoneminder/www/index.php line 209`

connortechnology, 2020-02-19:
I think this was fixed a long time ago.   Please confirm that this doesn't happen in 1.34 series and close if true.

alabamatoy, 2020-02-19:
These errors seem to have disappeared with upgrade to 1.34.x

alabamatoy, 2020-02-20:
I closed this on git yesterday, and it suddenly reappeared.

> Feb 19 18:21:14 garlakbch web_php[26993]: ERR [undefined]
> Feb 19 18:21:19 garlakbch web_php[26991]: ERR [undefined]
> Feb 19 18:21:21 garlakbch web_php[7369]: ERR [undefined]
> Feb 19 18:21:23 garlakbch web_php[26973]: ERR [undefined]
> Feb 19 18:21:25 garlakbch web_php[27525]: ERR [undefined]

...

Review of the Apache error log is enlightening:

```
[Wed Feb 19 18:21:25.981699 2020] [php7:notice] [pid 27525] [client 192.168.1.139:54531] ERR [undefined], referer: https://192.168.1.126/zm/index.php?view=event&eid=1709&filter[Query][terms][0][attr]=MonitorId&filter[Query][terms][0][op]==&filter[Query][terms][0][val]=3&sort_field=StartTime&sort_asc=1&popup=1
[Wed Feb 19 18:21:23.635128 2020] [php7:notice] [pid 26973] [client 192.168.1.139:54527] ERR [undefined], referer: https://192.168.1.126/zm/index.php?view=event&eid=1708&filter[Query][terms][0][attr]=MonitorId&filter[Query][terms][0][op]==&filter[Query][terms][0][val]=3&sort_field=StartTime&sort_asc=1&popup=1
[Wed Feb 19 18:21:21.392839 2020] [php7:notice] [pid 7369] [client 192.168.1.139:54528] ERR [undefined], referer: https://192.168.1.126/zm/index.php?view=event&eid=1707&filter[Query][terms][0][attr]=MonitorId&filter[Query][terms][0][op]==&filter[Query][terms][0][val]=3&sort_field=StartTime&sort_asc=1&popup=1
[Wed Feb 19 18:21:19.333715 2020] [php7:notice] [pid 26991] [client 192.168.1.139:54523] ERR [undefined], referer: https://192.168.1.126/zm/index.php?view=event&eid=1706&filter[Query][terms][0][attr]=MonitorId&filter[Query][terms][0][op]==&filter[Query][terms][0][val]=3&sort_field=StartTime&sort_asc=1&popup=1
[Wed Feb 19 18:21:14.568942 2020] [php7:notice] [pid 26993] [client 192.168.1.139:54521] ERR [undefined], referer: https://192.168.1.126/zm/index.php?view=event&eid=1705&filter[Query][terms][0][attr]=MonitorId&filter[Query][terms][0][op]==&filter[Query][terms][0][val]=3&page=1&popup=1
```

The Apache access log offers no further insight.  I dont know how to ID the actual web client.

Event IDs 1705 through 1709....hmmmm....web_js.log shows
```
02/19/20 18:21:14.567764 web_js[26993].ERR [192.168.1.139] [undefined] at ?view=event line 
02/19/20 18:21:19.331508 web_js[26991].ERR [192.168.1.139] [undefined] at ?view=event line 
02/19/20 18:21:21.391269 web_js[7369].ERR [192.168.1.139] [undefined] at ?view=event line 
02/19/20 18:21:23.633646 web_js[26973].ERR [192.168.1.139] [undefined] at ?view=event line 
02/19/20 18:21:25.980568 web_js[27525].ERR [192.168.1.139] [undefined] at ?view=event line
```
Client javascript error?  I dont know what else to do to troubleshoot.

connortechnology, 2020-04-23:
I think I've got a handle on this.  Will likely be fixed in 1.34.10