ZoneMinder/zoneminder #2797: Monitors.json API returns multiple copies of a monitor if that monitor belongs to multiple groups

neillbell, 2020-01-05:
**Describe Your Environment**
- Version of ZoneMinder: 1.33.16 (1.33.16~20191229135312-bionic)
- How you installed ZoneMinder: ppa:iconnor/zoneminder-master
- Full name and version of OS: Ubuntu 18.04.3 LTS

**Describe the bug**
When a monitor is included in multiple groups, the API returns multiple entries for that monitor in the monitors.json request.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a ZoneMinder system with a single monitor
2. Create two groups, each containing your single monitor
3. Execute https://example.org/zm/api/monitors.json
4. Observe the data for your single monitor occur twice in the output

**Expected behavior**
The monitors.json API request should return one entry for each monitor without regard to the groups to which that monitor belongs.

**Debug Logs**

With a single monitor and no groups:
https://example.org/zm/api/monitors.json
```
{"monitors":[{"Monitor":{"Id":"1","Name":"Monitor1","Notes":null,"ServerId":"0","StorageId":"0","Type":"Local","Function":"None","Enabled":"0","LinkedMonitors":"","Triggers":"","Device":"\/dev\/video0","Channel":"0","Format":"45056","V4LMultiBuffer":null,"V4LCapturesPerFrame":"1","Protocol":null,"Method":"v4l2","Host":null,"Port":"","SubPath":"","Path":null,"Options":null,"User":null,"Pass":null,"Width":"1920","Height":"1080","Colours":"4","Palette":"0","Orientation":"ROTATE_0","Deinterlacing":"0","DecoderHWAccelName":null,"DecoderHWAccelDevice":null,"SaveJPEGs":"3","VideoWriter":"0","OutputCodec":null,"OutputContainer":null,"EncoderParameters":"# Lines beginning with # are a comment \r\n# For changing quality, use the crf option\r\n# 1 is best, 51 is worst quality\r\n#crf=23","RecordAudio":"0","RTSPDescribe":false,"Brightness":"-1","Contrast":"-1","Hue":"-1","Colour":"-1","EventPrefix":"Event-","LabelFormat":"%N - %d\/%m\/%y %H:%M:%S","LabelX":"0","LabelY":"0","LabelSize":"1","ImageBufferCount":"100","WarmupCount":"0","PreEventCount":"0","PostEventCount":"0","StreamReplayBuffer":"0","AlarmFrameCount":"1","SectionLength":"600","MinSectionLength":"10","FrameSkip":"0","MotionFrameSkip":"0","AnalysisFPSLimit":null,"AnalysisUpdateDelay":"0","MaxFPS":null,"AlarmMaxFPS":null,"FPSReportInterval":"100","RefBlendPerc":"6","AlarmRefBlendPerc":"6","Controllable":"0","ControlId":null,"ControlDevice":null,"ControlAddress":null,"AutoStopTimeout":null,"TrackMotion":"0","TrackDelay":null,"ReturnLocation":"-1","ReturnDelay":null,"DefaultRate":"100","DefaultScale":"100","DefaultCodec":"auto","SignalCheckPoints":"0","SignalCheckColour":"#0000BE","WebColour":"red","Exif":false,"Sequence":"1","TotalEvents":"0","TotalEventDiskSpace":"0","HourEvents":"0","HourEventDiskSpace":"0","DayEvents":"0","DayEventDiskSpace":"0","WeekEvents":"0","WeekEventDiskSpace":"0","MonthEvents":"0","MonthEventDiskSpace":"0","ArchivedEvents":null,"ArchivedEventDiskSpace":null,"ZoneCount":"1","Refresh":null},"Monitor_Status":{"MonitorId":null,"Status":null,"CaptureFPS":null,"AnalysisFPS":null,"CaptureBandwidth":null}}]}
```

With a single monitor and a single group containing that monitor:
https://example.org/zm/api/monitors.json
```
{"monitors":[{"Monitor":{"Id":"1","Name":"Monitor1","Notes":null,"ServerId":"0","StorageId":"0","Type":"Local","Function":"None","Enabled":"0","LinkedMonitors":"","Triggers":"","Device":"\/dev\/video0","Channel":"0","Format":"45056","V4LMultiBuffer":null,"V4LCapturesPerFrame":"1","Protocol":null,"Method":"v4l2","Host":null,"Port":"","SubPath":"","Path":null,"Options":null,"User":null,"Pass":null,"Width":"1920","Height":"1080","Colours":"4","Palette":"0","Orientation":"ROTATE_0","Deinterlacing":"0","DecoderHWAccelName":null,"DecoderHWAccelDevice":null,"SaveJPEGs":"3","VideoWriter":"0","OutputCodec":null,"OutputContainer":null,"EncoderParameters":"# Lines beginning with # are a comment \r\n# For changing quality, use the crf option\r\n# 1 is best, 51 is worst quality\r\n#crf=23","RecordAudio":"0","RTSPDescribe":false,"Brightness":"-1","Contrast":"-1","Hue":"-1","Colour":"-1","EventPrefix":"Event-","LabelFormat":"%N - %d\/%m\/%y %H:%M:%S","LabelX":"0","LabelY":"0","LabelSize":"1","ImageBufferCount":"100","WarmupCount":"0","PreEventCount":"0","PostEventCount":"0","StreamReplayBuffer":"0","AlarmFrameCount":"1","SectionLength":"600","MinSectionLength":"10","FrameSkip":"0","MotionFrameSkip":"0","AnalysisFPSLimit":null,"AnalysisUpdateDelay":"0","MaxFPS":null,"AlarmMaxFPS":null,"FPSReportInterval":"100","RefBlendPerc":"6","AlarmRefBlendPerc":"6","Controllable":"0","ControlId":null,"ControlDevice":null,"ControlAddress":null,"AutoStopTimeout":null,"TrackMotion":"0","TrackDelay":null,"ReturnLocation":"-1","ReturnDelay":null,"DefaultRate":"100","DefaultScale":"100","DefaultCodec":"auto","SignalCheckPoints":"0","SignalCheckColour":"#0000BE","WebColour":"red","Exif":false,"Sequence":"1","TotalEvents":"0","TotalEventDiskSpace":"0","HourEvents":"0","HourEventDiskSpace":"0","DayEvents":"0","DayEventDiskSpace":"0","WeekEvents":"0","WeekEventDiskSpace":"0","MonthEvents":"0","MonthEventDiskSpace":"0","ArchivedEvents":null,"ArchivedEventDiskSpace":null,"ZoneCount":"1","Refresh":null},"Monitor_Status":{"MonitorId":null,"Status":null,"CaptureFPS":null,"AnalysisFPS":null,"CaptureBandwidth":null}}]}
```

With a single monitor and two groups, each containing that single monitor:
https://example.org/zm/api/monitors.json
```
{"monitors":[{"Monitor":{"Id":"1","Name":"Monitor1","Notes":null,"ServerId":"0","StorageId":"0","Type":"Local","Function":"None","Enabled":"0","LinkedMonitors":"","Triggers":"","Device":"\/dev\/video0","Channel":"0","Format":"45056","V4LMultiBuffer":null,"V4LCapturesPerFrame":"1","Protocol":null,"Method":"v4l2","Host":null,"Port":"","SubPath":"","Path":null,"Options":null,"User":null,"Pass":null,"Width":"1920","Height":"1080","Colours":"4","Palette":"0","Orientation":"ROTATE_0","Deinterlacing":"0","DecoderHWAccelName":null,"DecoderHWAccelDevice":null,"SaveJPEGs":"3","VideoWriter":"0","OutputCodec":null,"OutputContainer":null,"EncoderParameters":"# Lines beginning with # are a comment \r\n# For changing quality, use the crf option\r\n# 1 is best, 51 is worst quality\r\n#crf=23","RecordAudio":"0","RTSPDescribe":false,"Brightness":"-1","Contrast":"-1","Hue":"-1","Colour":"-1","EventPrefix":"Event-","LabelFormat":"%N - %d\/%m\/%y %H:%M:%S","LabelX":"0","LabelY":"0","LabelSize":"1","ImageBufferCount":"100","WarmupCount":"0","PreEventCount":"0","PostEventCount":"0","StreamReplayBuffer":"0","AlarmFrameCount":"1","SectionLength":"600","MinSectionLength":"10","FrameSkip":"0","MotionFrameSkip":"0","AnalysisFPSLimit":null,"AnalysisUpdateDelay":"0","MaxFPS":null,"AlarmMaxFPS":null,"FPSReportInterval":"100","RefBlendPerc":"6","AlarmRefBlendPerc":"6","Controllable":"0","ControlId":null,"ControlDevice":null,"ControlAddress":null,"AutoStopTimeout":null,"TrackMotion":"0","TrackDelay":null,"ReturnLocation":"-1","ReturnDelay":null,"DefaultRate":"100","DefaultScale":"100","DefaultCodec":"auto","SignalCheckPoints":"0","SignalCheckColour":"#0000BE","WebColour":"red","Exif":false,"Sequence":"1","TotalEvents":"0","TotalEventDiskSpace":"0","HourEvents":"0","HourEventDiskSpace":"0","DayEvents":"0","DayEventDiskSpace":"0","WeekEvents":"0","WeekEventDiskSpace":"0","MonthEvents":"0","MonthEventDiskSpace":"0","ArchivedEvents":null,"ArchivedEventDiskSpace":null,"ZoneCount":"1","Refresh":null},"Monitor_Status":{"MonitorId":null,"Status":null,"CaptureFPS":null,"AnalysisFPS":null,"CaptureBandwidth":null}},
{"Monitor":{"Id":"1","Name":"Monitor1","Notes":null,"ServerId":"0","StorageId":"0","Type":"Local","Function":"None","Enabled":"0","LinkedMonitors":"","Triggers":"","Device":"\/dev\/video0","Channel":"0","Format":"45056","V4LMultiBuffer":null,"V4LCapturesPerFrame":"1","Protocol":null,"Method":"v4l2","Host":null,"Port":"","SubPath":"","Path":null,"Options":null,"User":null,"Pass":null,"Width":"1920","Height":"1080","Colours":"4","Palette":"0","Orientation":"ROTATE_0","Deinterlacing":"0","DecoderHWAccelName":null,"DecoderHWAccelDevice":null,"SaveJPEGs":"3","VideoWriter":"0","OutputCodec":null,"OutputContainer":null,"EncoderParameters":"# Lines beginning with # are a comment \r\n# For changing quality, use the crf option\r\n# 1 is best, 51 is worst quality\r\n#crf=23","RecordAudio":"0","RTSPDescribe":false,"Brightness":"-1","Contrast":"-1","Hue":"-1","Colour":"-1","EventPrefix":"Event-","LabelFormat":"%N - %d\/%m\/%y %H:%M:%S","LabelX":"0","LabelY":"0","LabelSize":"1","ImageBufferCount":"100","WarmupCount":"0","PreEventCoun [... cut ...] oints":"0","SignalCheckColour":"#0000BE","WebColour":"red","Exif":false,"Sequence":"1","TotalEvents":"0","TotalEventDiskSpace":"0","HourEvents":"0","HourEventDiskSpace":"0","DayEvents":"0","DayEventDiskSpace":"0","WeekEvents":"0","WeekEventDiskSpace":"0","MonthEvents":"0","MonthEventDiskSpace":"0","ArchivedEvents":null,"ArchivedEventDiskSpace":null,"ZoneCount":"1","Refresh":null},"Monitor_Status":{"MonitorId":null,"Status":null,"CaptureFPS":null,"AnalysisFPS":null,"CaptureBandwidth":null}}]}
```

connortechnology, 2020-01-06:
The problem is that that is how SQL joins work.  In order to support filtering by GroupId we have to join the Groups_Monitors table.  So you get two entries, with each GroupId in the GroupId column.  

In order to fix this we would either have to disable filtering by GroupId or add a lot of code to merge the Group information.

neillbell, 2020-01-06:
I opened this because of an [issue](https://github.com/pliablepixels/zmNinja/issues/881) reported that zmNinja was displaying multiple copies of monitors. @pliablepixels suggested that it was an API problem.

I guess I should re-open the zmNinja issue and reference this issue.

Shall I go ahead and close this one?

connortechnology, 2020-01-06:
Yeah don't close it.  It's a serious problem, I'm leaning towards deprecating filtering Monitors by GroupId.  I think I need to add objects for the Group_Monitors association so that we can achieve the same result by loading those first.

pliablepixels, 2020-01-10:
In the mean time, I've added code to remove duplicate monitors from the API in [that issue](https://github.com/pliablepixels/zmNinja/issues/881)

commit f5ce39ae (2020-01-15): Merge pull request #2809 from gonzalezcalleja/master

Avoid duplicate monitors in API
Fixes #2797
closed 2020-01-15 by connortechnology