ZoneMinder/zoneminder #2799: Unable to connect to ZM db.could not find driver on debian buster rpi

joshuacox, 2020-01-07:
Unable to connect to ZM db.could not find driver

**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit]

rpi3 with dietPi on it (which is debian buster armhf) `Linux champ 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux`
```
apt-cache madison zoneminder
zoneminder |   1.32.3-2 | http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages
```

<insert debug logs here, please make sure they are within the ``` quotes so they are formatted properly>

```
● zoneminder.service - ZoneMinder CCTV recording and surveillance system
   Loaded: loaded (/lib/systemd/system/zoneminder.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2020-01-07 01:37:02 GMT; 2s ago
     Docs: http://zoneminder.readthedocs.org/en/latest/
  Process: 1360 ExecStart=/usr/bin/zmpkg.pl start (code=exited, status=255/EXCEPTION)

Jan 07 01:37:02 champ zmpkg.pl[1360]: DBI connect('database=;host=','',...) failed: Access denied for user 'www-data'@'localhost' (using password: 
NO) at /usr/share/perl5/ZoneMinder/Config.pm line 117.
Jan 07 01:37:02 champ zmpkg.pl[1360]: Can't connect to db at /usr/share/perl5/ZoneMinder/Config.pm line 166.
Jan 07 01:37:02 champ zmpkg.pl[1360]: BEGIN failed--compilation aborted at /usr/share/perl5/ZoneMinder/Config.pm line 166.
Jan 07 01:37:02 champ zmpkg.pl[1360]: Compilation failed in require at /usr/share/perl5/ZoneMinder.pm line 33.
Jan 07 01:37:02 champ zmpkg.pl[1360]: BEGIN failed--compilation aborted at /usr/share/perl5/ZoneMinder.pm line 33.
Jan 07 01:37:02 champ zmpkg.pl[1360]: Compilation failed in require at /usr/bin/zmpkg.pl line 34.
Jan 07 01:37:02 champ zmpkg.pl[1360]: BEGIN failed--compilation aborted at /usr/bin/zmpkg.pl line 34.
Jan 07 01:37:02 champ systemd[1]: zoneminder.service: Control process exited, code=exited, status=255/EXCEPTION
Jan 07 01:37:02 champ systemd[1]: zoneminder.service: Failed with result 'exit-code'.
Jan 07 01:37:02 champ systemd[1]: Failed to start ZoneMinder CCTV recording and surveillance system.
```
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]
- Full name and version of OS
- Browser name and version (if this is an issue with the web interface)

**If the issue concerns a camera**
- Make and Model
- frame rate
- resolution
- ZoneMinder Source Type:

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Debug Logs**
```

I went ahead and created the database myself and added a test user, but still:

/etc/zm/conf.d/test.conf
```
ZM_DB_HOST=localhost
ZM_DB_NAME=zm
ZM_DB_USER=test
ZM_DB_PASS=testpass
```

which is where /etc/zm/zm.conf indicates you should place such things but I still get the above issues.

info.php:
```

PHP Version 7.3.11-1~deb10u1
--

```

connortechnology, 2020-01-07:
You mention an error given by php, but post logs from the perl side of things.  The logs show that it is not picking up your test user settings.  If there is another file in /etc/zm/conf.d it may be overriding the values.

joshuacox, 2020-01-07:
My zm.conf is the only file in there aside from the README, should I erase the README?

```
root@champ:~# ls -alh /etc/zm/conf.d/
total 16K
drwxr-xr-x 2 root root     4.0K Jan  7 14:50 .
drwxr-xr-x 3 root root     4.0K Jan  7 01:46 ..
-rw-r--r-- 1 root root      740 Dec  8  2018 README
-rw-r----- 1 root www-data   90 Jan  7 01:32 zm.conf
```

joshuacox, 2020-01-07:
I did try removing the readme, and funny enough things changed in terms of systemctl messages:

```
root@champ:~# systemctl status zoneminder
● zoneminder.service - ZoneMinder CCTV recording and surveillance system
   Loaded: loaded (/lib/systemd/system/zoneminder.service; disabled; vendor preset: enabled)
   Active: inactive (dead)
     Docs: http://zoneminder.readthedocs.org/en/latest/
root@champ:~# systemctl restart zoneminder
Job for zoneminder.service failed because the control process exited with error code.
See "systemctl status zoneminder.service" and "journalctl -xe" for details.
root@champ:~# systemctl status zoneminder
● zoneminder.service - ZoneMinder CCTV recording and surveillance system
   Loaded: loaded (/lib/systemd/system/zoneminder.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2020-01-07 19:01:31 GMT; 1s ago
     Docs: http://zoneminder.readthedocs.org/en/latest/
  Process: 1255 ExecStart=/usr/bin/zmpkg.pl start (code=exited, status=11)

Jan 07 19:01:31 champ zmpkg.pl[1255]: DBD::mysql::st execute failed: Table 'zm.Config' doesn't exist at /usr/share/perl5/ZoneMinder/Config.pm line 
120.
Jan 07 19:01:31 champ zmpkg.pl[1255]: Can't execute: Table 'zm.Config' doesn't exist at /usr/share/perl5/ZoneMinder/Config.pm line 166.
Jan 07 19:01:31 champ zmpkg.pl[1255]: BEGIN failed--compilation aborted at /usr/share/perl5/ZoneMinder/Config.pm line 166.
Jan 07 19:01:31 champ zmpkg.pl[1255]: Compilation failed in require at /usr/share/perl5/ZoneMinder.pm line 33.
Jan 07 19:01:31 champ zmpkg.pl[1255]: BEGIN failed--compilation aborted at /usr/share/perl5/ZoneMinder.pm line 33.
Jan 07 19:01:31 champ zmpkg.pl[1255]: Compilation failed in require at /usr/bin/zmpkg.pl line 34.
Jan 07 19:01:31 champ zmpkg.pl[1255]: BEGIN failed--compilation aborted at /usr/bin/zmpkg.pl line 34.
Jan 07 19:01:31 champ systemd[1]: zoneminder.service: Control process exited, code=exited, status=11/n/a
Jan 07 19:01:31 champ systemd[1]: zoneminder.service: Failed with result 'exit-code'.
Jan 07 19:01:31 champ systemd[1]: Failed to start ZoneMinder CCTV recording and surveillance system.
```

full message from journalctl:
```
journalctl -u zoneminder
-- Logs begin at Tue 2020-01-07 18:59:17 GMT, end at Tue 2020-01-07 19:01:31 GMT. --
Jan 07 19:01:31 champ systemd[1]: Starting ZoneMinder CCTV recording and surveillance system...
Jan 07 19:01:31 champ zmpkg.pl[1255]: WARNING: ZoneMinder configuration file found but is not readable. Check file permissions on /etc/zm/zm.conf
Jan 07 19:01:31 champ zmpkg.pl[1255]: DBD::mysql::st execute failed: Table 'zm.Config' doesn't exist at /usr/share/perl5/ZoneMinder/Config.pm line 
120.
Jan 07 19:01:31 champ zmpkg.pl[1255]: Can't execute: Table 'zm.Config' doesn't exist at /usr/share/perl5/ZoneMinder/Config.pm line 166.
Jan 07 19:01:31 champ zmpkg.pl[1255]: BEGIN failed--compilation aborted at /usr/share/perl5/ZoneMinder/Config.pm line 166.
Jan 07 19:01:31 champ zmpkg.pl[1255]: Compilation failed in require at /usr/share/perl5/ZoneMinder.pm line 33.
Jan 07 19:01:31 champ zmpkg.pl[1255]: BEGIN failed--compilation aborted at /usr/share/perl5/ZoneMinder.pm line 33.
Jan 07 19:01:31 champ zmpkg.pl[1255]: Compilation failed in require at /usr/bin/zmpkg.pl line 34.
Jan 07 19:01:31 champ zmpkg.pl[1255]: BEGIN failed--compilation aborted at /usr/bin/zmpkg.pl line 34.
Jan 07 19:01:31 champ systemd[1]: zoneminder.service: Control process exited, code=exited, status=11/n/a
Jan 07 19:01:31 champ systemd[1]: zoneminder.service: Failed with result 'exit-code'.
Jan 07 19:01:31 champ systemd[1]: Failed to start ZoneMinder CCTV recording and surveillance system.

```

closed 2020-03-14 by stale[bot]