ZoneMinder/zoneminder #2800: zmtelemetry.pl abnormal exit code 9

DKnight110, 2020-01-07:
**Describe Your Environment**
- Version of ZoneMinder: ZoneMinder, v1.33.16.
- How you installed ZoneMinder: PPA
- Full name and version of OS: Ubuntu 18.04.3 LTS x86_64
- Perl version: v5.26.1

**Describe the bug**
The ZM log is filled with messages of "'zmtelemetry.pl' exited abnormally, exit status 9"

**To Reproduce**
Steps to reproduce the behavior:
1. Install Zoneminder from https://launchpad.net/~iconnor/+archive/ubuntu/zoneminder
2. Run manually /usr/bin/zmtelemetry.pl
3. See the following message in console:
```
zoneminder@zoneminder-virtual-machine:/var/log/zm$ sudo zmtelemetry.pl --force
ZoneMinder Telemetry Agent starting at 20/01/07 17:21:33
Collecting data to send to ZoneMinder Telemetry server.
07.01.2020 17:21:34.316546 zmtelemetry[18236].INF [main:104] [Sending data to ZoneMinder Telemetry server.]
HTTP::Message content must be bytes at /usr/bin/zmtelemetry.pl line 169.
```

**Expected behavior**
No error should be encountered

**Debug Logs**
```
2020-01-07 17:02:51.572550 zmtelemetry[18011].INF-zmtelemetry.pl [Sending data to ZoneMinder Telemetry server.]
2020-01-07 17:02:41.838790 zmdc[1930].ERR-zmdc.pl ['zmtelemetry.pl' exited abnormally, exit status 9]
2020-01-07 17:02:41.829050 zmtelemetry[18006].INF-zmtelemetry.pl [Sending data to ZoneMinder Telemetry server.]
2020-01-07 16:59:10.922090 zmtelemetry[17986].INF-zmtelemetry.pl [Sending data to ZoneMinder Telemetry server.]
2020-01-07 16:59:00.405590 zmtelemetry[17979].INF-zmtelemetry.pl [Sending data to ZoneMinder Telemetry server.]
2020-01-07 16:57:21.802220 zmdc[1930].ERR-zmdc.pl ['zmtelemetry.pl' exited abnormally, exit status 9]
2020-01-07 16:57:21.793810 zmtelemetry[17958].INF-zmtelemetry.pl [Sending data to ZoneMinder Telemetry server.]
2020-01-07 16:56:59.115320 zmtelemetry[17954].INF-zmtelemetry.pl [Sending data to ZoneMinder Telemetry server.]
2020-01-07 16:54:41.251810 zmdc[1930].ERR-zmdc.pl ['zmtelemetry.pl' exited abnormally, exit status 9]
2020-01-07 16:54:41.241480 zmtelemetry[17929].INF-zmtelemetry.pl [Sending data to ZoneMinder Telemetry server.]
2020-01-07 16:53:20.817870 zmdc[1930].ERR-zmdc.pl ['zmtelemetry.pl' exited abnormally, exit status 9]
2020-01-07 16:53:20.808230 zmtelemetry[17920].INF-zmtelemetry.pl [Sending data to ZoneMinder Telemetry server.]
2020-01-07 16:52:40.097190 zmdc[1930].ERR-zmdc.pl ['zmtelemetry.pl' exited abnormally, exit status 9]
2020-01-07 16:52:40.076940 zmtelemetry[14667].INF-zmtelemetry.pl [Sending data to ZoneMinder Telemetry server.]
2020-01-07 16:52:18.395520 zmdc[1930].ERR-zmdc.pl ['zmtelemetry.pl' exited abnormally, exit status 9]
2020-01-07 16:52:18.381290 zmtelemetry[8034].INF-zmtelemetry.pl [Sending data to ZoneMinder Telemetry server.]
2020-01-07 16:52:07.144570 zmdc[1930].ERR-zmdc.pl ['zmtelemetry.pl' exited abnormally, exit status 9]
2020-01-07 16:52:07.134420 zmtelemetry[4491].INF-zmtelemetry.pl [Sending data to ZoneMinder Telemetry server.]
```

**Possible fix**

By using the Perl Encode module and replacing line 169 with
```
$req->content(encode("UTF-8",$msg));
```

the issue is no longer seen. I will send a pull request if you think this fix makes sense.

connortechnology, 2020-01-07:
We can certainly add that line, but I was wondering if adding a use utf8; at the top after the use strict would do the same thing.

DKnight110, 2020-01-07:
> We can certainly add that line, but I was wondering if adding a use utf8; at the top after the use strict would do the same thing.
Thanks so much for having a look at this!

Unfortunately I just tried adding use utf8; instead of use Encode and no luck: I'm getting the same error:

```
ZoneMinder Telemetry Agent starting at 20/01/07 18:27:04
Collecting data to send to ZoneMinder Telemetry server.
07.01.2020 18:27:05.202625 zmtelemetry[18923].INF [main:105] [Sending data to ZoneMinder Telemetry server.]
HTTP::Message content must be bytes at /usr/bin/zmtelemetry.pl line 169.
```

I don't know if the use Encode path is the right way to go, maybe someone with more Perl experience can chime in.

connortechnology, 2020-01-07:
It probably is the way to go, given what I've read, I will commit it.

commit 971d62f6 (2020-01-07): encode the telemetry to utf-8 before sending. Fixes #2800
closed 2020-01-07 by connortechnology