ZoneMinder/zoneminder #2802: Zoneminder's local build/install from the modified source

pree-t1, 2020-01-10:
Environment details

ZoneMinder development version 1.33.16, just cloned tip of the master
Followed the instruction from documentation: https://zoneminder.readthedocs.io/en/1.32.3/ (Harder Way: Build Package From Source)
OS Ubuntu 18.04 (bionic)

I was able to build and install successfully and run Zoneminder in "Modect" mode. I used the below command from the documentation to build the latest master snapshot:

> ./do_debian_package.sh --snapshot=NOW --branch=master --type=local

It also creates a copy of source code _zoneminder_release_ that I believe one could use to rebuild some components independently. I modified some source code (e.g. zoneminder_release/src/zma.cpp) and tried to rebuild it. Below is the sequence of all steps:

`cd zoneminder_release`
`sudo cmake .`
`sudo make`
`sudo cp src/zma /usr/bin`
`sudo service zoneminder restart`

However, I am running into some issues after this. If I just use the original existing /usr/bin/zma (the one came from installation), it works fine. So I wanted to ask what is the best way to modify the local copy of source and rebuild/install independent modules or components (e.g. zma, perl scripts etc.).

connortechnology, 2020-01-10:
the zoneminder_release is really meant to be discarded.

What I do, is clone the repo:
git clone git@github.com:ZoneMinder/ZoneMinder
cd ZoneMinder
git submodule init
git submodule update --init --recursive
cmake -DCMAKE_INSTALL_PREFIX=/usr \
  -DCMAKE_SKIP_RPATH=ON \
  -DCMAKE_VERBOSE_MAKEFILE=OFF \
  -DCMAKE_COLOR_MAKEFILE=ON \
  -DZM_RUNDIR=/var/run/zm \
  -DZM_SOCKDIR=/var/run/zm \
  -DZM_TMPDIR=/var/tmp/zm \
  -DZM_LOGDIR=/var/log/zm \
  -DZM_WEBDIR=/usr/share/zoneminder/www \
  -DZM_CONTENTDIR=/var/cache/zoneminder \
  -DZM_CGIDIR=/usr/lib/zoneminder/cgi-bin \
  -DZM_CACHEDIR=/var/cache/zoneminder/cache \
  -DZM_WEB_USER=www-data \
  -DZM_WEB_GROUP=www-data \
  -DCMAKE_INSTALL_SYSCONFDIR=etc/zm \
  -DZM_CONFIG_DIR=/etc/zm \
  -DCMAKE_BUILD_TYPE=Debug
make
sudo make install

So basically I overwrite the installed package files. Then I do my work in src, web, whereever and from those dirs make and sudo make install are a little faster because they only do the files in src or web or wherever.

BTW this is not a bug or issue with ZM.  github Issues is for Bugs/Issues.  If you are going to be doing ZM deelopment please join us in our slack channel https://zoneminder-chat.herokuapp.com/ or irc (#zoneminder on freenode)

closed 2020-01-10 by connortechnology

pree-t1, 2020-01-10:
Sounds good. Also, I tried the above build method, however, the web interface is totally different and doesn't seem full functional as before. I can't see camera preview or see logs anymore.