ZoneMinder/zoneminder #2805: double free in mysql client on ubuntu build

dannysauer, 2020-01-11:
**Describe Your Environment**
- Version of ZoneMinder zoneminder
  - 1.32.3-2build1 amd64
- How you installed ZoneMinder
  - Ubuntu package (http://us.archive.ubuntu.com/ubuntu eoan/universe amd64 Packages)
- Full name and version of OS
  - Ubuntu 19.10
- Browser name and version
  - N/A

Other useful information:
MySQL is a remote system, running mysql-server-5.7 on an Ubuntu 18.04 system.
Other zoneminder components on this system work fine, such as the audit daemon and web interface.  This is part of a multi-server configuration, consisting of this Intel system running Ubuntu 19.10 and four Raspberry Pi 3 systems running Raspbian Buster.  The monitors technically work on the Pis, but performance is not particularly good. :)

**Describe the bug**
`zmc` fails to run, crashing repeatedly with double-free error.

**To Reproduce**
Set up a multi-server zoneminder with a remote MySQL on Ubuntu 18.04, and then:
```
sauer@cerberus:~$ sudo -u www-data zmc -m 2
double free or corruption (out)
Aborted
```

**Expected behavior**
`zmc` doesn't crash. :)

**Debug Logs**
There isn't anything useful in the logs other than "zmc exited".  However, I've got this from gdb:
```
sauer@cerberus:~$ sudo gdb /usr/bin/zmc
...
Reading symbols from /usr/bin/zmc...
(No debugging symbols found in /usr/bin/zmc)
(gdb) run -m 2
Starting program: /usr/bin/zmc -m 2
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
double free or corruption (out)

Program received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) backtrace
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff5798899 in __GI_abort () at abort.c:79
#2  0x00007ffff580338e in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff592c3a5 "%s\n") at ../sysdeps/posix/libc_fatal.c:181
#3  0x00007ffff580b4dc in malloc_printerr (str=str@entry=0x7ffff592e0a0 "double free or corruption (out)") at malloc.c:5332
#4  0x00007ffff580d170 in _int_free (av=0x7ffff595db80 <main_arena>, p=0x7ffff51928d0, have_lock=<optimized out>) at malloc.c:4314
#5  0x00007ffff7cc78e0 in SSL_CTX_free () from /usr/lib/x86_64-linux-gnu/libgnutls-openssl.so.27
#6  0x00007ffff768281f in ?? () from /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21
#7  0x00007ffff7682e7d in ?? () from /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21
#8  0x00007ffff7632871 in ?? () from /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21
#9  0x00007ffff7632c1e in ?? () from /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21
#10 0x00007ffff7636129 in mysql_real_connect () from /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21
#11 0x00005555555aaeab in zmDbConnect() ()
#12 0x00005555555a72d8 in zmLoadConfig() ()
#13 0x000055555559f263 in main ()
```

Looking around online, it appears that sometimes this is caused by the way mysql client is used?  It appears to be happening in the SSL negotiation based on the backtrace, but I'm not doing anything weird with the cert; it's just mysql running basically out of the box on an up-to-date  Ubuntu 10.04 LTS.  Here's the cert chain per OpenSSL, which appears to be an auto-generated self-signed certificate:

```
sauer@cerberus:~$ openssl s_client -starttls mysql -connect mysql:3306 </dev/null
CONNECTED(00000003)
Can't use SSL_get_servername
depth=1 CN = MySQL_Server_5.7.28_Auto_Generated_CA_Certificate
verify error:num=19:self signed certificate in certificate chain
verify return:1
depth=1 CN = MySQL_Server_5.7.28_Auto_Generated_CA_Certificate
verify return:1
depth=0 CN = MySQL_Server_5.7.28_Auto_Generated_Server_Certificate
verify return:1
---
Certificate chain
 0 s:CN = MySQL_Server_5.7.28_Auto_Generated_Server_Certificate
   i:CN = MySQL_Server_5.7.28_Auto_Generated_CA_Certificate
 1 s:CN = MySQL_Server_5.7.28_Auto_Generated_CA_Certificate
   i:CN = MySQL_Server_5.7.28_Auto_Generated_CA_Certificate
```

I'm guessing this is a defect in zmc, since the other components on this machine like the web interface and perl scripts are able to connect to the MySQL server ok, and I can use the command-line mysql client fine as well.  For example (I'm using an alternate db name - `zmpi` in this cluster to separate it from a legacy install) :

```
sauer@cerberus:~$ mysql -u zoneminder -p --ssl -h mysql -e 'SHOW TABLES;' zmpi
Enter password:
+-----------------+
| Tables_in_zmpi  |
+-----------------+
| Config          |
| ControlPresets  |
```

connortechnology, 2020-03-03:
The first step now would be to update to 1.34.

connortechnology, 2020-03-04:
We have seen something like this when zm is linked against both gnutls-openssl and libssl.  We are moving to get rid of gnutls-openssl in favour of wolfssl to resolve this.

dannysauer, 2020-03-04:
I've been away from home for a few days, but will definitely try the new version (and try locally compiling) in the next day or so when I'm back. :D

dannysauer, 2020-03-09:
I switched to the iconnor PPA and updated to 1.34.5-eoan1.  After way more work upgrading the database than anticipated :), it works properly out of the box.  Hooray!

Works in current supported version, so I'd call this closed.

closed 2020-03-09 by dannysauer