ZoneMinder/zoneminder #2806: Video not displaying when a monitor is selected

neillbell, 2020-01-11:
**Describe Your Environment**
- Version of ZoneMinder: 1.33.16 (1.33.16~20200110190654-bionic)
- How you installed ZoneMinder: ppa:iconnor/zoneminder-master
- Ubuntu 18.04.3 LTS

**Describe the bug**
When a monitor is clicked on in the ZoneMinder console, no video is displayed.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the ZoneMinder console in a browser
2. Click on a monitor
3. Observe the blank browser window

**Expected behavior**
Video from the monitor should be displayed.

**Debug Logs**
The apache error log shows an error. There is nothing in web_php.log or the browser console.
```
==> /var/log/apache2/error.log <==
[Sat Jan 11 12:41:07.277177 2020] [php7:error] [pid 9879] [client 192.168.1.1:55812] PHP Fatal error:  Uncaught Error: Call to a member function UrlToIndex() on null in /usr/share/zoneminder/www/includes/Monitor.php:472\nStack trace:\n#0 /usr/share/zoneminder/www/skins/classic/views/js/watch.js.php(51): ZM\\Monitor->UrlToIndex()\n#1 /usr/share/zoneminder/www/skins/classic/includes/functions.php(179): require_once('/usr/share/zone...')\n#2 /usr/share/zoneminder/www/skins/classic/views/watch.php(60): xhtmlHeaders('/usr/share/zone...', 'Front Door - Fe...')\n#3 /usr/share/zoneminder/www/index.php(273): require_once('/usr/share/zone...')\n#4 {main}\n  thrown in /usr/share/zoneminder/www/includes/Monitor.php on line 472, referer: https://example.org/zm/index.php?view=console

==> /var/log/apache2/access.log <==
192.168.1.1 - - [11/Jan/2020:12:41:07 -0600] "GET /zm/index.php?view=watch&mid=4 HTTP/1.1" 200 2345 "https://example.org/zm/index.php?view=console" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Ubuntu Chromium/79.0.3945.79 Chrome/79.0.3945.79 Safari/537.36"
192.168.1.1 - - [11/Jan/2020:12:41:08 -0600] "GET /zm/index.php?view=console HTTP/1.1" 200 5968 "https://example.org/zm/index.php?view=event&eid=570125&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Battr%5D=StartDateTime&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Bop%5D=%3E%3D&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Bval%5D=-1+hour&filter%5BQuery%5D%5Bterms%5D%5B1%5D%5Bcnj%5D=and&filter%5BQuery%5D%5Bterms%5D%5B1%5D%5Battr%5D=MonitorId&filter%5BQuery%5D%5Bterms%5D%5B1%5D%5Bop%5D=%3D&filter%5BQuery%5D%5Bterms%5D%5B1%5D%5Bval%5D=5&sort_field=StartTime&sort_asc=0&page=1" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Ubuntu Chromium/79.0.3945.79 Chrome/79.0.3945.79 Safari/537.36"
```

fennec622, 2020-01-11:
Yes same error 
1.33.16~20200111140945-xenial

neillbell, 2020-01-12:
I've restarted things and rebooted the machine. I still see the problem.

fennec622, 2020-01-12:
> I've restarted things and rebooted the machine. I still see the problem.

have you update zoneminder to last ?
20200111140945

neillbell, 2020-01-12:
I hadn't noticed that there was an update. It all works great now. I had to restart apache, as you suggested, before it worked correctly.

Thanks for reminding me to check for an update!

closed 2020-01-12 by neillbell