ZoneMinder/zoneminder #2810: 1.34.0: OpenSSL versus GPL versus jwt-cpp

onlyjob, 2020-01-18:
As you may already know, due to incompatibility between GPL and OpenSSL licenses it is illegal to link GPL-licensed software (without OpenSSL linking exception) including ZoneMinder with _libssl_.

  https://people.gnome.org/~markmc/openssl-and-the-gpl

This is why in Debian we build ZoneMinder against `libcurl4-gnutls-dev` to avoid dependency on `libssl-dev`.

Note how `src/zm_rtsp_auth.h` and `src/zm_user.cpp` have (conditional) detection of GnuTLS that makes OpenSSL optional: 

  * https://github.com/ZoneMinder/zoneminder/blob/master/src/zm_rtsp_auth.h#L25
  * https://github.com/ZoneMinder/zoneminder/blob/master/src/zm_user.cpp#L33

But ZoneMinder 1.34.0 gained direct unconditional dependency on OpenSSL in `src/zm_crypt.cpp`: https://github.com/ZoneMinder/zoneminder/blob/master/src/zm_crypt.cpp#L6

To complicate this problem further, with introduction of _jwt-cpp_ that explicitly requires `openssl/evp.h` from `libssl-dev`, ZoneMinder 1.34.0 also gained dependency on OpenSSL through _jwt-cpp_.

Even with `libssl-dev` vendored _jwt-cpp_ FTBFS in Debian "unstable" as follows:

~~~~
./dbuild/src/./src/jwt-cpp/include/jwt-cpp/jwt.h:142: undefined reference to `HMAC' 
/usr/bin/ld: ./dbuild/src/./src/jwt-cpp/include/jwt-cpp/jwt.h:142: undefined reference to `HMAC' 
~~~~

Fortunately there is a GnuTLS flavour of JWT library in Debian -- [libjwt-gnutls-dev](https://tracker.debian.org/pkg/libjwt) but ZoneMinder needs some work to use the system library instead of a vendored one.

Due to licensing conflict, please avoid using OpenSSL.

Please consider using alternatives such as [GnuTLS](https://gnutls.org/) and/or GPL-licensed [wolfSSL](https://www.wolfssl.com/products/wolfssl/).

onlyjob, 2020-02-05:
As far as I'm concerned this is the most serious outstanding bug that requires urgent attention.

onlyjob, 2020-02-19:
Any progress?? There were 3 minor releases already, none of which addressed this outstanding issue... This problem prevents upgrade of ZoneMinder 1.32.3 in Debian. Please help.

hax0kartik, 2020-02-20:
Hey, I am working on this and I have got a couple of questions to ask.

I got jwt-cpp to work with wolfssl. Therefore from now on, wolfssl will become a neccessary dependency for zoneminder to compile.

My first question is would the developers like to see me include wolfssl as a gitmodule or should I rather use cmake to check for it. ( I'm planning to go with former, since we need to compile the latest commit of wolfssl to make it work.)

Also, since I will be already be using wolfSSL should I completely remove dependency on OpenSSL and GNUTLS, since it wouldn't have any sense to use two crypto libraries simultaneously?

Any other suggestion or questions are welcome.

connortechnology, 2020-02-20:
We do not want to import wolfssl at all. That is how we got into this situation in the first place.  We need to work with whatever libraries are found on a debian system.

It would be a nice option if found on the system, but ideally we would give people choice. 

My understanding based on @onlyjob's issue is that we just need to do a bit of work to use debian's jwt, so perhaps if wolf-ssl isn't at the required version, that is the way to go.  I do want your wolf-ssl code too though, for the day when everyone has a sufficient version installed.  What is the feature needed in the latest wolfssl?

Please note that as buster has already sailed, the idea is to get zoneminder included into whatever version of debian is used for the next ubuntu. So debian testing.  So whatever is available there is likely what we have to work with.

hax0kartik, 2020-02-20:
@connortechnology Alright thanks for the reply.  
What you say is correct, we shouldn't be shipping wolfssl. I can probably mention it in INSTALL.md that the latest commit of wolfssl needs to be build for linking with wolfssl. There are a few missing openssl APIs which are not shipped with the latest stable version of wolfssl and hence we need to build it from the latest commit.

So, if I understand correctly, I will need to keep all the dependencies as it is and add extra ifdefs to deal with wolfssl, correct?

onlyjob, 2020-02-20:
Thanks for working on this, @hax0kartik. You might want to check [Debian Upstream Guide](https://wiki.debian.org/UpstreamGuide), especially

  * https://wiki.debian.org/UpstreamGuide#No_inclusion_of_third_party_code
  * https://wiki.debian.org/UpstreamGuide#Dependencies

Cmake is preferable, please try to use conditional compilation whenever possible and avoid vendoring or sub-moduling. Please only ever depend on released versions of dependency libraries and _never on latest commit_. If latest _released_ version is not suitable then we should either wait for release or seek other alternatives. Sometimes it may help to ask library developers to make a release.

Technically speaking we might still include ZoneMinder to _buster-backports_ but in order to do that it is necessary for ZoneMinder to migrate to "testing" first. This issue is a blocker (as because of it currently I can't even build ZoneMinder from source) but also https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=922724 which we might re-classify with security team to lower its severity.

connortechnology, 2020-03-05:
@onlyjob wolfssl doesn't appear to be in buster.  It is in backports, but not buster proper.  Do you have any input about that?

We have been having trouble using libgnutls-openssl-dev so it is not an option.

closed 2020-03-05 by connortechnology

onlyjob, 2020-03-06:
Yes, WolfSSL is in backports:

```
$ rmadison libwolfssl-dev
libwolfssl-dev | 4.3.0+dfsg-2~bpo10+1 | buster-backports | amd64, arm64, armel, armhf, i386, mips, mips64el, mipsel, ppc64el, s390x
libwolfssl-dev | 4.3.0+dfsg-2         | testing          | amd64, arm64, armel, armhf, i386, mips64el, mipsel, ppc64el, s390x
libwolfssl-dev | 4.3.0+dfsg-2         | unstable         | amd64, arm64, armel, armhf, i386, mips64el, mipsel, ppc64el, s390x
```

_buster-backports_ is a suite that provides new or updated packages for _buster_ after release.  
Practically speaking the library is in Buster once it introduced to backports i.e. it is good enough to have a library in backports. One just have to enable backports suite as instructed here: https://backports.debian.org/Instructions/

onlyjob, 2020-03-11:
Thanks a lot for addressing the problem. Now ZoneMinder 1.34.6 builds successfully on Debian but there is a following warning:

```
CMake Warning (dev) at CMakeLists.txt:760 (message): 
  ZoneMinder requires a working MD5 function for hashed authenication but 

          none were found - hashed authenication will not be available 
```

Is there is way to fix the warning? Thanks.

(Note the typo in 'authenication').

connortechnology, 2020-03-27:
I think this was fixed with eed45ae66c449af305c1f24a944535b1c6c8e4fd

onlyjob, 2020-03-28:
Perfect, thanks.

PR #2874 (2020-03-05): Add libjwt and remove gnutls-openssl wrapper
Fixes #2810