ZoneMinder/zoneminder #2816: Amcrest control script error 401

alabamatoy, 2020-01-26:
**Describe Your Environment**
- Version of ZoneMinder 1.34.1
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc] installed from PPA as 1.32.3, then upgraded to 1.34.1
- Full name and version of OS Ubuntu 18.04 bionic something
- Browser name and version N/A

**If the issue concerns a camera**
- Make and Model: Amcrest IP2m-853EW
- frame rate: 10fps (set in camera)
- resolution
- ZoneMinder Source Type: ffmpeg

**Describe the bug**
Control is enabled, correct control device and control address are entered, "Amcrest HTTP API" is control type.  Selecting any of the control functions results in 
Camera control command FAILED: '401 Unauthorized' for URL [login]:[password]@10.1.1.101:80/cgi-bin/ptz.cgi?action=stop&channel=0&code=ZoomTele&arg1=0&arg2=0&arg3=0&arg4=0

**To Reproduce**
Steps to reproduce the behavior:
1. open camera monitor, click on control (pan, tilt, zoom), nothing happens.
2 - Log contains noted error.

**Expected behavior**
A clear and concise description of what you expected to happen.
Camera performs expected PTZ function

**Debug Logs**
Turned on debug for zmcontrol, all I got was:
01/26/2020 14:35:13.996098 zmcontrol_3[21706].ERR [ZoneMinder::Control::Amcrest_HTTP:147] [Camera control command FAILED: '401 Unauthorized' for URL [login]:[password]@10.1.1.101:80/cgi-bin/ptz.cgi?action=stop&code=Right&channel=0&arg1=0&arg2=1&arg3=0]

<insert debug logs here, please make sure they are within the ``` quotes so they are formatted properly>

```

NoNaym, 2020-02-11:
I am still seeing the same issue in ZM v1.34.2, and running pretty much the same configuration as @alabamatoy 

Entering URLs from the ZM log into Chrome causes the camera to start moving...

connortechnology, 2020-02-11:
You do have to restart zmcontrol.pl after turning on debug.... 

One thing that is interesting... in order for the script to even open communications, it has to get a successful request to /cgi-bin/ptz.cgi... so why do subsequent requests fail?  

Looking at the code now... I see that we are assuming http://, which we probably shouldn't...

Now when I reworked the code for 1.34 I added digest auth support, which means NOT including the username and password with each request in the url.... oh crap... I see the problem.  I'm going to push some fixes to master, can you someone run master and let me know if they fix the problem?

I suspect they won't, but maybe the extra logging will.

alabamatoy, 2020-02-12:
" I'm going to push some fixes to master, can you someone run master and let me know if they fix the problem?"

I copied the latest amcrest_http file and attempted to test it, and got exactly same behavior, error 401 not auth.

connortechnology, 2020-02-12:
There should be lots of informative debugging lines.  Just saying you get a 401 isn't helpful to me.

Any chance I can get access to the camera?  teamviewer or something?

alabamatoy, 2020-02-12:
Isaac, send me an email at alabamatoy at gmail dot com and I will send you link to site and user creds.

connortechnology, 2020-02-12:
I just updated master.  After testing with @alabamatoy's camera, it seems that the first request to start movement fails with 401 Unauth, but if we just resend the command, it works.

NoNaym, 2020-02-18:
Just got this working after adding the port to the Control Address in ZM Console (and some other server cleanup/updating). If I understand correct, @connortechnology I am still seeing errors in the System Log because each command is being "resent," correct? My camera is responding to the ZM Web UI controls now, but I am seeing these errors that _look_ bad, but I assume are mostly harmless. Either way, I wonder if there is a way to clean this last bit up...

`2020-02-17 21:32:31	zmcontrol_8		23105	ERR	Camera control command FAILED: '401 Unauthorized' for URL 192.168.1.106:60/cgi-bin/ptz.cgi?action=start&channel=0&code=GotoPreset&&arg1=0&arg2=3&arg3=0&arg4=0	zmcontrol.pl	
2020-02-17 21:32:31	zmcontrol_8		23105	INF	Camera control: '200 OK' for URL userid:password@192.168.1.106:60/cgi-bin/ptz.cgi?action=start&channel=0&code=GotoPreset&&arg1=0&arg2=2&arg3=0&arg4=0	zmcontrol.pl	
`

Perhaps I might have something else going on here, as I'm not sure why ZM would be sending the command once without the credentials, and once with them...

alabamatoy, 2020-02-18:
There is other cleanup needed as well.  
1 - The script as written uses a fixed delay between start and stop explicitly set in each function ("usleep(500);", instead of the time entered in the control config ("Auto Stop Timeout" presumably).  This value needs to be set once at beginning of script, or passed in as a externally set option.
2 - I believe the PTZ function should be sent, then the sleep, then a single stop sent, instead of the start, sleep and stop within each PTZ function.  Am I wrong?
3 - The script needs to be changed to have an option somehow to send a start based on left/right/up/down/zoom/wide and a stop based on click in the center of the left/right/up/down rose versus using the timeout in item 1.  I personally would prefer to go back to the press-and-hold design (onmousedown and onmouseup or ontouch for phones) either as an option (a switch in the script or in the config?  I dont know how to add a switch to the config...maybe use "Auto Stop Timeout" set to "-1" as a switch to use press-and-hold?) or as a second version of the script (maintenance becomes a concern).  I dont know how the script could export a requirement for an additional config item in the options.

I dont like having the script throw errors, but at this point, I dont know how to fix that problem.

connortechnology, 2020-02-18:
Please test the latest version.

alabamatoy, 2020-02-18:
Copied latest version of Amcrest_HTTP.pm, overwrote existing dot pm file on ZM.  Restarted ZM.  Confirmed that the camera monitor control is correctly pointed at the right "protocol" script.  Here's what I get:

`2020-02-18 12:47:19 | zmcontrol_3 |   | 18421 | ERR | Camera  control command FAILED: '401 Unauthorized' for URL  http://[login:password]@10.1.1.101:80/cgi-bin/ptz.cgi?action=stop&code=Right&channel=0&arg1=0&arg2=1&arg3=0 | zmcontrol.pl |  
-- | -- | -- | -- | -- | -- | -- | --
2020-02-18 12:47:19 | zmcontrol_3 |   | 18421 | DBG | cgi-bin/ptz.cgi?action=stop&code=Right&channel=0&arg1=0&arg2=1&arg3=0[69] | zmcontrol.pl |  
2020-02-18 12:47:19 | zmcontrol_3 |   | 18421 | DBG | Move Right Stop | zmcontrol.pl |  
2020-02-18 12:47:19 | zmcontrol_3 |   | 18421 | ERR | Camera  control command FAILED: '401 Unauthorized' for URL  http://[login:password]@10.1.1.101:80/cgi-bin/ptz.cgi?action=start&code=Right&channel=0&arg1=0&arg2=1&arg3=0 | zmcontrol.pl |  
2020-02-18 12:47:19 | zmcontrol_3 |   | 18421 | DBG | cgi-bin/ptz.cgi?action=start&code=Right&channel=0&arg1=0&arg2=1&arg3=0[70] | zmcontrol.pl |  
2020-02-18 12:47:19 | zmcontrol_3 |   | 18421 | DBG | Move Right | zmcontrol.pl |  
2020-02-18 12:47:19 | zmcontrol_3 |   | 18421 | DBG | $VAR1 = {          'command' => 'moveConRight'        };
`
So the 401 error is back.  No camera movement is evident.

connortechnology, 2020-02-18:
How about the debug lines before sending the command?

alabamatoy, 2020-02-18:
Here's everything for zmcontrol_3
[zm-log.txt](https://github.com/ZoneMinder/zoneminder/files/4221394/zm-log.txt)

connortechnology, 2020-02-18:
try latest

alabamatoy, 2020-02-18:
`

2020-02-18 14:34:14 | zmcontrol_3 |   | 23129 | ERR | Camera  control command FAILED: '401 Unauthorized' for URL  http://admin:admin123@10.1.1.101:80/cgi-bin/ptz.cgi?action=stop&code=Right&channel=0&arg1=0&arg2=1&arg3=0 | zmcontrol.pl |  
-- | -- | -- | -- | -- | -- | -- | --
2020-02-18 14:34:14 | zmcontrol_3 |   | 23129 | DBG | cgi-bin/ptz.cgi?action=stop&code=Right&channel=0&arg1=0&arg2=1&arg3=0[69] | zmcontrol.pl |  
2020-02-18 14:34:14 | zmcontrol_3 |   | 23129 | DBG | Move Right Stop | zmcontrol.pl |  
2020-02-18 14:34:14 | zmcontrol_3 |   | 23129 | ERR | Camera  control command FAILED: '401 Unauthorized' for URL  http://admin:admin123@10.1.1.101:80/cgi-bin/ptz.cgi?action=start&code=Right&channel=0&arg1=0&arg2=1&arg3=0 | zmcontrol.pl |  
2020-02-18 14:34:14 | zmcontrol_3 |   | 23129 | DBG | cgi-bin/ptz.cgi?action=start&code=Right&channel=0&arg1=0&arg2=1&arg3=0[70] | zmcontrol.pl |  
2020-02-18 14:34:14 | zmcontrol_3 |   | 23129 | DBG | Move Right | zmcontrol.pl |  
2020-02-18 14:34:14 | zmcontrol_3 |   | 23129 | DBG | $VAR1 = {          'command' => 'moveConRight'        }; | zmcontrol.pl |  
2020-02-18 14:30:57 | zmcontrol_3 |   | 23129 | DBG | Header connection => close | zmcontrol.pl |  
2020-02-18 14:30:57 | zmcontrol_3 |   | 23129 | DBG | Header client-response-num => 1 | zmcontrol.pl |  
2020-02-18 14:30:57 | zmcontrol_3 |   | 23129 | DBG | Header client-date => Tue, 18 Feb 2020 20:30:57 GMT | zmcontrol.pl |  
2020-02-18 14:30:57 | zmcontrol_3 |   | 23129 | DBG | Header ::std_case => HASH(0x561c54c0f2e8) | zmcontrol.pl |  
2020-02-18 14:30:57 | zmcontrol_3 |   | 23129 | DBG | Header  www-authenticate => Digest realm="Login to AMC0449301F7564FAD",  qop="auth", nonce="661160577",  opaque="0a24ebda7ac1c37dd7164c3839e5d541d40897ef" | zmcontrol.pl |  
2020-02-18 14:30:57 | zmcontrol_3 |   | 23129 | DBG | Header client-peer => 10.1.1.101:80 | zmcontrol.pl |  
2020-02-18 14:30:57 | zmcontrol_3 |   | 23129 | DBG | Header content-length => 0 | zmcontrol.pl |  
2020-02-18 14:30:57 | zmcontrol_3 |   | 23129 | ERR | Authentication still failed after updating REALM401 Unauthorized | zmcontrol.pl |  
2020-02-18 14:30:57 | zmcontrol_3 |   | 23129 | DBG | Changing REALM to (Login to AMC0449301F7564FAD) | zmcontrol.pl |  
2020-02-18 14:30:57 | zmcontrol_3 |   | 23129 | DBG | Initial Header content-length => 0 | zmcontrol.pl |  
2020-02-18 14:30:57 | zmcontrol_3 |   | 23129 | DBG | Initial  Header www-authenticate => Digest realm="Login to  AMC0449301F7564FAD", qop="auth", nonce="782908133",  opaque="0a24ebda7ac1c37dd7164c3839e5d541d40897ef" | zmcontrol.pl |  
2020-02-18 14:30:57 | zmcontrol_3 |   | 23129 | DBG | Initial Header client-peer => 10.1.1.101:80 | zmcontrol.pl |  
2020-02-18 14:30:57 | zmcontrol_3 |   | 23129 | DBG | Initial Header connection => close | zmcontrol.pl |  
2020-02-18 14:30:57 | zmcontrol_3 |   | 23129 | DBG | Initial Header client-date => Tue, 18 Feb 2020 20:30:57 GMT | zmcontrol.pl |  
2020-02-18 14:30:57 | zmcontrol_3 |   | 23129 | DBG | Initial Header client-response-num => 1 | zmcontrol.pl |  
2020-02-18 14:30:57 | zmcontrol_3 |   | 23129 | DBG | Initial Header ::std_case => HASH(0x561c54bd0200) | zmcontrol.pl |  
2020-02-18 14:30:57 | zmcontrol_3 |   | 23129 | DBG | Using  initial credentials for admin:admin123@10.1.1.101:80:80, Login to  AMC0447E0A9B08D59C, admin, admin123, base_url:  http://admin:admin123@10.1.1.101:80 auth:admin:admin123@10.1.1.101:80 | zmcontrol.pl |  
2020-02-18 14:30:57 | zmcontrol_3 |   | 23129 | DBG | Address: http://admin:admin123@10.1.1.101:80:80 | zmcontrol.pl |  
2020-02-18 14:30:57 | zmcontrol_3 |   | 23129 | INF | Control server 3/Amcrest_HTTP_master starting at 20/02/18 14:30:57 | zmcontrol.pl |  
2020-02-18 14:30:57 | zmcontrol_3 |   | 23129 | INF | Starting control server 3/Amcrest_HTTP_master | zmcontrol.pl |  
2020-02-18 14:30:57 | zmcontrol_3 |   | 23129 | DBG | zmcontrol: arg string: --id 3 sock file /var/run/zm/zmcontrol-3.sock | zmcontrol.pl |  
2020-02-18 14:30:57 | zmcontrol_3 |   | 23129 | DBG | LogOpts: level=DBG/DBG, screen=OFF, database=DBG, logfile=ERR->/var/log/zm/zmcontrol_3.log, syslog=WAR

`

connortechnology, 2020-02-18:
Dang it, why is there still :80:80

try latest please.

alabamatoy, 2020-02-18:
I think you nailed it this time.

`

2020-02-18 16:03:29 | zmcontrol_3 |   | 28701 | DBG | $VAR1 = {          'command' => 'moveConRight'        }; | zmcontrol.pl |  
-- | -- | -- | -- | -- | -- | -- | --
2020-02-18 16:03:29 | zmcontrol_3 |   | 28701 | DBG | Move Right | zmcontrol.pl |  
2020-02-18 16:03:29 | zmcontrol_3 |   | 28701 | DBG | cgi-bin/ptz.cgi?action=start&code=Right&channel=0&arg1=0&arg2=1&arg3=0[70] | zmcontrol.pl |  
2020-02-18 16:03:29 | zmcontrol_3 |   | 28701 | INF | Camera  control 2: '200 OK' for URL  http://[login:password]@10.1.1.101/cgi-bin/ptz.cgi?action=start&code=Right&channel=0&arg1=0&arg2=1&arg3=0 | zmcontrol.pl |  
2020-02-18 16:03:29 | zmcontrol_3 |   | 28701 | DBG | Move Right Stop | zmcontrol.pl |  
2020-02-18 16:03:29 | zmcontrol_3 |   | 28701 | DBG | cgi-bin/ptz.cgi?action=stop&code=Right&channel=0&arg1=0&arg2=1&arg3=0[69] | zmcontrol.pl |  
2020-02-18 16:03:29 | zmcontrol_3 |   | 28701 | INF | Camera  control 2: '200 OK' for URL  http://[login:password]@10.1.1.101/cgi-bin/ptz.cgi?action=stop&code=Right&channel=0&arg1=0&arg2=1&arg3=0 | zmcontrol.pl |  
2020-02-18 16:02:57 | zmcontrol_3 |   | 28701 | INF | Camera  control 2: '200 OK' for URL  http://[login:password]@10.1.1.101/cgi-bin/ptz.cgi?action=stop&code=Right&channel=0&arg1=0&arg2=1&arg3=0 | zmcontrol.pl |  
2020-02-18 16:02:57 | zmcontrol_3 |   | 28701 | DBG | cgi-bin/ptz.cgi?action=stop&code=Right&channel=0&arg1=0&arg2=1&arg3=0[69] | zmcontrol.pl |  
2020-02-18 16:02:57 | zmcontrol_3 |   | 28701 | DBG | Move Right Stop | zmcontrol.pl |  
2020-02-18 16:02:57 | zmcontrol_3 |   | 28701 | INF | Camera  control 2: '200 OK' for URL  http://[login:password]@10.1.1.101/cgi-bin/ptz.cgi?action=start&code=Right&channel=0&arg1=0&arg2=1&arg3=0 | zmcontrol.pl |  
2020-02-18 16:02:57 | zmcontrol_3 |   | 28701 | DBG | cgi-bin/ptz.cgi?action=start&code=Right&channel=0&arg1=0&arg2=1&arg3=0[70] | zmcontrol.pl |  
2020-02-18 16:02:57 | zmcontrol_3 |   | 28701 | DBG | Move Right | zmcontrol.pl |  
2020-02-18 16:02:57 | zmcontrol_3 |   | 28701 | DBG | $VAR1 = {          'command' => 'moveConRight'        }; | zmcontrol.pl |  
`

connortechnology, 2020-02-18:
Good.  So to be clear, this is still sending each command twice.  So weird.

NoNaym, 2020-02-21:
Updated my ZM yesterday to v1.35.0, and I don't think I am seeing the duplicate commands being sent anymore. First command send is a little delayed, but all commands appear to be working for me now! Thanks for working on this @connortechnology!

alabamatoy, 2020-02-23:
I have a new version of this which eliminates the 'usleep' between start and stop and instead sends the PTZ command start, which makes the camera keep going until either the user clicks on the X in the center of the PTZ arrows or the camera hits a physical stop.  The 'usleep' approach would take a zillion clicks and half of forever to slew the camera 90 degrees.  This new approach takes 2 clicks, start, watch the movement, stop. I have made no changes to iconnor's authentication fixes which are working correctly.

I will clean up the code, add some comments, and offer the new version.  I don't know how to do this correctly using this git site, can someone educate me, or point me at a guide?

connortechnology, 2020-02-23:
So basically you fork the zm repo into your own.  

You can actually do all the editing on github, or you can clone it to your own system and do the edits there.  Knowing that you operate through webmin generally you might want to operate entirely within github.  Create a branch, commit your changes to the branch, push them to github.  Now you click Create New Pull Request, submit the request.  I will receive a notification about it and can review/merge.

closed 2020-05-02 by stale[bot]