ZoneMinder/zoneminder #2819: 1.32.3 -> 1.34.1 update: output garbled for non-default resolution

abishai, 2020-01-31:
**Describe Your Environment**
- Version of ZoneMinder 1.34.1
- How you installed ZoneMinder from-source
- FreeBSD 12.1

**If the issue concerns a camera**
- Make and Model No idea, it's behind firewall. BrickCom something
- frame rate 5 fps
- 2032 x 1536
- ZoneMinder Source Type: ffmpeg, rstp/tcp

**Describe the bug**
This is a regression, as 1.32 works. Only ZM was updated, all other dependencies are the same.
Here is output for all monitors with 2032 x 1536 resolution. https://imgur.com/0b0wNj3
No errors are printed in logs.
If I change resolution to 2048 x 1536, ZM emits warning **Monitor dimensions are 2048x1536 but camera is sending 2032x1536** but works (=output is normal).
Monitor configuration (after resolution workaround applied)

```
                    Id: 10
                  Name: cam10
                 Notes: 
              ServerId: 0
             StorageId: 0
                  Type: Ffmpeg
              Function: Mocord
               Enabled: 1
        LinkedMonitors: NULL
              Triggers: 
                Device: 
               Channel: 0
                Format: 0
        V4LMultiBuffer: NULL
   V4LCapturesPerFrame: NULL
              Protocol: NULL
                Method: rtpRtsp
                  Host: NULL
                  Port: 80
               SubPath: 
                  Path: rtsp://192.168.1.212/channel1
               Options: NULL
                  User: NULL
                  Pass: NULL
                 Width: 2048
                Height: 1536
               Colours: 3
               Palette: 0
           Orientation: ROTATE_0
         Deinterlacing: 0
    DecoderHWAccelName: NULL
  DecoderHWAccelDevice: NULL
             SaveJPEGs: 0
           VideoWriter: 2
           OutputCodec: NULL
       OutputContainer: NULL
     EncoderParameters: # Lines beginning with # are a comment 
# For changing quality, use the crf option
# 1 is best, 51 is worst quality
#crf=23
           RecordAudio: 0
          RTSPDescribe: 0
            Brightness: -1
              Contrast: -1
                   Hue: -1
                Colour: -1
           EventPrefix: Event-
           LabelFormat: %%N - %d/%m/%y %H:%M:%S
                LabelX: 0
                LabelY: 0
             LabelSize: 2
      ImageBufferCount: 40
           WarmupCount: 25
         PreEventCount: 25
        PostEventCount: 25
    StreamReplayBuffer: 1000
       AlarmFrameCount: 1
         SectionLength: 600
      MinSectionLength: 10
             FrameSkip: 0
       MotionFrameSkip: 0
      AnalysisFPSLimit: NULL
   AnalysisUpdateDelay: 0
                MaxFPS: 30.000
           AlarmMaxFPS: 30.00
     FPSReportInterval: 1000
          RefBlendPerc: 12
     AlarmRefBlendPerc: 6
          Controllable: 0
             ControlId: NULL
         ControlDevice: NULL
        ControlAddress: NULL
       AutoStopTimeout: NULL
           TrackMotion: 0
            TrackDelay: NULL
        ReturnLocation: -1
           ReturnDelay: NULL
           DefaultRate: 100
          DefaultScale: 50
          DefaultCodec: auto
     SignalCheckPoints: 0
     SignalCheckColour: #0000BE
             WebColour: red
                  Exif: 0
              Sequence: 10
           TotalEvents: 1139
            HourEvents: 8
             DayEvents: 145
            WeekEvents: 1000
           MonthEvents: 1139
        ArchivedEvents: 0
ArchivedEventDiskSpace: NULL
             ZoneCount: 0
               Refresh: NULL
   MonthEventDiskSpace: 620921300772
    WeekEventDiskSpace: 530179850109
     DayEventDiskSpace: 73569213512
    HourEventDiskSpace: 3126272041
   TotalEventDiskSpace: 620921300772
```

connortechnology, 2020-02-13:
Can you try switch to 32bit instead of 24?  I have a suspicion about where the problem lies

abishai, 2020-02-16:
32bit colour mode fixes the issue as well!

commit c035376f (2020-02-23): Merge pull request #2848 from connortechnology/fix_2819_garbled_output_weird_resolution_24bit

Only align buffers to 32bit if using a 32bit pix format.  Fixes #2819
commit d2872bec (2020-02-23): Only align buffers to 32bit if using a 32bit pix format.  Fixes #2819
closed 2020-02-23 by connortechnology