ZoneMinder/zoneminder #2824: Version 1.34.2 Release shows Version 1.34.1, triggering

rossbcan, 2020-02-08:
Need to update zm_create.sql or zm_update-1.34.2.sql for correct version

connortechnology, 2020-02-08:
Um, release 1.34.2 has the version file correct, and a zm-update.1.34.2.sql.  zm_create.sql doesn't reference any version #s.   SO not sure what the issue is here.

rossbcan, 2020-02-08:
Well, zm_create.sql  has ZM_DYN_CURR_VERSION at 1.34.1 (should be 1.34.2?) and with a fresh git bulld from 1.34.2, the update zm window keep popping up til disabled. In addition, the GUI displays V1.34.1

I would expect, at latest release, no update window to popup and correct version to display in GUI 

I can work around this, up to you if issue and fix or not.

Regards;
Bill

connortechnology, 2020-02-10:
zm_create.sql is a generated file.  You havn't updated it.  it is generated from zm_create.sql.in

rossbcan, 2020-02-10:
have since updated my build to replace zm_create.sql (after config) with new zm_create.sql with some dir changes. My issue, not yours, sorry for trouble...

closed 2020-02-10 by rossbcan