ZoneMinder/zoneminder #2825: Onvif parsing lacks resilience

remi-crypto, 2020-02-08:
**Describe Your Environment**
- Version of ZoneMinder v1.32.3
- How you installed ZoneMinder deb https://zmrepo.zoneminder.com/debian/release stretch/
- Full name and version of OS: Debian GNU/Linux 9 (stretch)
- Browser name and version: Not a browser issue

**If the issue concerns a camera**
- Make and Model: Wanscam K22 1080P WiFi IP Camera Wireless CCTV 2MP Outdoor Waterproof Onvif Security Camera Support 64G TF Card
- frame rate: 12
- resolution: 1920x1080 and 640x360
- ZoneMinder Source Type: unknown

**Describe the bug**
During profiling: /usr/bin/zmonvif-probe.pl  profiles http://<ip>/onvif/device_service 
the results are:
Nothing returned ffrom get_PTZ
Nothing returned ffrom get_Media
Nothing returned ffrom get_Imaging
Nothing returned ffrom get_Events
Nothing returned ffrom get_Device
MainStreamTooken, MainStream, H264, 1920, 1080, 12, 10/media/wsdl"/
SubStreamTooken, SubStream, H264, 640, 360, 15, 10/media/wsdl"/
Unknown profile AudioStreamTooken AudioStream

*Note* the '10/media/wsdl' - pretty wrong and crapping on the camera's part. However the correct values do exist ( my fix ):
Nothing returned ffrom get_PTZ
Nothing returned ffrom get_Media
Nothing returned ffrom get_Imaging
Nothing returned ffrom get_Events
Nothing returned ffrom get_Device
MainStreamTooken, MainStream, H264, 1920, 1080, 12, rtsp://192.168.100.26:554/ch0_0.h264
SubStreamTooken, SubStream, H264, 640, 360, 15, rtsp://192.168.100.26:554/ch0_1.h264
AudioStreamTooken, AudioStream, , , , , rtsp://192.168.100.26:554/ch0_3.h264

**To Reproduce**
Steps to reproduce the behavior:
  Attached is the GetProfilesResponse from the camera, just parse it with ZoneMinder::ONVIF::profiles
Also attached is my fix for ```sub profiles```, attached as ONVIF.pm.txt

**Expected behavior**
The expect behaviour is to scan for appropriate media url and not pick the first likely match, perhaps one that starts with rtsp:://?

**Debug Logs**
``` Nothing returned ffrom get_PTZ
Nothing returned ffrom get_Media
Nothing returned ffrom get_Imaging
Nothing returned ffrom get_Events
Nothing returned ffrom get_Device
MainStreamTooken, MainStream, H264, 1920, 1080, 12, 10/media/wsdl"/
SubStreamTooken, SubStream, H264, 640, 360, 15, 10/media/wsdl"/
Unknown profile AudioStreamTooken AudioStream
 ```

[ONVIF.pm.txt](https://github.com/ZoneMinder/zoneminder/files/4174670/ONVIF.pm.txt)

[CameraWanscamK22Profile.txt](https://github.com/ZoneMinder/zoneminder/files/4174671/CameraWanscamK22Profile.txt)

*Summary*, Thanks for your work on this project. My contribution is just to make the profile parser 
work for every camera even if they give good results. It wasn't until I started debugging ZoneMinder's profile code did I discover the questionable soap results and made changes to the profile parser to make it more resilient. Not 100% perfect but it fixed my issues and hopefully NOT break existing implementations.

Thx.

connortechnology, 2020-02-09:
I will look through your ONVIF.pm, however as I have already made many changes to it since 1.32.3, this may take some time.  In future, please work with the latest development version and talk to us, as we may save you some time.

Thanks for your efforts though!

closed 2020-02-09 by connortechnology

connortechnology, 2020-02-10:
In sub profiles, you seem to return empty values if there is not VideoEncoderConfiguration.  My fix was to not return anything for the profile at all if there wasn't a vec. I'm not sure which is the best approach... I'm kinda thinking yours, as it would then return profiles that aren't video....

I see that you added multicast streamtype support... I just wonder if the RTP-multicast shouldn't be RTP_multicast (underscore instead of hyphen).  I suppose I will have to look at the specification. Having done so... 

Another note, where you test for rtsp:// I wonder if we should do it case-insensitive.

Anyways, I have merged your ideas and gone a little farther.  Please update to latest master and let me know if it works for your cameras.

remi-crypto, 2020-02-10:
First, I'll yield to your experience Zoneminder. As I don't have a library of cameras to test against nor the experience in onvif. Just a lot of experience in integrating with third party systems that don't always follow the rules. :)

My goal in practice is to get all practical options and discard the garbage. And I have dealt with my share of garbage data before.

As for returning the audio channel (no video) was an option that I thought was interesting as you could record audio without a video channel, like for an intercom system. Sort of a 'future' proofing thing. Not sure if there is an rtsp device for just audio or even if Zoneminder is equipped to handle such a thing.

As for the multicast, I followed the ONVIF documentation and your style in case there was a case-sensitive issue. In hind sight, that might have been premature, as I wasn't sure if zoneminder supported multicast or not.

also the rtsp protocol could be case insensitive check but as a result it should lowercase the protocol  if that were the case.

I'll be sure to update to the latest version as soon as possible.

Thx.

commit 3b30d0a8 (2020-02-09): code restructure, return all valid profiles instead  of just the first one. Also add multicase stream types.  This incorporates ideas from patch from remi-crypto.  Fixes #2825