ZoneMinder/zoneminder #2827: Add button doesn't work

cowanmax, 2020-02-09:
Hi guys!
Version of ZoneMinder 1.43.2, Ubuntu 18.04 LTS. I can't add cams. 
I see just _add_circle_ Add button and there is no reaction? What I need to do??

connortechnology, 2020-02-09:
You havn't adjusted your apache config.  This must be an upgrade (you didn't say).  You might just have to restart apache, or you will have to modify it's config to match the one that would have come with zoneminder.

Here is an example of mine, pay attention to line 17.
.https://gist.github.com/connortechnology/9ca83f812d94b2777b93ad54ee45067e
.

cowanmax, 2020-02-09:
Thanks for your reply but I use nginx with mariadb. The same?

cowanmax, 2020-02-10:
I have deleted nginx, php, mysql and install LAMP. So now is works :) Many thanks!

closed 2020-03-03 by connortechnology