ZoneMinder/zoneminder #2836: Ffmpeg vaapi no effect 1.34.2

multijohn, 2020-02-16:
**Describe Your Environment**
Ubuntu 16
- Version of ZoneMinder [release version, development version, or commit]
1.34.2
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]
ppa 
- Full name and version of OS
Ubuntu server 16 ffmpeg 4
- Browser name and version (if this is an issue with the web interface)

**If the issue concerns a camera**
- Make and Model
- frame rate
- resolution
- ZoneMinder Source Type:
9 Escam rtsp cameras 25 fps 720p
**Describe the bug**
A clear and concise description of what the bug is.
With remote method i'm happy with the performance , but i'm reading that the remote method is not the correct way and i must use ffmpeg, so i tryed it!
But the performance is getting worse hits 50 percent cpu! Ffmpeg supports vaapi on my system i7 processor!
Vaapi settings inside monitors no have any effect
**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.
Reduce cpu and load to remote method levels or lower
**Debug Logs**
```

<insert debug logs here, please make sure they are within the ``` quotes so they are formatted properly>

```

danlsgiga, 2020-03-24:
I believe I have the same problem. Configured ZM to use `vaapi` but don't see any improvement in performance. I'm running `1.35.1` via Unraid docker `dlandon/zoneminder.master`

madhacker3kxl, 2020-05-09:
I have similar issue with v1.34.11 in Debian 10 Buster. It doesn't actually do any GPU decoding, used intel_gpu_top to verify. Tried manually in ffmpeg directly in cmdline and it decodes fine.

connortechnology, 2020-05-12:
You have to turn on debugging and check the logs.  It will likely tell you why.

I've noticed that on ubuntu2004, the /dev/dri/renderD129 ar enow owned by group render instead of video.  So you need to put www-data into the render group as well.

multijohn, 2020-05-15:
Here is the debugging, 
is it looking for h264_mmal ?
I have set it for VAAPI
Thank u!

`05/16/20 01:39:04.478440 zmc_m1[13298].DB1-zm_logger.cpp/255 [LogOpts: level=DB1 effective=DB1, screen=OFF, database=DB9, logfile=DB1->/var/log/zm/zm_debug.log.13298, syslog=OFF]
05/16/20 01:39:04.492579 zmc_m1[13298].DB1-zm_utils.cpp/282 [Detected a x86\x86-64 processor with AVX2]
05/16/20 01:39:04.499132 zmc_m1[13298].DB1-zm_monitor.cpp/2116 [Got 1 for v4l_captures_per_frame]
05/16/20 01:39:04.506311 zmc_m1[13298].INF-zm_ffmpeg.cpp/82 [Enabling ffmpeg logs, as LOG_DEBUG+LOG_FFMPEG are enabled in options]
05/16/20 01:39:04.512508 zmc_m1[13298].DB1-zm_monitor.cpp/423 [monitor purpose=1]
05/16/20 01:39:04.519420 zmc_m1[13298].DB1-zm_monitor.cpp/437 [mem.size(8) SharedData=600 TriggerData=560 VideoStoreData=4128 timestamps=800 images=50x3686400 = 184320000 total=184326152]
05/16/20 01:39:04.525860 zmc_m1[13298].DB1-zm_storage.cpp/84 [Loaded Storage area 1 'ZoneminderStorage']
05/16/20 01:39:04.532743 zmc_m1[13298].DB1-zm_monitor.cpp/441 [Storage path: /media/zoneminder/events]
05/16/20 01:39:04.543522 zmc_m1[13298].DB1-zm_monitor.cpp/595 [Unable to map file /dev/shm/zm.mmap.1 (184326152 bytes) to locked memory, trying unlocked]
05/16/20 01:39:04.550324 zmc_m1[13298].DB1-zm_monitor.cpp/598 [Mapped file /dev/shm/zm.mmap.1 (184326152 bytes) to unlocked memory]
05/16/20 01:39:04.639986 zmc_m1[13298].DB1-zm_monitor.cpp/514 [Monitor Garden-Front has function 3,
label format = '%N - %d/%m/%y %H:%M:%S', label X = 0, label Y = 0, label size = 1,
image buffer count = 50, warmup count = 25, pre-event count = 25, post-event count = 100, alarm frame count = 1,
fps report interval = 1000, ref blend percentage = 12, alarm ref blend percentage = 12, track motion = 0]
05/16/20 01:39:04.646926 zmc_m1[13298].DB1-zm_zone.cpp/884 [Got 2 zones for monitor Garden-Front]
05/16/20 01:39:04.656554 zmc_m1[13298].DB1-zm_monitor.cpp/2389 [Loaded monitor 1(Garden-Front), 2 zones]
05/16/20 01:39:04.662572 zmc_m1[13298].INF-zmc.cpp/223 [Starting Capture version 1.34.14]
05/16/20 01:39:04.669549 zmc_m1[13298].INF-zm_ffmpeg_camera.cpp/202 [Priming capture from rtsp://192.168.2.80:554/user=user_password=pass_channel=1_stream=0.sdp]
05/16/20 01:39:04.675512 zmc_m1[13298].DB1-zm_ffmpeg_camera.cpp/365 [Calling avformat_open_input for rtsp://192.168.2.80:554/user=user_password=pass_channel=1_stream=0.sdp]
05/16/20 01:39:04.929087 zmc_m1[13298].WAR-zm_ffmpeg.cpp/70 [Invalid UE golomb code]
05/16/20 01:39:04.936653 zmc_m1[13298].WAR-zm_ffmpeg.cpp/70 [Invalid UE golomb code]
05/16/20 01:39:06.127771 zmc_m1[13298].DB1-zm_ffmpeg_camera.cpp/458 [Failed to find decoder (h264_mmal)]
05/16/20 01:39:06.136468 zmc_m1[13298].DB1-zm_ffmpeg.cpp/335 [Dumping stream index i(0) index(0)]
05/16/20 01:39:06.142932 zmc_m1[13298].DB1-zm_ffmpeg.cpp/345 [    Stream #0:0]
05/16/20 01:39:06.149718 zmc_m1[13298].DB1-zm_ffmpeg.cpp/356 [frames:28, frame_size:0 stream timebase: 1/90000]
05/16/20 01:39:06.156110 zmc_m1[13298].DB1-zm_ffmpeg.cpp/363 [codec: Video: h264 (Main), 1 reference frame, yuv420p(left), 1280x720, 1/44]
05/16/20 01:39:06.161988 zmc_m1[13298].DB1-zm_ffmpeg.cpp/286 [22 fps]
05/16/20 01:39:06.168101 zmc_m1[13298].DB1-zm_ffmpeg.cpp/288 [90k stream tb numerator]
05/16/20 01:39:06.173966 zmc_m1[13298].WAR-zm_ffmpeg_camera.cpp/540 [HWAccel support not compiled in.]
05/16/20 01:39:06.186028 zmc_m1[13298].DB1-zm_ffmpeg.cpp/330 [Dumping codec_context codec_type(0) codec_id(28 h264) width(1280) height(720)  timebase(1/44) format(unsupported on avconv) gop_size 12 max_b_frames 0 me_cmp 0 me_range 0 qmin 2 qmax 31]
05/16/20 01:39:06.191675 zmc_m1[13298].DB1-zm_ffmpeg_camera.cpp/561 [HWACCEL not in use]
05/16/20 01:39:06.199253 zmc_m1[13303].WAR-zm_ffmpeg.cpp/70 [Invalid UE golomb code]
05/16/20 01:39:06.206378 zmc_m1[13304].WAR-zm_ffmpeg.cpp/70 [Invalid UE golomb code]
05/16/20 01:39:06.213348 zmc_m1[13305].WAR-zm_ffmpeg.cpp/70 [Invalid UE golomb code]
05/16/20 01:39:06.220041 zmc_m1[13306].WAR-zm_ffmpeg.cpp/70 [Invalid UE golomb code]
05/16/20 01:39:06.227254 zmc_m1[13307].WAR-zm_ffmpeg.cpp/70 [Invalid UE golomb code]
05/16/20 01:39:06.234207 zmc_m1[13308].WAR-zm_ffmpeg.cpp/70 [Invalid UE golomb code]
05/16/20 01:39:06.241197 zmc_m1[13309].WAR-zm_ffmpeg.cpp/70 [Invalid UE golomb code]
05/16/20 01:39:06.248200 zmc_m1[13310].WAR-zm_ffmpeg.cpp/70 [Invalid UE golomb code]
05/16/20 01:39:06.255378 zmc_m1[13298].DB1-zm_ffmpeg_camera.cpp/276 [raw frame from decoder: format 0 unsupported 1280x720 linesize:1280x640 pts: -9223372036854775808]
05/16/20 01:39:06.263013 zmc_m1[13311].WAR-zm_ffmpeg.cpp/70 [Invalid UE golomb code]
05/16/20 01:39:06.276046 zmc_m1[13298].DB1-zm_ffmpeg_camera.cpp/1122 [Setup conversion context for 1280x720 yuv420p to 1280x720 rgba]
05/16/20 01:39:06.294097 zmc_m1[13298].INF-zm_monitor.cpp/2544 [Garden-Front: images:1 - Capturing at 0.50 fps, capturing bandwidth 34521bytes/sec]
05/16/20 01:39:06.301485 zmc_m1[13298].DB1-zm_ffmpeg_camera.cpp/276 [raw frame from decoder: format 0 unsupported 1280x720 linesize:1280x640 pts: -9223372036854775808]
05/16/20 01:39:06.318910 zmc_m1[13298].DB1-zm_ffmpeg_camera.cpp/276 [raw frame from decoder: format 0 unsupported 1280x720 linesize:1280x640 pts: -9223372036854775808]
05/16/20 01:39:06.328588 zmc_m1[13298].DB1-zm_ffmpeg_camera.cpp/276 [raw frame from decoder: format 0 unsupported 1280x720 linesize:1280x640 pts: -9223372036854775808]
[... cut ...]
05/16/20 01:39:06.381374 zmc_m1[13298].DB1-zm_ffmpeg_camera.cpp/276 [raw frame from decoder: format 0 unsupported 1280x720 linesize:1280x640 pts: -9223372036854775808]
05/16/20 01:39:06.388290 zmc_m1[13298].DB1-zm_ffmpeg_camera.cpp/276 [raw frame from decoder: format 0 unsupported 1280x720 linesize:1280x640 pts: -9223372036854775808]
05/16/20 01:39:06.395359 zmc_m1[13298].DB1-zm_ffmpeg_camera.cpp/276 [raw frame from decoder: format 0 unsupported 1280x720 linesize:1280x640 pts: -9223372036854775808]`

Gransi, 2020-11-06:
I have the same behavior. There is no difference between vaapi and cpu only.

I'm using ubuntu 20.04 and zm 1.34.21.

[zm_debug.log](https://github.com/ZoneMinder/zoneminder/files/5503424/zm_debug.log)

kabadisha, 2020-12-30:
I have been looking into this and found that dlandon's docker container uses Ubuntu 18.04 which does not have very up to date Intel graphics drivers and VAAPI support.

I have found a workaround, which you can find here:
https://github.com/ZoneMinder/zoneminder/issues/3063#issuecomment-752787126

closed 2025-12-27 by SteveGilvarry