ZoneMinder/zoneminder #2845: Image freeze

observer1, 2020-02-20:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit]
1.34.1-1
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]
RPMFusion
- Full name and version of OS
CentOS Linux release 8.1.1911 (Core)
- Browser name and version (if this is an issue with the web interface)
Firefox 72.0.1
**If the issue concerns a camera**
All are diferent cameras the common of all is all are wifi cameras using http or rtsp for video
- ZoneMinder Source Type:
All are ffmpeg
**Describe the bug**
I setup three cameras and after some hours the image freezes until I restart the service, this doesn't happen at the same moment for all the cameras but eventually all freeze

**To Reproduce**
Steps to reproduce the behavior:
This happens after some time after start the service, no need to do anything

**Expected behavior**
The cameras should keep running

**Debug Logs**
```
There are a lot  of log files in /var/log/zoneminder which should I include?

<insert debug logs here, please make sure they are within the ``` quotes so they are formatted properly>

```

connortechnology, 2020-02-24:
typically zmc_m*.log are what we might want.  

Are the cameras still recording events or is it just the live view that stops?

observer1, 2020-02-25:
It doesn't record any event

connortechnology, 2020-03-03:
Nothing further is going to happen here unless you post some logs.

observer1, 2020-03-04:
It's been working fine for about a week
I think you want to know what I did
I noticed the cameras freeze at night mostly and I saw the database was in a mariadb host that has a lot of work during nights making everything on it to time out, I changed to another host and it fixed.
I think zm should be able to continue on this circumstances
Thanks for your help

closed 2020-05-10 by stale[bot]