ZoneMinder/zoneminder #2847: Failed to get HW surface format for cuda

Ilia-SB, 2020-02-21:
**THIS FORUM IS FOR BUG REPORTS ONLY**

Do not post feature or enhancement requests, general discussions, or support questions here.

Feature and enhancement requests, general discussions, and support questions should occur in one of the following areas:

- The [ZoneMinder-Chat Slack channel](https://zoneminder-chat.herokuapp.com/)
- The [ZoneMinder Forum](https://forums.zoneminder.com/)

Docker related issues should be posted here: https://github.com/ZoneMinder/zmdockerfiles

In order to submit a bug report, please populate the fields below this line. This is required.

----------------------------------------------------------------------------------------------------

**Describe Your Environment**
- Version of ZoneMinder: 1.34.3
- How you installed ZoneMinder: PPA
- Full name and version of OS: Ubuntu 18.04.4 LTS
- Browser name and version (if this is an issue with the web interface)

**If the issue concerns a camera**
- Make and Model
- Frame rate
- Resolution
- ZoneMinder Source Type:

**Describe the bug**
I tried enabling hardware acceleration using cuda on GeForce GT 710. Compiled ffmpeg to support hardware acceleration, set DecoderHWAccelName to 'cuda' in monitor source options. Decoding the stream from in terminal seems to work (see attached screenshot). In Zoneminder hardware acceleration gets enabled but leads to errors and capturing fails. Relevant Zoneminder log is attached.

**To Reproduce**
Steps to reproduce the behavior:
1. Set DecoderHWAccelName to 'cuda'
2. Restart Zoneminder

**Expected behavior**
Zoneminder runs using hardware acceleration.

**Debug Logs**
```
02/22/20 00:37:12.199174 zmc_m1[6982].DB1-zm_logger.cpp/255 [LogOpts: level=DB3 effective=DB3, screen=OFF, database=WAR, logfile=DB3->/var/log/zm/zm_debug.log.06982, syslog=WAR]
02/22/20 00:37:12.199314 zmc_m1[6982].DB1-zm_utils.cpp/297 [Detected a x86\x86-64 processor with SSE3]
02/22/20 00:37:12.199548 zmc_m1[6982].DB1-zm_monitor.cpp/2096 [Capture Delay!? 0.000]
02/22/20 00:37:12.199571 zmc_m1[6982].DB1-zm_monitor.cpp/2118 [Got 1 for v4l_captures_per_frame]
02/22/20 00:37:12.199593 zmc_m1[6982].DB2-zm_camera.cpp/55 [New camera id: 1 width: 1920 height: 1080 colours: 4 subpixelorder: 8 capture: 1]
02/22/20 00:37:12.199616 zmc_m1[6982].INF-zm_ffmpeg.cpp/82 [Enabling ffmpeg logs, as LOG_DEBUG+LOG_FFMPEG are enabled in options]
02/22/20 00:37:12.203265 zmc_m1[6982].DB3-zm_image.cpp/443 [Setting up static colour tables]
02/22/20 00:37:12.203400 zmc_m1[6982].DB1-zm_monitor.cpp/423 [monitor purpose=1]
02/22/20 00:37:12.203413 zmc_m1[6982].DB1-zm_monitor.cpp/437 [mem.size(8) SharedData=600 TriggerData=560 VideoStoreData=4128 timestamps=2416 images=151x8294400 = 1252454400 total=1252462168]
02/22/20 00:37:12.203424 zmc_m1[6982].DB2-zm_storage.cpp/66 [Loading Storage for 1 using SELECT `Id`, `Name`, `Path`, `Type`, `Scheme` FROM `Storage` WHERE `Id`=1]
02/22/20 00:37:12.203669 zmc_m1[6982].DB1-zm_storage.cpp/84 [Loaded Storage area 1 'Default']
02/22/20 00:37:12.203746 zmc_m1[6982].DB1-zm_monitor.cpp/441 [Storage path: /mnt/sda1/zoneminder/events]
02/22/20 00:37:12.203777 zmc_m1[6982].DB3-zm_monitor.cpp/553 [Connecting to monitor.  Purpose is 1]
02/22/20 00:37:12.203830 zmc_m1[6982].DB3-zm_monitor.cpp/560 [Success opening mmap file at (/dev/shm/zm.mmap.1)]
02/22/20 00:37:12.203856 zmc_m1[6982].DB3-zm_monitor.cpp/586 [MMap file size is 0]
02/22/20 00:37:12.203879 zmc_m1[6982].DB1-zm_monitor.cpp/591 [Unable to map file /dev/shm/zm.mmap.1 (1252462168 bytes) to locked memory, trying unlocked]
02/22/20 00:37:12.203904 zmc_m1[6982].DB1-zm_monitor.cpp/594 [Mapped file /dev/shm/zm.mmap.1 (1252462168 bytes) to unlocked memory]
02/22/20 00:37:12.203920 zmc_m1[6982].DB3-zm_monitor.cpp/606 [mmapped to 0x7fa38c9d6000]
02/22/20 00:37:12.203936 zmc_m1[6982].DB3-zm_monitor.cpp/626 [Aligning shared memory images to the next 64 byte boundary]
02/22/20 00:37:12.203951 zmc_m1[6982].DB3-zm_monitor.cpp/629 [Allocating 151 image buffers]
02/22/20 00:37:12.204118 zmc_m1[6982].DB3-zm_monitor.cpp/660 [Success connecting]
02/22/20 00:37:13.141639 zmc_m1[6982].DB1-zm_monitor.cpp/514 [Monitor Cam-1-HiRes has function 6,
label format = '%N - %d/%m/%y %H:%M:%S', label X = 0, label Y = 0, label size = 1,
image buffer count = 151, warmup count = 0, pre-event count = 30, post-event count = 125, alarm frame count = 1,
fps report interval = 100, ref blend percentage = 12, alarm ref blend percentage = 12, track motion = 0]
02/22/20 00:37:13.141901 zmc_m1[6982].DB1-zm_zone.cpp/884 [Got 1 zones for monitor Cam-1-HiRes]
02/22/20 00:37:13.141917 zmc_m1[6982].DB3-zm_zone.cpp/759 [Parsing polygon string '0,0 10,0 10,10 0,10']
02/22/20 00:37:13.141928 zmc_m1[6982].DB3-zm_zone.cpp/791 [Got coordinate 0,0 from polygon string]
02/22/20 00:37:13.141936 zmc_m1[6982].DB3-zm_zone.cpp/791 [Got coordinate 10,0 from polygon string]
02/22/20 00:37:13.141945 zmc_m1[6982].DB3-zm_zone.cpp/791 [Got coordinate 10,10 from polygon string]
[... 871 lines cut ...]
02/22/20 00:37:27.839309 zmc_m1[6982].WAR-zm_ffmpeg.cpp/70 [decode_slice_header error]
02/22/20 00:37:27.859086 zmc_m1[6982].WAR-zm_ffmpeg.cpp/70 [no frame!]
02/22/20 00:37:27.873457 zmc_m1[6982].ERR-zm_ffmpeg.cpp/548 [Unable to send packet Invalid data found when processing input, continuing]
02/22/20 00:37:27.893148 zmc_m1[6982].WAR-zm_ffmpeg_camera.cpp/263 [Unable to receive frame 0: code -1094995529 Invalid data found when processing input. error count is 85]
02/22/20 00:37:27.907942 zmc_m1[6982].DB3-zm_ffmpeg.cpp/70 [nal_unit_type: 1, nal_ref_idc: 2]
02/22/20 00:37:27.908037 zmc_m1[6982].ERR-zm_ffmpeg_camera.cpp/57 [Failed to get HW surface format for cuda.]
02/22/20 00:37:27.927572 zmc_m1[6982].ERR-zm_ffmpeg_camera.cpp/60 [Available HW surface format was vdpau.]
02/22/20 00:37:27.941796 zmc_m1[6982].ERR-zm_ffmpeg_camera.cpp/60 [Available HW surface format was vaapi_vld.]
02/22/20 00:37:27.961543 zmc_m1[6982].ERR-zm_ffmpeg_camera.cpp/60 [Available HW surface format was yuvj420p.]
02/22/20 00:37:27.975903 zmc_m1[6982].WAR-zm_ffmpeg.cpp/70 [decode_slice_header error]
02/22/20 00:37:27.995422 zmc_m1[6982].WAR-zm_ffmpeg.cpp/70 [no frame!]
02/22/20 00:37:28.010500 zmc_m1[6982].ERR-zm_ffmpeg.cpp/548 [Unable to send packet Invalid data found when processing input, continuing]
02/22/20 00:37:28.029902 zmc_m1[6982].WAR-zm_ffmpeg_camera.cpp/263 [Unable to receive frame 0: code -1094995529 Invalid data found when processing input. error count is 86]
02/22/20 00:37:28.044848 zmc_m1[6982].DB3-zm_ffmpeg.cpp/70 [nal_unit_type: 1, nal_ref_idc: 2]
02/22/20 00:37:28.044913 zmc_m1[6982].ERR-zm_ffmpeg_camera.cpp/57 [Failed to get HW surface format for cuda.]
02/22/20 00:37:28.063715 zmc_m1[6982].ERR-zm_ffmpeg_camera.cpp/60 [Available HW surface format was vdpau.]
02/22/20 00:37:28.078295 zmc_m1[6982].ERR-zm_ffmpeg_camera.cpp/60 [Available HW surface format was vaapi_vld.]
02/22/20 00:37:28.097828 zmc_m1[6982].ERR-zm_ffmpeg_camera.cpp/60 [Available HW surface format was yuvj420p.]
02/22/20 00:37:28.111950 zmc_m1[6982].WAR-zm_ffmpeg.cpp/70 [decode_slice_header error]
02/22/20 00:37:28.131504 zmc_m1[6982].WAR-zm_ffmpeg.cpp/70 [no frame!]
02/22/20 00:37:28.146438 zmc_m1[6982].ERR-zm_ffmpeg.cpp/548 [Unable to send packet Invalid data found when processing input, continuing]
02/22/20 00:37:28.165833 zmc_m1[6982].WAR-zm_ffmpeg_camera.cpp/263 [Unable to receive frame 0: code -1094995529 Invalid data found when processing input. error count is 87]
02/22/20 00:37:28.181660 zmc_m1[6982].DB2-zm_ffmpeg_camera.cpp/667 [CloseFfmpeg called.]
02/22/20 00:37:28.220446 zmc_m1[6982].DB3-zm_packetqueue.cpp/204 [Deleted (0) packets]
```

connortechnology, 2020-02-24:
Having investigated, this error basically means that the libraries that zm is linking against do not support cuda.  

We have seen reports of this, where someone has compiled their own ffmpeg libs, but zm is still at runtime linking against the standard libraries.  I think this is because the compiled libs are a different version and so both are present.  

Unfortunately I'm not sure of the legality of me providing you with ffmpeg libs with cuda support, and I don't think there is anything I can do code-wise in zm.

Ilia-SB, 2020-02-25:
Hi. Thanks for looking into this. Could you please name the libs in question so that I could check my system for duplicates? Would it make sense to print the libs used and their versions to debug?
I tried copying some of the compiled libs over the installed ones, but no luck so far.

connortechnology, 2020-02-25:
The libs are those that come with ffmpeg. ubuntu 18.04 comes with ffmpeg 3.4.6.
/usr/lib/x86_64-linux-gnu/lib/avcodec.so.57 libavutil.so.55 libavformat.so.57 etc... 

Your best bet is to use apt-get source ffmpeg to get the correct source to replace the ffmpeg libs with the correct version.

Ilia-SB, 2020-02-28:
Sorry to bring it up again, but is it possible that something is still not right here?
I was not able to compile ffmpeg 3.4 with cuda (nvdec) support, and my understanding is that it has been added only in ffmpeg 4.
So with ffmpeg 3.4 the only option is cuvid which is not a valid option for zm.
ffmpeg 4 can't be used because zm requires older libraries.

closed 2020-05-09 by stale[bot]

Jim-Noble, 2020-05-18:
Anyone managed to resolve this?

I'm in a similar situation in that I have a repo-installed version 4.2.2 of ffmpeg, which is linked against libavformat.so.58, and zmc (v1.34.14)  is insisting on linking to libavformat.so.57.

Being a H/W guy, I'm flying blind a little here, but it looks like zm_ffmpeg.cpp is looking for a libavformat minor version of 64. 

https://github.com/ZoneMinder/zoneminder/blob/1.34.14/src/zm_ffmpeg.cpp#L87

Where as the minor version of libavformat v4.2.2 is 29:

https://github.com/FFmpeg/FFmpeg/blob/n4.2.2/libavformat/version.h#L35

Is that the reason it won't link against the libavformat.so.58 that I have?

None of the release versions of ffmpeg have a libavformat with a minor version > 29, and even v4.3-dev is only at 30.

Santiago-vdk, 2020-09-16:
Same issue.
```
root@afc009150e13:/build#  ldd /usr/bin/zmc | grep libav
        libavformat.so.57 => /usr/lib/x86_64-linux-gnu/libavformat.so.57 (0x00007fb649546000)
        libavcodec.so.57 => /usr/lib/x86_64-linux-gnu/libavcodec.so.57 (0x00007fb647e24000)
        libavutil.so.55 => /usr/lib/x86_64-linux-gnu/libavutil.so.55 (0x00007fb647b97000)
root@afc009150e13:/build# ffmpeg -hwaccels
ffmpeg version 3.4.8-0ubuntu0.2 Copyright (c) 2000-2020 the FFmpeg developers
  built with gcc 7 (Ubuntu 7.5.0-3ubuntu1~18.04)
  configuration: --prefix=/usr --extra-version=0ubuntu0.2 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --enable-gpl --disable-stripping --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librubberband --enable-librsvg --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libopencv --enable-libx264 --enable-shared
  libavutil      55. 78.100 / 55. 78.100
  libavcodec     57.107.100 / 57.107.100
  libavformat    57. 83.100 / 57. 83.100
  libavdevice    57. 10.100 / 57. 10.100
  libavfilter     6.107.100 /  6.107.100
  libavresample   3.  7.  0 /  3.  7.  0
  libswscale      4.  8.100 /  4.  8.100
  libswresample   2.  9.100 /  2.  9.100
  libpostproc    54.  7.100 / 54.  7.100
Hardware acceleration methods:
vdpau
vaapi
cuvid
```

TimDowker, 2020-10-23:
Same error here. /usr/bin/zmc shows up for a few seconds in nvidia-smi then disappears.  ffmpeg reports it has the necessary cuda "hardware acceleration method".  Log file full of 

`10/23/20 19:18:20.073905 zmc_m4[1637].ERR-zm_ffmpeg_camera.cpp/56 [Failed to get HW surface format for cuda.]
10/23/20 19:18:20.075348 zmc_m4[1637].ERR-zm_ffmpeg_camera.cpp/59 [Available HW surface format was vdpau.]
10/23/20 19:18:20.076656 zmc_m4[1637].ERR-zm_ffmpeg_camera.cpp/59 [Available HW surface format was vaapi_vld.]
10/23/20 19:18:20.077351 zmc_m4[1637].ERR-zm_ffmpeg_camera.cpp/59 [Available HW surface format was yuvj420p.]
10/23/20 19:18:20.078038 zmc_m4[1637].ERR-zm_ffmpeg_camera.cpp/59 [Available HW surface format was yuvj420p.]
10/23/20 19:18:20.078717 zmc_m4[1637].ERR-zm_ffmpeg.cpp/547 [Unable to send packet Invalid data found when processing input, continuing]
10/23/20 19:18:20.079430 zmc_m4[1637].WAR-zm_ffmpeg_camera.cpp/277 [Unable to receive frame 0: code -1094995529 Invalid data found when processing input. error count is 90]`

Any help would be appreciated.  Looks like it's an issue with the zm_ffmpeg_camera.cpp.

mihaiblaga89, 2020-11-26:
same issue here. Ubuntu 18.04 in docker, ZM 1.34.22

Tony763, 2020-11-27:
I seems that ZM check exact version of libraries: 
https://github.com/ZoneMinder/zoneminder/blob/a67fb0bb16cb9718c5bee74a64f5c6c8fc5f015a/src/zm_ffmpeg_camera.cpp#L43

connortechnology, 2020-11-27:
That check is a minimum.  Prior versions don't have the required support.

Tony763, 2020-11-27:
If i run l`dconfig -v | grep libavcodec `   I get:

```
libavcodec.so.57 -> libavcodec.so.57.107.100 
libavcodec.so.58 -> libavcodec.so.58.113.100
```
 so that should be ok?

Jim-Noble, 2020-11-27:
It's the libavformat check that's failing here isn't it? ZM is looking for 58.64 (major.minor version, with micro version >100). There is no 58.64. Even ffmpeg v4.4dev is only using 58.46.

(There was a 57.64...?)

James

mihaiblaga89, 2020-11-27:
on my end I have 57 that was installed from Ubuntu repos and 58 that I compiled with CUDA support. ZM looked like it was using some of the libs with cuda and some without. I tried deleting 57 with the thought that it will fallback to 58 and also symlinked 58 into 57, no luck, "computer says no".

Tony763, 2020-11-27:
This seems to me as culprit:
```
--enable-nonfree         allow use of nonfree code, the resulting libs
                           and binaries will be unredistributable [no]
```
Libav*  from repo will be without support for cuda. I will try to compile it as well.

Tony763, 2020-11-27:
libav compiled, but produced version is 58.

Tony763, 2020-11-27:
I tried to recompile ZM with new libav, but compilation hangs at:
```
/usr/include/x86_64-linux-gnu/libswresample/swresample.h:191:7: error: ‘AVClass’ does not name a type; did you mean ‘class’?
  191 | const AVClass *swr_get_class(void);
      |       ^~~~~~~
      |       class

```