ZoneMinder/zoneminder #2854: ZoneMinder redirect loop when authenticating with apache basic auth

gitmesomepizza, 2020-02-25:
**Environment**
- ZoneMinder 1.34.3 (Just updated from 1.32)
- Installed via PPA
- Ubuntu Server 18.04.4 LTS (Bionic Beaver)
- Chrome (latest on Windows 10)
- Apache basic auth

**Describe the bug**
After entering apache basic auth credentials into the browser, I get the Chrome "Too many redirects" error.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to http://<IP>/zm
2. Enter Apache basic auth credentials
3. URL changes to "http://<IP>/zm/index.php?view=login"
4. Too many redirects error (Many 302's)

**Expected behavior**
Successful login to zoneminder. My setup worked on ZoneMinder 1.32

**Debug Logs**
```
WAR [No includes/actions/postlogin.php for action login]
WAR [No includes/actions/postlogin.php for action login]
WAR [No includes/actions/postlogin.php for action login]
WAR [No includes/actions/postlogin.php for action login]
...
```
Apache access logs:
```
192.168.x.x - zmuser [24/Feb/2020:20:14:31 -0500] "GET /zm/index.php?view=login HTTP/1.1" 302 570 "-" "Mozilla...
192.168.x.x - zmuser [24/Feb/2020:20:14:31 -0500] "GET /zm/index.php?view=login HTTP/1.1" 302 570 "-" "Mozilla...
192.168.x.x - zmuser [24/Feb/2020:20:14:31 -0500] "GET /zm/index.php?view=login HTTP/1.1" 302 570 "-" "Mozilla...
192.168.x.x - zmuser [24/Feb/2020:20:14:31 -0500] "GET /zm/index.php?view=login HTTP/1.1" 302 570 "-" "Mozilla...
...
```
I will continue to troubleshoot.
My first thought was that it may be related to the includes/actions/postlogin.php file? (which doesn't exist)

gitmesomepizza, 2020-02-29:
It appears that the file www/includes/actions/login.php is not being executed when a login is attempted via index.php?view=login
I can't seem to find the code that includes this file.
Was support for apache basic auth dropped?
I had originally followed this guide which worked for zoneminder 1.32: [https://github.com/pliablepixels/zmNinja/wiki/Apache-Proxy-Authorization-with-HTTPS,-Basic-Auth,-and-ProxyPass](url)

gitmesomepizza, 2020-03-01:
So the updated authentication method using tokens, is what broke my setup.
I enabled OPT_USE_LEGACY_API_AUTH and edited the includes\auth.php file to pass the basic auth credentials from $_SERVER['PHP_AUTH_USER'] and $_SERVER['PHP_AUTH_PW'].
Hopefully there will be an updated way to get ZoneMinder working with basic auth again.

mjnewton, 2020-03-05:
I've tripped over this myself and [raised it on the forum](https://forums.zoneminder.com/viewtopic.php?f=40&t=29150&p=114230) without checking here first.

A workaround for me has been to keep HTTP AUTH enabled but disable OPT_USE_AUTH within ZoneMinder. This means that whilst I can restrict access to ZoneMinder I cannot be more specific on a user-by-user basis once in. This happens to be fine for me and so I may keep it this way even if/when it is fixed.

gitmesomepizza, 2020-03-07:
Thanks for suggesting your workaround. I saw iconnor responded to your forum post, and I can confirm the issue is fixed for me in 1.34.6! Thank you both!

closed 2020-03-07 by gitmesomepizza