ZoneMinder/zoneminder #2862: Tag for 1.24.5 missing version bump commit

mtfurlan, 2020-02-28:
I'm ignoring the bug report template cause this is a git/process thing not a bug in zoneminder.

The tag `1.34.5` doesn't have commit `52cb2441f Bump version to 1.34.5 for release` which is on the `release-1.34` branch.
This means that when on the tag `1.34.5`, packpack builds packages labeled `1.34.4`.
Is this expected, or was this a mistake that should be fixed?

Also 1.34.5 isn't an annotated tag, so `git describe` returns `1.34.1-74-g1b70ed207`.
Not sure if y'all care about this, just slightly confused me cause I use  `__git_ps1` in my PS1, so it didn't show `1.34.5` after I checked out `1.34.5`.

Thanks for developing this project!

connortechnology, 2020-02-28:
This was a mistake I made during the release, but I have corrected it in github... your local repo may need some extra commands to fix it.  

git fetch -P

mtfurlan, 2020-02-28:
Thanks!
I had to delete the local tag manually.

closed 2020-02-28 by mtfurlan