ZoneMinder/zoneminder #2869: Event space not updating

MatoXD99, 2020-03-04:
Environment description
- Zoneminder version: v1.35.0
- OS: Debian 10
- Chrome, Firefox, Explorer, Edge

Bug:
At the beginning, Zoneminder was doing pretty much okay, but through some time it started breaking. I managed to fix the issues, but there is still one remaining. Size of events is not updating, even with checked "Update used disk space". There is still enough space available on the disks for recordings, it's at 40%.

I don't know if that has any connection with anything I described, but I'm getting this error every now and then:

"'zmx10.pl -c start' exited abnormally, exit status 9"

Thank you for help.

connortechnology, 2020-03-04:
There not much information in here to go on.  I would look in the zmfilter logs.  As to the x10 thing, you may have to run it from the command line to find out why it is failing.

sudo zmx10.pl 

should do it.

MatoXD99, 2020-03-05:
I am getting this message:

root@debian:~# sudo zmx10.pl                                                                                                                                                                                                                                                                         Can't locate X10/ActiveHome.pm in @INC (you may need to install the X10::ActiveHome module) (@INC contains: /etc/perl /usr/local/lib/x86_64-linux-gnu/perl/5.28.1 /usr/local/share/perl/5.28.1 /usr/lib/x86_64-linux-gnu/perl5/5.28 /usr/share/perl5 /usr/lib/x86_64-linux-gnu/perl/5.28 /usr/share/perl/5.28 /usr/local/lib/site_perl /usr/lib/x86_64-linux-gnu/perl-base) at /usr/bin/zmx10.pl line 171.                                                                                                                                                                                                BEGIN failed--compilation aborted at /usr/bin/zmx10.pl line 171.

connortechnology, 2020-03-06:
sudo perl -MCPAN -e shell

install X10::ActiveHome

MatoXD99, 2020-03-09:
Thank you, that issue has been fixed, but I am still getting "'zmx10.pl -c start' exited abnormally, exit status 9" for some reason, it shows one of those every 3 minutes :(

connortechnology, 2020-03-09:
I think the x10 talks to a serial port, which the web user probably doesn't have access to.  You might try adding www-data to the dialout group
sudo adduser www-data dialout

I'm just guessing though, really need output/logs

closed 2020-05-16 by stale[bot]