ZoneMinder/zoneminder #2875: ZM needs a websocket

connortechnology, 2020-03-05:
Ideally this would be implemented using websocketpp or better.  It should likely use it's own thread in the zmc.  It should listen on a configurable port range (See Options->Network->MIN_STREAMING_PORT).  

A client should be able to connect to it and request status updates (capture fps, analysis fps, etc.) the current image in various formats (h264, jpeg, raw).  It should be possible to make a single request, or subscribe to a constant stream of events.  Perhaps even at a specified interval.

From the zm side, capture process should be able to queue messages to the socket to be sent to clients. So events like capturing failing, the capture thread can send out a message saying that capture failed, and then when capturing has resumed, another message.  There should be no blocking of the capture thread.

jeroenst, 2020-10-31:
The websocket setting should be a full path, so it's also possible to configure a proxypass for websockets in apache so all trafic can be handled by port 80 or 443.

connortechnology, 2026-05-10:
bountysource is gone and took the money, but I would still honour the bounties.  People might have to be a little patient in payout we would get it done.  I don't remember what the actual $ value of the bounty was on this.

AJ0070, 2026-08-16:
Sorry, I was busy with my GSoC work and could not get back to this.

AJ0070, 2026-08-16:
@connortechnology picking this back up properly now.

Where the websocket work stands: #4830 is out of draft and ready for review. Every review thread is addressed and resolved, including the dedicated port range, the image vs video stream separation, the switch to the packetqueue, and moving the general string helpers into `zm_utils`. It is merged up to master and CI is green.

One question of yours I could not settle on my own: whether to move from the in-tree server to websocketpp. I documented the tradeoffs you asked for and kept it in-tree for now. If you would rather have the migration, I will do it as a follow-up rather than hold the PR on it.

I also split the unrelated pieces out so they can be reviewed on their own: #5050 (comms test reliability) and #5051 (a `UriEncode` non-ASCII bug plus JSON escaping in the go2rtc payloads, both pre-existing on master). And I reworked #4864 to drop the frontend name lookup entirely after Steve's review.

Two things I would like to settle when you have a moment:

1. **The bounty for this issue.** You mentioned you could assign a new amount. Whatever you think is fair works for me, and GitHub Sponsors is fine for payout, including staged over time if that is easier.

2. **The ongoing arrangement.** Back in May you said that if I was interested in ongoing ZM dev, something more regular could be arranged. I am interested. Rather than guess at the shape of it, what did you have in mind in terms of scope, cadence and budget? I would rather fit my availability around what actually works for the project than propose something that does not.

No urgency on either. I will keep picking off smaller issues in the meantime.

AJ0070, 2026-09-01:
@connortechnology back in May you said "I can just assign a new amount. If you are interested in ongoing zm Dev perhaps something more regular could be arranged." I said I was interested at the time, and I still am.

Is that still on the table? A no is genuinely fine, I would just rather know either way.

Separately, #4830 needs your call on websocketpp, and on whether the two concurrency bugs get fixed there or it gets parked.

singhharsh1708, 2026-09-07:
@connortechnology on the ongoing arrangement you mentioned in May, separate from the websocket bounty above, which is @AJ0070's: I would be interested in that for the bug queue. Fifteen of my PRs merged between 2 and 6 September, across the C++, the web layer, the Perl modules, the schema and the docs, with five more open. GitHub Sponsors works for payout, monthly or per issue, and I am flexible on the amount. Give me a list and I will work through it.

connortechnology, 2026-11-06:
> [@connortechnology](https://github.com/connortechnology) back in May you said "I can just assign a new amount. If you are interested in ongoing zm Dev perhaps something more regular could be [...]

Possibly, we don't have a ton of money.  Please email me privately to discuss: isaac@zoneminder.com