ZoneMinder/zoneminder #2883: DCS5020L.pm: conflicting license grants

onlyjob, 2020-03-10:
Header states GPL-2+ but footer disagrees (LGPL-3):

  * https://github.com/ZoneMinder/zoneminder/blob/master/scripts/ZoneMinder/lib/ZoneMinder/Control/DCS5020L.pm#L6-L9
  * https://github.com/ZoneMinder/zoneminder/blob/master/scripts/ZoneMinder/lib/ZoneMinder/Control/DCS5020L.pm#L354

Please clarify.

FYI @zlodag.

connortechnology, 2020-03-30:
Fixed by 3117f29f716193114bd3b1fb7aa48742dbe4d663

closed 2020-03-30 by connortechnology

onlyjob, 2020-03-30:
Thanks. Small cosmetic issues are still there:

  * two GPL-2+ grants (one is redundant).
  * two copyright statements on great distance from each other.
  * inconsistent GPL disclaimers.

Why not just remove the footer and move copyright statement to the header?

connortechnology, 2020-03-31:
because the footer is perldoc.