ZoneMinder/zoneminder #2884: favicon.ico with PNG content

onlyjob, 2020-03-10:
There is an inconsistency between file name and its content:

```
$ file web/api/app/webroot/favicon.ico
web/api/app/webroot/favicon.ico: PNG image data, 32 x 32, 8-bit colormap, non-interlace
```

connortechnology, 2020-03-10:
This file comes with cakephp I think...

onlyjob, 2020-03-10:
There is another file `web/api/lib/Cake/Console/Templates/skel/webroot/favicon.ico` that is definitely came from CakePHP so the `web/api/app/webroot/favicon.ico` appears to be a copy of that file since they both have identical md5sum.

`web/graphics/favicon.ico` have no such problem.

onlyjob, 2020-05-10:
@stale bot practices disrespectful and inefficient issue management approach. :(

connortechnology, 2020-07-10:
So I havn't addressed this because I'm not sure what the correct course of action is.
We could replace it with a png version... but this file comes with cakephp.  It's really cakephp's problem.
We really should update our cakephp code to version 3, which I believe allows us to get rid of all cakephp boilerplate including this file.

connortechnology, 2021-03-10:
This might get fixed with us updating cakephp for fix php8.  Reminder to check