ZoneMinder/zoneminder #2891: User's language can not setup yet

sipi58, 2020-03-18:
- Version of ZoneMinder 1.34.6
- ZoneMinder installed from PPA
- Ubuntu 18.04 LTS 64 bit
- Browser name are Firefox and Chrome

**Describe the bug**
I used individual language setup for users but after update all user language are default. I can not change it but removed the comment two row in skin.php.

I updated zm 1.32.5 to 1.34.6. The update not fixed the problem.

I restarted zm, deleted user and create it newly nothing changed. Please help.

Thanks and Regards

connortechnology, 2020-03-24:
I am unable to reproduce this.

Please confirm that editing  the user, and changing the language, after click save in the user list, is the language updated?

sipi58, 2020-03-25:
I editeded /usr/share/zoneminder/www/skins/classic/skin.php and changed this

// Uncomment if there are language overrides
//if ( $skinLangFile = loadLanguage( ZM_SKIN_PATH ) )
    //require_once( $skinLangFile );
to
// Uncomment if there are language overrides
if ( $skinLangFile = loadLanguage( ZM_SKIN_PATH ) )
    require_once( $skinLangFile );

Next step I edited user and changed language. Save and logout by user. I login again and checked user's language setup. The user's language changed but webcosole not changed.  I choose whatever user's language the webconsole remains English. I tried to restart ZM but anything not changed.

Do I need to change the source elsewhere?

connortechnology, 2020-03-25:
You shouldn't have to edit any code... I don't even know how old those lines are.

To change a user's language you edit the user, select the new language and click save.  It should take effect immediately, no logout/login required.

sipi58, 2020-03-25:
OK. I tried what you recommend without editing the source code. Not logout/login and not restart.

Nothing changed. I tried more user's language selection but nothing....

Zoneminder 1.34.6 64 bit, Ubuntu 18.04 LTS 64 bit (fresh update)

What can I do?

connortechnology, 2020-03-26:
I think we need to do a remote desktop share session.  Teamviewer, zoom, google remote desktop or whatever works for you.

sipi58, 2020-03-26:
This was a fresh install of zm 1.32.5 and was constantly updated to 1.34.6. The web console only English, the user"s language selection is ineffective on the web console since 1.32.5.

I installed a NEW FRESH 1.34.6 and the same problem. Not edited, only user language selected. French, English, German, Russian, Hungarian, Romanian language selection doesn't work. I haven't tried the rest.

Thanks for your suggestion but if the recent installation isn't working then the problem isn't with my installed one.

Sorry, but the remote connection not enabled by policy...

connortechnology, 2020-03-27:
Just a thought here, but are you actually logged in as any of these users?  Is Options->System->OPT_USE_AUTH set to on?  
You can edit and add users without actually turning on  using users.  In which case, the language used in the interface will be set to Options->System->LANG_DEFAULT

connortechnology, 2020-03-27:
Nevermind, I found the problem. I will release 1.34.8 today with the fix. In the meantime, you can move the line at 156
require_once('includes/lang.php');
to around 192 after require_once('includes/auth.php');

sipi58, 2020-03-27:
The OPT_USE_AUTH is checked (ON).

If I change LANG_DEFAULT then web console language will change. It works fine.

Different language for different user not works.  Admin change user's language and user login but web console's language is default.

Sorry, I don't understand what need to change and how... in which file? 'includes/lang.php' includes only 73 rows... 

Thank you for your help.

connortechnology, 2020-03-27:
Sorry /usr/share/zoneminder/www/index.php

sipi58, 2020-03-27:
... and see a miracle! So perfect, it's working.

Many thanks and Regards.

closed 2020-04-02 by connortechnology