ZoneMinder/zoneminder #2894: [api] Call to force an alarm returns wrong event id

mhilbush, 2020-03-21:
- Version of ZoneMinder: 1.34.6
- How you installed ZoneMinder: PPA
- Full name and version of OS: Ubuntu 18.04

**Describe the bug**
The response to `http://server/zm/api/monitors/alarm/id:1/command:on.json` returns a json string that contains the wrong event id. A identical call a second or more later returns the correct event id.

**To Reproduce**
1. The call to `http://server/zm/api/monitors/alarm/id:1/command:on.json` returns
```
{"status":"Alarmed event id: 2702"}
```
2. Wait a second or more

3. The call to `http://server/zm/api/monitors/alarm/id:1/command:on.json` returns
```
{"status":"Alarmed event id: 2795"}
```

**Expected behavior**
In the above scenario, 2795 is the correct event id. That's what should be returned in step 1.

The above behavior would suggest a timing issue or race condition somewhere. While I logged this as an issue with the API, I haven't determined if this is an API issue, or an issue somewhere in the zoneminder core. I'll see if I can figure out how to force an alarm using the command line, at which point I be able to get a better idea whether it's an API issue or not.

mhilbush, 2020-03-21:
I'm going to remove the [api] from the subject.

I can reproduce the problem by calling `zmu -m 2 -a`

On the first invocation of `zmu -m 2 -a` I get:
```
Alarmed event id: 2797
```

On the second invocation of `zmu -m 2 -a` a second or so later I get:
```
Alarmed event id: 2798
```

The correct event id is 2798.

connortechnology, 2020-03-21:
I believe the only way this can happen is if the monitor was already alarmed.

connortechnology, 2020-03-21:
Okay, so this is happening because we are already alarmed, but we only add the Forced Web notes to the new event that follows.  

So I think the fix is to update the notes of the currently running event.

mhilbush, 2020-03-22:
Not sure f his is helpful, but I also saw a case where the returned event id was 0 after the first invocation of `zmu -m 2 -a`. The second invocation a couple seconds later returned the correct event id. I think this was on a monitor that had never been previously alarmed.

commit 3658c3b5 (2020-07-16): Update state to ALARM AFTER we have opened the event, so that we get the correct event Id.  Fixes #2978 Fixes #2894
commit d849ee54 (2020-07-16): Update state to ALARM AFTER we have opened the event, so that we get the correct event Id.  Fixes #2978 Fixes #2894
closed 2020-07-16 by connortechnology