ZoneMinder/zoneminder #2895: 500 Http Status Caused by Missing ZM_TIMEZONE

sluebbert, 2020-03-22:
**Describe Your Environment**
quantumobject/docker-zoneminder:1.34

**If the issue concerns a camera**
It doesn't

**Describe the bug**
Immediately after logging in, I get a 500 http status error and a blank white page.

**To Reproduce**
Pull above docker image and run. I happened to update from the same repo v 1.32.

**Expected behavior**
No 500 http status error with a blank page.

**Debug Logs**

From /var/log/apache2/error.log
```
[Sat Mar 21 21:36:21.797312 2020] [php7:warn] [pid 215] [client 172.18.0.1:34994] PHP Warning:  Use of undefined constant ZM_TIMEZONE - assumed 'ZM_TIMEZONE' (this will throw an Error in a future version of PHP) in /usr/share/zoneminder/www/includes/config.php on line 196, referer: http://host:805/zm/?
[Sat Mar 21 21:36:21.797380 2020] [php7:warn] [pid 215] [client 172.18.0.1:34994] PHP Warning:  Use of undefined constant ZM_TIMEZONE - assumed 'ZM_TIMEZONE' (this will throw an Error in a future version of PHP) in /usr/share/zoneminder/www/includes/config.php on line 197, referer: http://host:805/zm/?
[Sat Mar 21 21:36:21.797440 2020] [php7:warn] [pid 215] [client 172.18.0.1:34994] PHP Warning:  ini_set(): Invalid date.timezone value 'ZM_TIMEZONE', we selected the timezone 'UTC' for now. in /usr/share/zoneminder/www/includes/config.php on line 197, referer: http://host:805/zm/?
[Sat Mar 21 21:36:21.797796 2020] [php7:warn] [pid 215] [client 172.18.0.1:34994] PHP Warning:  Use of undefined constant ZM_COOKIE_LIFETIME - assumed 'ZM_COOKIE_LIFETIME' (this will throw an Error in a future version of PHP) in /usr/share/zoneminder/www/includes/session.php on line 9, referer: http://host:805/zm/?
[Sat Mar 21 21:36:21.797950 2020] [php7:warn] [pid 215] [client 172.18.0.1:34994] PHP Warning:  Use of undefined constant ZM_COOKIE_LIFETIME - assumed 'ZM_COOKIE_LIFETIME' (this will throw an Error in a future version of PHP) in /usr/share/zoneminder/www/includes/session.php on line 31, referer: http://host:805/zm/?
[Sat Mar 21 21:36:21.797968 2020] [php7:warn] [pid 215] [client 172.18.0.1:34994] PHP Warning:  A non-numeric value encountered in /usr/share/zoneminder/www/includes/session.php on line 31, referer: http://host:805/zm/?
[Sat Mar 21 21:36:21.799137 2020] [php7:warn] [pid 215] [client 172.18.0.1:34994] PHP Warning:  localtime(): Invalid date.timezone value 'ZM_TIMEZONE', we selected the timezone 'UTC' for now. in /usr/share/zoneminder/www/includes/auth.php on line 181, referer: http://host:805/zm/?
[Sat Mar 21 21:36:21.799614 2020] [php7:error] [pid 215] [client 172.18.0.1:34994] PHP Fatal error:  Uncaught Exception: DateTime::__construct(): Invalid date.timezone value 'ZM_TIMEZONE', we selected the timezone 'UTC' for now. in /usr/share/zoneminder/www/includes/functions.php:2389\nStack trace:\n#0 /usr/share/zoneminder/www/includes/functions.php(2389): DateTime->__construct()\n#1 /usr/share/zoneminder/www/index.php(207): check_timezone()\n#2 {main}\n  thrown in /usr/share/zoneminder/www/includes/functions.php on line 2389, referer: http://host:805/zm/?
```

I can fix the issue by adding `define( 'ZM_TIMEZONE', 'America/Chicago' );` to the top of the file `/usr/share/zoneminder/www/includes/config.php` and restarting.
Searching this repo, I'm not sure where that is ever defined but I'm not too familiar with php.

brianara3, 2020-03-29:
I get the same issue on a new install under CentOS 7.7.1908 from the RPM Fusion release v1.34.5. The stated fix seems to resolve the issue.

surrealse7en, 2020-04-02:
I can confirm the fix presented works.
CentOS 3.10.0-957.el7.x86_64
rpmfusion-free-updates/x86_64 RPM Fusion for EL 7 - Free - Updates
epel/x86_64                   Extra Packages for Enterprise Linux 7 - x86_64

connortechnology, 2020-04-02:
This is caused by your Configuration table not being up to date.
Do zmupdate.pl -f

closed 2020-04-04 by connortechnology

dahmen, 2020-08-03:
On Debian 10.5 it helps fine:
vi /usr/share/zoneminder/www/includes/config.php
`define( 'ZM_TIMEZONE', 'Europe/Berlin');`

perfect - thanks

DimasenBro, 2022-08-10:
@connortechnology , Hello thank you very much!

commit 3bfcd8eb (2020-04-04): Check for definition of ZM_TIMEZONE before using it. Fixes #2895