ZoneMinder/zoneminder #2896: Eventview: Disable timestamp broken

BlueMax, 2020-03-22:
Disabling the overlay timestamp for eventview by removing the string completely in camera options doesn't work and still overlays a default date mask.
This has been discussed before [here](https://forums.zoneminder.com/viewtopic.php?f=40&t=29164).

ZoneMinder v1.34.6
Alpine Linux

connortechnology, 2020-03-26:
This was an issue in 1.34.5 and was fixed in 1.34.6.  Also in further testing, I cannot recreate this.

I did find an issue where a monitor on a remote server in a multi-server setup was not restarting the capture process so it was proceeding with the old values.  So please confirm whether this may be this issue.

Further, confirm that the actual value stored in the db is being updated.  So bring up the monitor options and check that the timestamp is cleared.

BlueMax, 2020-03-26:
Yes, i can change the value to e.g. '12345' successfully but removing the line falls back to default again.
I'm not using multi-server.

Eightplant, 2020-04-28:
Possibly similar?  I noticed one camera was missing the timestamp.  The string was in the timestamp field.  I removed the field from another camera to test and the overlay went away in viewing.  I pasted the string (%N - %m/%d/%y %H:%M:%S) back in and there is no timestamp overlay but when looking at the timestamp field, the string is there.  Now I have two cameras without a timestamp overlay.  Thanks!

~$ zmc -v                                                  1.34.10

~$ mariadb -V                                              mariadb  Ver 15.1 Distrib 10.1.44-MariaDB, for debian-linux-gnu (x86_64) using r eadline 5.2
--

connortechnology, 2020-06-15:
Can we get an update on this issue since we are at 1.34.16 now?

BlueMax, 2020-06-21:
Still the same here. If field is empty i get default overlay. Values like '%H:%M' do work.

ZM 1.34.16
Alpine Linux
php7-7.4.7

connortechnology, 2020-08-11:
I wonder if this is something like the db having a default value assigned for the column.

Can we get a DESCRIBE Monitors output from mysql?

connortechnology, 2020-08-11:
Yes I think that is it.  We are still creating the db with LabelFormat having a default value, but we dropped it in 1.31.0.  

Do 
ALTER TABLE Monitors MODIFY LabelFormat varchar(64);

BlueMax, 2020-08-11:
Doesn't work here. LabelFormat has been set to NULL manually indeed.

Somehow there are two different overlay timestamps now (at least i've never noticed that). In the 'Preview/realtime' window its a small black/white overlay and on 'event view' its still the transparent bar (with yellow font) at the bottom. They seem to have different fonts and regex results are different. First one shows '%H %H %H' as '23 23 23' and the 2nd one shows as '%23 %23 %23' (or something like that). The first one honors empty string, 2nd (still) does not.

connortechnology, 2020-08-11:
The preview window is controlled by label format.  The yellow font thing is put there by video.js.  we have less control over that.

BlueMax, 2020-08-12:
> The yellow font thing is put there by video.js. we have less control over that.
Not sure if we were talking about the same overlay then. Its exactly this one i was talking about. Can this one be removed by NULL string (or is it intended to be removed this way) as i'm not sure if i do have an individual problem (mariadb or sth) on my side?

BlueMax, 2020-08-13:
You closed the bug report but the issue is about the "yellow font thing" not the b/w overlay. Nothing has changed. Its still there and i can change the label but i cannot remove the bar completely. Isn't there a simple option in video.js to disable the whole bar?

connortechnology, 2020-08-13:
yeah I thought we were talking about the bw thing.  Now we know that it is about the video.js overlay we can move on to addressing that.

commit 56fbfa0d (2020-08-13): Add db update to drop default value for LabelFormat. Fixes #2896
commit bf1c27e2 (2020-08-13): Add db update to drop default value for LabelFormat. Fixes #2896
closed 2020-08-17 by connortechnology