ZoneMinder/zoneminder #2897: Buttons in watch and event view are all gray when using dark CSS

jjakob, 2020-03-23:
**Describe Your Environment**
- Zoneminder v1.34.6
- ppa:iconnor/zoneminder-1.34
- Ubuntu 16.04
- Firefox 74.0

**Describe the bug**
The buttons in the watch and event view are all gray when using dark CSS.

**To Reproduce**
- Go to Options, set CSS_DEFAULT to "dark", save
- logout, close the tab and reopen it, relogin
- click on a camera, all the buttons are gray, even though they can be clicked

**Expected behavior**
The buttons should be different colors as they are with the base css.

alabamatoy, 2020-03-23:
related to https://github.com/ZoneMinder/zoneminder/issues/2849  I thought this was fixed?

jjakob, 2020-03-23:
That was for the default css with thite background (which is fine), this is for the dark css. Here the buttons are all gray and never become blue.

On March 23, 2020 8:27:22 PM GMT+01:00, alabamatoy <notifications@github.com> wrote:
>related to https://github.com/ZoneMinder/zoneminder/issues/2849  I thought this was fixed?
>

jjakob, 2020-05-25:
This is still an issue in 1.34.15

connortechnology, 2021-01-22:
Fixed by cd9f10695dea88f2505096f516d5eabcd365ee0a

closed 2021-01-22 by connortechnology