ZoneMinder/zoneminder #2898: "Step forward" and "Step back" buttons only work 2 times

jjakob, 2020-03-24:
**Describe Your Environment**
- Zoneminder v1.34.6
- ppa:iconnor/zoneminder-1.34
 - Ubuntu 16.04
- Firefox 74.0

**Describe the bug**
In Event view, when paused, the buttons for the next and previous frames ("Step forward", "Step back") only work 2 times, subsequent presses do nothing. The cameras are set to capture MJPEG.

Similar to #2499 but not the same, in that issue more buttons weren't working (next/prev do work here)

**To Reproduce**
Steps to reproduce the behavior:
1. click on Console
2. click on camera's events number (any - hour, day,...)
3. click on a event name
4. pause the event during playback
5. click the "Step forward" or "Step back" buttons.
  They work only 2 times, after that they have no effect.

**Expected behavior**
They work a unlimited number of times.

**Debug Logs**
None that I can see at the time of pressing these buttons, there are other errors that occur when navigating:

These occur when a live view is closed:
```
2020-03-24 10:40:10 | web_php |   | 25170 | ERR | Timed out waiting for msg /var/run/zm/zms-832593s.sock | /usr/share/zoneminder/www/ajax/stream.php | 93
2020-03-24 10:40:10 | web_php |   | 25170 | ERR | No data to read from socket | /usr/share/zoneminder/www/includes/functions.php | 2170
```

Navigating between events with Next/Prev buttons while they are playing:
```
2020-03-24 10:43:42 | web_php |   | 25170 | ERR | Socket  /var/run/zm/zms-893729s.sock does not exist.  This file is created by  zms, and since it does not exist, either zms did not run, or zms exited  early.  Please check your zms logs and ensure that CGI is enabled in  apache and check that the PATH_ZMS is set correctly.  Make sure that ZM  is actually recording.  If you are trying to view a live stream and the  capture process (zmc) is not running then zms will exit. Please go to  http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc  for more information. | /usr/share/zoneminder/www/includes/functions.php | 2170
2020-03-24 10:43:41 | zms_e320615 |   | 15085 | ERR | Unable to send raw frame 53: Broken pipe | zm_eventstream.cpp | 791
```

connortechnology, 2020-03-26:
So I have a problem here.  My problem is that step fwd/back don't work at all and looking at the code, I can see why.  Which makes me really wonder how yours is working twice.    Please verify that it is working at all because it shouldn't be.  

Please let's not confuse issues by talking about things that happen when navigating, when the issue at hand is Step Back and Step Forward.

jjakob, 2020-03-26:
When I click pause, either of the buttons advances the view by 1 frame (the step back button advances it forward too) and exactly 2 times, even if I press forward and back alternately. If I leave the view in paused state, the view advances automatically for one frame at exactly 5 and 10 second points after the moment I pause it, then it stays paused for the rest of the time. It's strange behavior indeed and I can't explain why you're seeing a different result.

I agree to not mix issues, those were only the logs I've seen and can be ignored if they're irrelevant.

jjakob, 2020-05-25:
Retested in 1.34.15: the step forward/back buttons now work more than 2 times, mostly as expected, but:
- when paused, the frame still advances by itself every 5 seconds for a maximum of 2 frames
- the first time a step button is pressed, its action is the reverse of what it should be (e.g. pressing forward the 1st time steps one frame backward, pressing it a 2nd time advances it, it then behaves correctly after the 1st press)

jjakob, 2022-04-17:
Restested on v1.36.14, still the same behavior. Additionally, I haven't noticed previously, but when clicking the step buttons, when you change direction (e.g. the previous step was back, now you click forward) the action is in the previous (opposite) direction on the 1st click (e.g. clicking "back, forward, forward" will do "back, back, forward"). The stepping every 5 seconds when paused is still present but sometimes it steps just once or twice.