ZoneMinder/zoneminder #2903: Script to send created videos to telegram

DanielBorgesOliveira, 2020-04-05:
Hi guys,

I would like to share a python script I created and proved to be very useful for me.
This script monitors the video folder and sends the created videos to telegram, in case of some alarm be triggered.

Here some instructions to make it work:

1. Getting the bot token or create a bot :: instructions here [https://www.siteguarding.com/en/how-to-get-telegram-bot-api-token==(https://www.siteguarding. com / en / how-to-get-telegram-bot-api-token);
2. Getting the chat ID of the group: instructions here
[zm_watchdog.tar.gz] (https://github.com/ZoneMinder/zoneminder/files/4433927/zm_watchdog.tar.gz)
[zm_watchdog.tar.gz] (https://github.com/ZoneMinder/zoneminder/files/4433930/zm_watchdog.tar.gz)

 [https://pupli.net/2019/02/get-chat-id-from-telegram-bot/==(https://pupli.net/2019/02/get-chat-id-from-telegram-bot /);
3. Change line 104 with your bot token;
4. Change line 105 with your chat ID;
5. Change line 100 with the path where the videos are stored.

I run the script in a screen session:
`/ usr / bin / screen -S WATCHDOG -m -d / bin / bash -c '/ usr / bin / python3 / root / watchdog / watchdog_events_zm.py'`

The code can be found here: [https://mega.nz/file/5hMETSxK#3SubuDmfwqG59wGRueYvyqJXvIzhTUjAPL2tCGCFWBs▪(https://mega.nz/file/5hMETSxK#3SubuDmfwqGJTUHFBQFJLQFQLQFBQ

Sorry if this should not be posted here.
If that the case, please deleted this post, please.

Thank you.

pliablepixels, 2020-04-07:
Nice. You might want to post this in the [User Contributions forum](https://forums.zoneminder.com/viewforum.php?f=9)

connortechnology, 2020-04-07:
Yeah I think this belongs in the User Contributions forum.

closed 2020-04-07 by connortechnology

DanielBorgesOliveira, 2020-04-07:
Okay. I will post it in the user forum.

Thank you pals.

caddik5, 2020-05-29:
@DanielBorgesOliveira 
- This is a really interesting script you've written. Thanks for posting. 
I've used Telegram previously but am struggling with getting push notifications setup for Zoneminder. 
Would you mind posting a guide for getting this script setup? 
- I'm sure it would help others also.

DanielBorgesOliveira, 2020-06-01:
Hi @caddik5,

I just updated the code. Please, get the last version [here](https://gist.github.com/DanielBorgesOliveira/d3e578e2b677245cec550e965eae1755).
The setup is pretty straight forward. I thought the provided instructions were sufficient.
Can you describe what is the problem are you facing?

caddik5, 2020-06-01:
Hi @DanielBorgesOliveira 
Thanks for posting and updating. My issues are probably more to do with my skills rather than your explanations. I have have another go this week at your update.

titof2375, 2021-04-04:
Change line 78 with your bot token;
Change line 79 with your chat ID;
Change line 74 with the path where the videos are stored.
hello you find them or is it online?

DanielBorgesOliveira, 2021-04-04:
Hi @titof2375,

I'm not sure what you are asking, but I'm assuming you are asking how to obtain the Bot Token and the Chat ID.
If this is the case, you can get the Bot Token following the instructions [here](https://www.siteguarding.com/en/how-to-get-telegram-bot-api-token) and get the Chat ID following the instructions [here](https://stackoverflow.com/questions/32423837/telegram-bot-how-to-get-a-group-chat-id).

titof2375, 2021-04-04:
I already have a telegram bot
my request is that I understand how you install your script

DanielBorgesOliveira, 2021-04-04:
You have to download the **telegram.py**, **watchdog_events_zm.py**  and **watchdog_logging.py**  from [here](https://gist.github.com/DanielBorgesOliveira/d3e578e2b677245cec550e965eae1755).

In line 78 of the file **watchdog_events_zm.py** you have to change the variable **TelegramToken** to your Bot Token,
In line 79 of the file **watchdog_events_zm.py** you have to change the variable **TelegramChatID** to the chat ID of the chat where the bot has to send messages.
In line 74 of the file **watchdog_events_zm.py** you have to change the variable path to point to the **path** where zoneminder stores the video files. This is usually "/var/cache/zoneminder/events".

The last thing you should do is call the file **watchdog_events_zm.py** like this:
# /usr/bin/python3 watchdog_events_zm.py'

titof2375, 2021-04-11:
The last thing you should do is call the file watchdog_events_zm.py like this:
/usr/bin/python3 watchdog_events_zm.py'

hello, I do not understand what to do? do I have to rename it or place it in the directory?

DanielBorgesOliveira, 2021-04-12:
Hi @titof2375 

> The last thing you should do is call the file watchdog_events_zm.py like this:
/usr/bin/python3 watchdog_events_zm.py'

This video shows how to run python from Windows: [https://www.youtube.com/watch?v=Qi28uPKaH_A](https://www.youtube.com/watch?v=Qi28uPKaH_A).
This video shows how to run python from Linux: [https://www.youtube.com/watch?v=asnGiNwo4RU](https://www.youtube.com/watch?v=asnGiNwo4RU).

> hello, I do not understand what to do? do I have to rename it or place it in the directory?
and the three downloaded file I put them where?

Download all the files. Put it in some folder you choose. Lastly, you have to run the code.

sbethge, 2021-04-12:
Hello Daniel,
thanks for this useful script.
How about filters? 
I have set a filter for send Videos via email for out off ofice time. 
Your script seems send every event and ignore filters.

DanielBorgesOliveira, 2021-04-13:
Hi @sbethge,

That is true.
I did this script to send files that are created in a specific situation.
In this case, I used zones to defined what is filmed.
This is a real-time delivery system.
I made the script to send the video as fast as possible.
Because of this, I think that filters are not applicable.

ysechynskyy, 2022-12-30:
for some reason script send to me 2 videos. first one  /events/1/2022-12-30/7987/7987-video.mp4.incomplete (video dosent work) and second  /events/1/2022-12-30/7987/7987-video.mp4... 

So i fixed it by adding line `                not '.incomplete' in event.src_path: # This avoid send file than once.`

[image]

my Dockerfile
```
FROM python:3
WORKDIR /app
RUN apt update && apt install lsof -y
COPY requirements.txt .
RUN pip3 install -r requirements.txt
COPY *.py .

CMD python3 -u  watchdog_events_zm.py
```

requirements.txt

```
watchdog
requests
```

docker-compose.yaml

```
version: '3.1'
services:
  zoneminder:
    container_name: zoneminder
    image:  zoneminderhq/zoneminder:latest-el7
    restart: unless-stopped
    ports:
      - 8883:443/tcp
      - 9000:9000/tcp
      - 80:80
    network_mode: "bridge"
    privileged: false
    shm_size: 7G
    environment:
      - TZ=Europe/Warsaw
      - PUID=99
      - PGID=100
      - MULTI_PORT_START=0
      - MULTI_PORT_END=0
    volumes:
      - /dockerdata/zoneminder/events:/var/lib/zoneminder/events
      - /dockerdata/zoneminder/mysql:/var/lib/mysql
      - /dockerdata/zoneminder/logs:/var/log/zm

  watchdog:
    container_name: watchdog
    build: watchdog/.
    networks:
      - monitor-net
    volumes:
      - /dockerdata/zoneminder/events:/events
    environment:
      - TZ=Europe/Warsaw
    privileged: true

networks:
  monitor-net:
    driver: bridge

```

DanielBorgesOliveira, 2022-12-31:
Hi @ysechynskyy,

Thank you very much for the contribution.

leonopulo, 2023-06-28:
Hello Daniel.
Please write in the description that the screen script must be run under sudo.
Otherwise, lsof will not have access to the event.src_path file.